linuxwacom/xf86-input-wacom #103: No more right clicks?

nicola-d, 2020-03-08:
Hi,

I'm the owner of a Wacom Bamboo 16FG 4x5 (ID 056a:00de Wacom Co., Ltd CTH-470 [Bamboo Fun Pen & Touch]).

Using the driver up to version 0.37 I was able to use two fingers for a right click. Whenever I upgrade the version to a more recent one I no longer can make right clicks.

I tried to compare the output of xev -event button and the differences are (for a two fingers right click gesture):

Version 0.37:

ButtonPress event, serial 25, synthetic NO, window 0x6600001,
    root 0x247, subw 0x0, time 641800, (106,46), root:(464,2114),
    state 0x0, button 3, same_screen YES

ButtonRelease event, serial 25, synthetic NO, window 0x6600001,
    root 0x247, subw 0x0, time 641800, (106,46), root:(464,2114),
    state 0x400, button 3, same_screen YES

Version 0.39:

ButtonPress event, serial 25, synthetic NO, window 0x4c00001,
    root 0x247, subw 0x0, time 109310, (134,69), root:(136,1532),
    state 0x0, button 5, same_screen YES

ButtonRelease event, serial 25, synthetic NO, window 0x4c00001,
    root 0x247, subw 0x0, time 109310, (134,69), root:(136,1532),
    state 0x1000, button 5, same_screen YES

How can I remap the buttons in the newer versions?

Thank you in advance and best regards,
Nicola

whot, 2020-03-08:
button 5 is the scroll wheel button, so the two-finger interaction is interpreted as scrolling.

Random guess: 65b7c37f0fac5cdd03ff7f977124148d0bdf4be8 looks like it could be the cause. @jigpu, does that sound right?

jigpu, 2020-03-09:
Its possible. @nicola-d can you please use our [capture.sh tool](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/82fb9362998dd8ca582b4efb558ec3abfb11668d/capture.sh) to record yourself performing a right-click gesture? Choose your Bamboo's touchpad when prompted, perform the right-click gesture, and stop recording with CTRL+C. Attach the generated tarball to this bug for review.

nicola-d, 2020-03-09:
Here it is. Thank you
[record_1583795718.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/4309428/record_1583795718.tar.gz)

jigpu, 2020-03-11:
I can confirm that the behavior changed with commit 65b7c37. Now moving on to understanding the root cause.

sebasmonia, 2020-08-21:
I think this is causing a problem also with touchscreens for Lenovo Yoga laptops. I can see using `xev` that many taps with two fingers in a row might generate a `Button 3` event but it is completely unreliable.

Tried modifying `TapTime` and other parameters to no avail.

pjanx, 2021-02-03:
Yes, since some time it's almost impossible to trigger a right click on my Intuos Pro M and S likewise.

I only manage to do it when being extremely precise using the 2FG right click method as per wacom(4).

#28 might contain a workaround by using libinput; haven't tested it yet.

thstyl2000, 2021-06-21:
It's working, but one loses the configurations of the stylus also. I did not find out any way to separately reconfigure the "tablet" (or better "unconfigure" the touchpad, since one cannot match the stylus).

abishekmuthian, 2021-09-06:
I was lead to this discussion by the closing of #165 .

Using Libinput instead of xf86-input-wacom might indeed fix the trackpad woes especially the right clicks because I have now moved on to Wayland on Upstream GNOME and my Intuos CTH-480 works nearly flawless.

Nearly because, rarely when I wake the display from sleep there's some temporary weird quirks with cursor, selection of text in some websites. Not entirely sure whether it's related to the Libinput but right clicks are an non issue on Wayland AFAIK.

dodisoft, 2022-08-20:
Hello,
Lead here from #262, now closed.
As far as I'm concerned, trying Libinput didn't cure this behavior (but I'm not on Wayland).
With xserver-xorg-input.wacom 1.1.0-1 on debian testing (bookworm), my tablet acts now in a wacky way:
- 2 fingers right clicks not recognized anymore.
- 2 fingers scrolling and "pinch-to-zoom" sometimes mistaken.
- loss of accuracy in those gesture commands.
- (double-tap to click works well.)

I created a capture of a right-click according to jigpu's method (with 1.1.0-1 version). Find it attached if it can help...
[record_1661030948.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/9388081/record_1661030948.tar.gz)

My last working driver was xserver-xorg-input-wacom 0.34.99.1-1+b1.
I guess I'll stay stuck  with it as long as this bug is not figured out.
Thanks for your work !

abishekmuthian, 2022-08-21:
@dodisoft  On GNOME 42.3/Wayland + Arch Linux, All the issues I had with the tablet seems to have resolved. It hangs very rarely, Which I'm not sure is the fault of linuxwacom (or) the hardware.

Pinglinux, 2022-08-26:
Please test the above patch and let me know your result. Thanks!

dodisoft, 2022-08-27:
@Pinglinux 
I guessed I just had to install your last fork from sources ?
( https://github.com/Pinglinux/xf86-input-wacom/tree/fix-2fg-right-click )
I did this, but i didn't recover the 2fg right-click.
Single/double tap, scroll, pinch/spread stay operational.
Thanks for your time.

@abishekmuthian 
Thanks for your return. It seems that Wayland rely on libinput to deal with our wacom tools.
I had a try with libinput on my setup, but I totally lost the "Pinch" gesture by this way.

Pinglinux, 2022-08-28:
@dodisoft Yes, you basically need the one line change from my patch. I tested it myself. Right-click was back if I apply the patch. Way steps did you use to install the driver from source?

abishekmuthian, 2022-08-28:
> @abishekmuthian Thanks for your return. It seems that Wayland rely on libinput to deal with our wacom tools. I had a try with libinput on my setup, but I totally lost the "Pinch" gesture by this way.

Oh! I presumed it uses linuxwacom. Btw, Pinch to Zoom works fine for me on Firefox, Image Viewer (those were the places I tested). Another gesture I use is three finger navigation in the browser for (Back & Forward).

dodisoft, 2022-08-28:
@Pinglinux 
I uninstalled my installed version (1.1.0-1):
`sudo apt remove xserver-xorg-input-wacom`

I cloned your git sources (assuming that wcmTouchFilter.c was patched in your version?):
`git clone https://github.com/Pinglinux/xf86-input-wacom.git`

I installed the build dependencies:
`sudo apt-get install autoconf pkg-config make xutils-dev libtool xserver-xorg-dev$(dpkg -S $(which Xorg) | grep -Eo -- "-hwe-[^:]*") libx11-dev libxi-dev libxrandr-dev libxinerama-dev libudev-dev`

I opened my terminal in the cloned directory:
`cd xf86-input-wacom`

I launched the build/install from there, according to the autotools linuxwacom recipe :
`set -- --prefix="/usr" --libdir="$(readlink -e $(ls -d /usr/lib*/xorg/modules/input/../../../ | head -n1))"
if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"`

The installation, as such, went well. I rebooted, and nothing changed.

Pinglinux, 2022-08-28:
Ok, I tested again, with a display tablet and an opaque tablet, respectively. I see two potential issues.

1. With display tablet, you need to make sure gesture is turned on:
$xsetwacom list
Wacom HID 4900 Pen stylus       	id: 15	type: STYLUS    
Wacom HID 4900 Finger touch     	id: 16	type: TOUCH     
Wacom HID 4900 Pen eraser       	id: 26	type: ERASER
$xsetwacom get 16 gesture 
off
----------You need to turn gesture on for TOUCH
$xsetwacom set 16 gesture on

2. With Opaque tablet (you are most likely using an opaque tablet), make sure wacom_drv.so is driving the tablet, not libinput. Check your ~/.local/share/xorg/Xorg.0.log, you'd see something like below messages, if your touch device wasn't driving by wacom X driver:
[    36.860] (II) config/udev: Adding input device Wacom Intuos Pro M Finger (/dev/input/event13)
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "evdev touchpad catchall"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "evdev tablet catchall"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "libinput touchpad catchall"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "libinput tablet catchall"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "Wacom USB tablet class"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "Wacom USB touchpad class"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "Wacom tablet class"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "Wacom touchpad class"
[    36.860] (**) Wacom Intuos Pro M Finger: Applying InputClass "libinput overrides wacom touchpads"
[    36.860] (II) Using input driver 'libinput' for 'Wacom Intuos Pro M Finger"

Hope this helps.

dodisoft, 2022-08-28:
1. `xsetwacom list`

Wacom Bamboo 2FG 6x8 Pen stylus 	id: 8	type: STYLUS    
Wacom Bamboo 2FG 6x8 Pad pad    	id: 9	type: PAD       
Wacom Bamboo 2FG 6x8 Finger touch	id: 10	type: TOUCH     
Wacom Bamboo 2FG 6x8 Pen eraser 	id: 13	type: ERASER    

xsetwacom get 10 gesture
on

2. To make it short:
 `sudo cat /var/log/Xorg.0.log | grep -i "using input driver 'wacom'"`

[    40.029] (II) Using input driver 'wacom' for 'Wacom Bamboo 2FG 6x8 Pen'
[    40.182] (II) Using input driver 'wacom' for 'Wacom Bamboo 2FG 6x8 Pad'
[    40.343] (II) Using input driver 'wacom' for 'Wacom Bamboo 2FG 6x8 Finger'
[    40.473] (II) Using input driver 'wacom' for 'Wacom Bamboo 2FG 6x8 Pen eraser'

Wacom seems to deal with it, though the "touch" option may be unclear ?
Find here more of my Xorg.0.log file concerning Wacom:
[Xorg.0.log_part.txt](https://github.com/linuxwacom/xf86-input-wacom/files/9440504/Xorg.0.log_part.txt)

I could have done something wrong during the last installation ? 
How can I check which version of the driver is actually used ?
Thanks a lot for your time.

Pinglinux, 2022-08-29:
Your TOUCH is driving by wacom driver. That's good. You won't be able to tell the difference of the code by the driver version since they'll be the same (I didn't change the version number). But, I do see another potential that you are running the old code.
After "git clone https://github.com/Pinglinux/xf86-input-wacom.git", you need to switch to fix-2fg-right-click branch. Otherwise, you are building the driver from the master branch, which doesn't include the patch I submitted.

So, please do the following:

$git clone https://github.com/Pinglinux/xf86-input-wacom.git
$cd xf86-input-wacom
$git checkout -b fix-2fg-right-click remotes/origin/fix-2fg-right-click
$set -- --prefix="/usr" --libdir="$(readlink -e $(ls -d /usr/lib*/xorg/modules/input/../../../ | head -n1))" if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"

And test again...

dodisoft, 2022-08-29:
Tested successfully ! 
2fg-right-click back again on my Bamboo !
Thanks to guide me along all that Git Alchemy to achieve this.
Hoping this commit will meet our debian repositories very soon...
Have a nice day !

whot, 2022-08-31:
Fixed in #285

closed 2022-08-31 by whot