linuxwacom/xf86-input-wacom #104: Acquiring touch events

polygonalenippel, 2020-03-10:
Heya, not an issue but more of a question.

I am developing a userspace application which requires registering simultaneous touch events on a tablet on Linux. Roughly which part of the kernel/driver stack would be the most interesting to hack into to get this information? I am quite new to interfacing with devices on Linux.

whot, 2020-03-12:
What toolkit are you using? Both GTK and Qt have touch support, much easier to rely on that to take away the lower-level difficulties. Touch support in X and Wayland look quite different (XI2 vs `wl_touch`) and require different handling. Again, the toolkit would largely abstract this away from you.

polygonalenippel, 2020-03-12:
@whot I wish not to use any toolkit in particular, because I am merely interested in just getting the touch points at any given time as input for a different application.

By using a toolkit, you usually get such events whenever an event is enacted upon, say, a window of said toolkit.

I tested this with Qt: not only did I necessarily need to display a window, I could only get the touch events which came from that window. I did something along the lines of this if you're interested:

```cpp
class C : public QWidget {...}

T C::event(QEvent *event) {
  switch (event->type()) {
    case QEvent::TouchBegin:
    case QEvent::TouchUpdate:
    case QEvent::TouchEnd:
    { std::cout << "touch event" <<std::endl; break;}
    default: return QWidget::event(event);
    // ...
```
(For those playing along at home, make sure touch is enabled and gestures are turned off. Furthermore, it works quite spotty, the first touch does not get registered in my case until a second simultaneous touch is applied.)

Am I missing something? If not, what else could I try? Interfacing with X?

whot, 2020-03-12:
Two ways in that case: in XI2 there are raw events and you can register to those. It's not what I recommend though, they may work but have subtle behaviour policies (and bugs). They're also not available in Wayland for good reason.

The other alternative: you're effectively looking for a side-channel and for that you have to listen to the kernel device directly. This requires root (or the `input` group on your user) and comes with all the disadvantages. Depending on how processed you want the touches to be, you can either [parse the evdev protocol](http://who-t.blogspot.com/2016/09/understanding-evdev.html) yourself (use `libevdev` or `python-libevdev`) but note there are [*a lot* of subtleties](http://who-t.blogspot.com/2018/07/why-its-not-good-idea-to-handle-evdev.html). Or you could use libinput against that kernel device and then parse the touch events as they come out of libinput. For your own sanity, I recommend that. Search for the `libinput-debug-events.c` source file in the libinput git tree and that'll have most of what you need.

polygonalenippel, 2020-07-27:
@whot I realized I never replied back. I ended up using libinput's debug events. Thanks for your comment, much appreciated.

closed 2020-07-27 by polygonalenippel