linuxwacom/xf86-input-wacom #105: Intuos BT S - CTL-4100WL working but not being displayed by xsetwacom

mr-ubik, 2020-03-12:
I have recently bought a Wacom Intuos BT S.

`libwacom-list-local-devices` correctly lists it. 

The tablet works perfectly but `xsetwacom` does not see it (And the GUI for KDE as well).

The system is Arch Linux with a fresh installation `linuxwacom` packages.

whot, 2020-03-13:
does arch default to Wayland yet? If so, then xsetwacom can't talk to the driver because there is no wacom driver (it's handled by libinput within the compositor) but it should also complain about that.

Otherwise, it's likely that you're not using wacom driver for the tablet device. Check with `xinput list-props "device name"`, get the device name with `xinput list`. If the properties are prefixed with libinput you're using the libinput xorg driver. Make sure the xf86-input-wacom package is actually installed, it should also install an override to prefer the wacom driver. Otherwise, and Xorg.log will be useful to debug further.

mr-ubik, 2020-03-13:
I don't think it does. I am currently running the most recent Plasma which should still be X11.
`xf86-input-wacom` is installed but `libinput` is used.

```console-output
Device 'Wacom Intuos BT S Pad':
        Device Enabled (161):   1
        Coordinate Transformation Matrix (163): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Send Events Modes Available (281):     1, 0
        libinput Send Events Mode Enabled (282):        0, 0
        libinput Send Events Mode Enabled Default (283):        0, 0
        Device Node (284):      "/dev/input/event11"
        Device Product ID (285):        1386, 887
```

jigpu, 2020-03-13:
Can you run `lsusb` for me? Its possible your tablet is running in "Android mode", which changes its vendor ID from 0x56A to 0x2D1F. Normally this also causes other odd behavior (see [the relevant FAQ item](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect)), but it would be good to know for sure. You could also try the steps mentioned in the FAQ to change mode and see if that improves your situation.

mr-ubik, 2020-03-14:
I have already tried them when the tablet goes with the dimmer light (which by a Wacom video I gather should be the Android mode indicator) it stops working. Once I put it back in "PC" mode it works as before.

`lsusb` outputs the following:

```console-output
...
Bus 001 Device 005: ID 056a:0376 Wacom Co., Ltd CTL-4100WL [Intuos BT (S)]
...
```

mr-ubik, 2020-03-17:
I have found this in the Arch Wiki https://wiki.archlinux.org/index.php/Wacom_tablet#Tablet_recognized_but_xsetwacom_and_similar_tools_do_not_display_it I will try and see if that's the issue.

closed 2020-03-17 by mr-ubik