linuxwacom/xf86-input-wacom #107: Jumps when writing with pen

feijoas, 2020-03-31:
When writing with my Lenovo active Pen 2 on my Lenovo ThinkPad X1 Yoga (4th Gen), I have sudden jumps like in the screenshot:

[image]

I attached the recording from the screenshot in the capture.sh script:
[record_1585639488.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/4407397/record_1585639488.tar.gz)

tattsan, 2020-04-05:
I have the same PC, and the same problem with Wacom Bamboo Ink stylus pen (2nd genaration). I use it on Arch Linux.

feijoas, 2020-04-06:
Did you find a pen that works?

jigpu, 2020-04-06:
@feijoas and @tattsan can you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball to this issue. This will give me a little more information about the environment to go along with the recording analysis.

feijoas, 2020-04-06:
@jigpu Thanks for looking into this. Please find the info attached.
[sysinfo.c14bWadVBh.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/4439562/sysinfo.c14bWadVBh.tar.gz)

jigpu, 2020-04-06:
Thanks! Looks like the HID descriptor and used report ID is similar to the bug report you referenced, but the issue itself is different. Your sensor is reporting proper pressure values and seems to be properly tracking tip events most of the time. The artifact that you captured seems to occur near a moment where the pen is on the edge of being in-range of the sensor. The issue seems to be with the event stream that we're getting from the hardware rather than the kernel itself.

Have you always noticed the presence of this issue? If you have Windows installed on another partition, it would be interesting to know if you can replicate it there or not.

jigpu, 2020-04-06:
For reference, event snippets of the top stroke of the final "T" with the artifact:

### Evemu
~~~
E: 15.370540 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 15.380460 0003 0000 9916	# EV_ABS / ABS_X                9916
E: 15.380460 0003 0001 5518	# EV_ABS / ABS_Y                5518
E: 15.380460 0003 0018 0999	# EV_ABS / ABS_PRESSURE         999
E: 15.380460 0001 014a 0001	# EV_KEY / BTN_TOUCH            1
E: 15.380460 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.380460 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 15.385527 0003 0000 9907	# EV_ABS / ABS_X                9907
E: 15.385527 0003 0018 1501	# EV_ABS / ABS_PRESSURE         1501
E: 15.385527 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.385527 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 15.395457 0003 0000 9897	# EV_ABS / ABS_X                9897
E: 15.395457 0003 0001 5514	# EV_ABS / ABS_Y                5514
E: 15.395457 0003 0018 1853	# EV_ABS / ABS_PRESSURE         1853
E: 15.395457 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.395457 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 15.400490 0003 0000 9894	# EV_ABS / ABS_X                9894
E: 15.400490 0003 0001 5512	# EV_ABS / ABS_Y                5512
E: 15.400490 0003 0018 2125	# EV_ABS / ABS_PRESSURE         2125
E: 15.400490 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.400490 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 15.410502 0003 0000 9897	# EV_ABS / ABS_X                9897
E: 15.410502 0003 0001 5509	# EV_ABS / ABS_Y                5509
E: 15.410502 0003 0018 2297	# EV_ABS / ABS_PRESSURE         2297
E: 15.410502 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.410502 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 15.415477 0003 0000 9913	# EV_ABS / ABS_X                9913
E: 15.415477 0003 0001 5501	# EV_ABS / ABS_Y                5501
E: 15.415477 0003 0018 2443	# EV_ABS / ABS_PRESSURE         2443
E: 15.415477 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.415477 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 15.425477 0003 0000 9946	# EV_ABS / ABS_X                9946
E: 15.425477 0003 0001 5497	# EV_ABS / ABS_Y                5497
E: 15.425477 0003 0018 2560	# EV_ABS / ABS_PRESSURE         2560
E: 15.425477 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.425477 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 15.430497 0003 0000 9982	# EV_ABS / ABS_X                9982
E: 15.430497 0003 0001 5479	# EV_ABS / ABS_Y                5479
E: 15.430497 0003 0018 2673	# EV_ABS / ABS_PRESSURE         2673
E: 15.430497 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.430497 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 15.440524 0003 0000 10038	# EV_ABS / ABS_X                10038
E: 15.440524 0003 0001 5459	# EV_ABS / ABS_Y                5459
E: 15.440524 0003 0018 2789	# EV_ABS / ABS_PRESSURE         2789
E: 15.440524 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.440524 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 15.445456 0003 0000 10102	# EV_ABS / ABS_X                10102
E: 15.445456 0003 0001 5437	# EV_ABS / ABS_Y                5437
E: 15.445456 0003 0018 2900	# EV_ABS / ABS_PRESSURE         2900
E: 15.445456 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.445456 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 15.455499 0003 0000 10168	# EV_ABS / ABS_X                10168
E: 15.455499 0003 0001 5409	# EV_ABS / ABS_Y                5409
E: 15.455499 0003 0018 2992	# EV_ABS / ABS_PRESSURE         2992
E: 15.455499 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.455499 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 15.460480 0003 0000 10244	# EV_ABS / ABS_X                10244
E: 15.460480 0003 0001 5382	# EV_ABS / ABS_Y                5382
E: 15.460480 0003 0018 3072	# EV_ABS / ABS_PRESSURE         3072
E: 15.460480 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.460480 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 15.470508 0003 0000 10329	# EV_ABS / ABS_X                10329
E: 15.470508 0003 0001 5358	# EV_ABS / ABS_Y                5358
E: 15.470508 0003 0018 3120	# EV_ABS / ABS_PRESSURE         3120
E: 15.470508 0004 0000 -1324056558	# EV_MSC / MSC_SERIAL           -1324056558
E: 15.470508 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
[... 60 lines cut ...]
E: 000015.470536 18 06 21 59 28 ee 14 30 0c 19 00 12 80 14 b1 00 ff 00 00
# ReportID: 6 / Tip Switch: 1 | Barrel Switch: 0 | Eraser: 0 | Invert: 0 | Secondary Barrel Switch: 0 | In Range: 1 | # | X:  10411 | Y:   5329 | Tip Pressure:   3153 | 0xff00005b:     25 | Transducer Serial Number: -1324056558 | Undefined:    0 | Battery Strength:  255 | X Tilt:    0 | Y Tilt:    0 
E: 000015.475543 18 06 21 ab 28 d1 14 51 0c 19 00 12 80 14 b1 00 ff 00 00
# ReportID: 6 / Tip Switch: 1 | Barrel Switch: 0 | Eraser: 0 | Invert: 0 | Secondary Barrel Switch: 0 | In Range: 1 | # | X:  10501 | Y:   5305 | Tip Pressure:   3167 | 0xff00005b:     25 | Transducer Serial Number: -1324056558 | Undefined:    0 | Battery Strength:  255 | X Tilt:    0 | Y Tilt:    0 
[... cut ...]
# ReportID: 6 / Tip Switch: 0 | Barrel Switch: 0 | Eraser: 0 | Invert: 0 | Secondary Barrel Switch: 0 | In Range: 1 | # | X:  30935 | Y:   4860 | Tip Pressure:      0 | 0xff00005b:      0 | Transducer Serial Number:           0 | Undefined:    0 | Battery Strength:    0 | X Tilt:    0 | Y Tilt:    0 
E: 000015.584387 18 06 20 d7 78 fc 12 00 00 00 00 00 00 00 00 00 00 00 00

## Following events track the hovering pen (with a serial number) nicely from this XY position to the start of the next stroke
~~~

feijoas, 2020-04-06:
> Have you always noticed the presence of this issue? If you have Windows installed on another partition, it would be interesting to know if you can replicate it there or not.

This is a new laptop and I noticed this issue from day 1. I don't have Windows available, but noticed the same issue with a Ubuntu Live CD.

Tobstr02, 2021-01-03:
I have the same issue on Ubuntu 20.04.1 LTS
I tried to update the driver but the problem is still persistent.

Its only when i pressed a button on the pen recently, after a bunch of drawed lines the bug dissapears.
I used the pen for a few months and everything was fine until today.

jigpu, 2021-01-06:
@Tobstr02 can you please provide the output from our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) as well as a recording from the [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/82fb9362998dd8ca582b4efb558ec3abfb11668d/capture.sh) which contains an example of the bug occurring?

tormec, 2021-05-20:
I remember I had the same issue when I bought my Lifebook T4220 (second-hand). I managed to solve the problem by simply removing and reattaching the tip.

Pinglinux, 2022-08-11:
The "## Large timestamp jump" indicates missing tablet events, which could be a hardware or system configuration issue. Similar issue reported at https://gitlab.freedesktop.org/libinput/libinput/-/issues/464 was closed 2 years ago.

closed 2022-08-11 by Pinglinux