linuxwacom/xf86-input-wacom #108: How would I change the active area of the tablet?

RTSAjwad, 2020-04-02:
I'm confused as to how I could replicate this functionality on Linux:
https://imgur.com/a/pHBhG

For example, If i wanted a certain area of my tablet that is smaller than its actual area to map to my screen.

jigpu, 2020-04-02:
I don't know if we have a clear tutorial, but what you want to do is modify your tablet's "Area" property to set the active area. Here's an example of how you can use the Area property to set up different active areas and reset them if necessary. You aren't limited to using a quarter of the tablet either; its just much easier to demo since the math is easy (just divide X and Y by 2).

~~~
# List available devices:
$ xsetwacom list
Wacom Intuos Pro M Pen stylus   	id: 13	type: STYLUS    
Wacom Intuos Pro M Pen eraser   	id: 14	type: ERASER    
Wacom Intuos Pro M Pen cursor   	id: 15	type: CURSOR    
Wacom Intuos Pro M Pad pad      	id: 16	type: PAD       
Wacom Intuos Pro M Finger touch 	id: 17	type: TOUCH

# Read the current active area of the stylus. Result defines a rectangle (x0, y0) (x1, y1)
$ xsetwacom get "Wacom Intuos Pro M Pen stylus" area
0 0 44800 29600

# Limit active area to upper-left quarter of tablet (halve x1 and y1):
$ xsetwacom set "Wacom Intuos Pro M Pen stylus" area 0 0 22400 14800

# Read back the active area and see that it is set to what we requested:
$ xsetwacom get "Wacom Intuos Pro M Pen stylus" area
0 0 22400 14800

# Limit active area to the upper-right quarter of the tablet:
$ xsetwacom set "Wacom Intuos Pro M Pen stylus" area 22400 0 44800 14800

# Limit active area to the bottom-right quarter of the tablet:
$ xsetwacom set "Wacom Intuos Pro M Pen stylus" area 22400 14800 44800 29600

# Reset active area to the default value:
$ xsetwacom set "Wacom Intuos Pro M Pen stylus" area -1 -1 -1 -1

# Read the active area again and see that it has been reset to the default:
$ xsetwacom get "Wacom Intuos Pro M Pen stylus" area
0 0 44800 29600
~~~

Note that while using xsetwacom is great for experimenting, if you want a change to be "permanent", you should use an xorg.conf.d file to define the area rather than relying on xsetwacom. Note that you will need to fill in the "X0", "Y0", "X1", and "Y1" with the values used by xsetwacom.

Save as **/etc/X11/xorg.conf.d/90-wacom-area.conf**
~~~
# Restrict the tablet active area. Note that if you have multiple tablets connected
# you should add a rule like the following inside the InputClass to ensure that
# the active area only applies to the intended device.
#     MatchUSBID "XXXX:YYYY"

Section "InputClass"
        MatchDriver "wacom"
        MatchIsTouchpad "false"
        MatchIsTouchscreen "false"
        Option "Area" "X0 Y0 X1 Y1"
EndSection
~~~

RTSAjwad, 2020-04-03:
It seems like my problem was that I was trying to define the area of the tablet rather than the stylus and it wasn't returning an area. An xorg approach won't work for me I believe since I use Wayland. The issue is resolved though as I got it working.

Thank you for the quick and detailed response. I really appreciate it :)

jigpu, 2020-04-03:
Glad you were able to figure it out :) If you don't mind, could you share what you did so that others who use Wayland can benefit?

RTSAjwad, 2020-04-03:
Sorry. That was a bad reply from me. It seems I don't use wayland on my current distro. I got it working using your xsetwacom method. Will try the xorg method later.

jigpu, 2020-04-03:
Oops, my bad. I wrote that config file from memory and didn't actually test it. :fearful: It also breaks my system, if that's any consolation :D The following should work, however (remember to replace X0 Y0 X1 Y1):
~~~
# Restrict the tablet active area. Note that if you have multiple tablets connected
# you should add a rule like the following inside the InputClass to ensure that
# the active area only applies to the intended device.
#     MatchUSBID "XXXX:YYYY"

Section "InputClass"
	Identifier "Restrict wacom tablet area"
        MatchDriver "wacom"
        MatchIsTouchpad "false"
        MatchIsTouchscreen "false"
        Option "Area" "0 0 1000 1000"
EndSection
~~~

RTSAjwad, 2020-04-03:
Its all good :) Also it still doesn't work on my end. 

If it means anything, the file for my tablet is called "52-tablet.conf" and I'm using a UGEE M708 and not an actual wacom tablet.

jigpu, 2020-04-03:
Hmm... Weird. Should work as long as xsetwacom does, but its possible I'm just missing something obvious. I'll close the issue for now though since you seem to be happy with the xsetwacom route :+1:

closed 2020-04-03 by jigpu

whot, 2020-04-03:
ftr, the wayland solution here will be to implement this in the compositors.

out of curiosity - what is this functionality used for? just to match the right-proportional area to the screen? I **think** this may already be supported in GNOME at least, though magically hidden.

RTSAjwad, 2020-04-03:
It's just for a rhythm game called "osu!". Smaller active areas are much more comfortable in this scenario. If I'm drawing, I'll return it back to normal.