linuxwacom/xf86-input-wacom #109: S Pen: Only the first side button pressing is recognized

Aster-the-Med-Stu, 2020-04-10:
(Sequal to #97)

```libinput record``` output attached below. I have pressed side button multiple times during recording but it seems that only the first button event is recorded.

[Stylus_TEST.zip](https://github.com/linuxwacom/xf86-input-wacom/files/4460217/Stylus_TEST.zip)

And I have got a new stylus with side button and a dedicated eraser (HP Zbook x2), somehow side button pressing still shows up as eraser.

```
> sudo libinput record | grep BTN           
Available devices:
/dev/input/event0:      AT Translated Set 2 keyboard
/dev/input/event1:      Video Bus
/dev/input/event2:      Logitech Wireless Keyboard PID:4023
/dev/input/event3:      Logitech Wireless Mouse PID:4022
/dev/input/event4:      G2Touch Multi-Touch by G2TSP
/dev/input/event5:      Lid Switch
/dev/input/event6:      Power Button
/dev/input/event7:      Power Button
/dev/input/event8:      Intel Virtual Button driver
/dev/input/event9:      PC Speaker
/dev/input/event10:     HDA Intel PCH Mic
/dev/input/event11:     HDA Intel PCH Headphone
/dev/input/event12:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event13:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event14:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event15:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event16:     HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event17:     720p HD Camera: 720p HD Camera
/dev/input/event18:     720p HD Camera: Samsung IR Came
/dev/input/event19:     WCOM0033:00 2D1F:001E Mouse
/dev/input/event20:     WCOM0033:00 2D1F:001E
Select the device event number: 20
Recording to 'stdout'.
    #   Event code 320 (BTN_TOOL_PEN)
    #   Event code 321 (BTN_TOOL_RUBBER)
    #   Event code 330 (BTN_TOUCH)
    #   Event code 331 (BTN_STYLUS)
    #   Event code 332 (BTN_STYLUS2)
    #   Pressing side button 
    - [  0,      0,   1, 320,       1] # EV_KEY / BTN_TOOL_PEN              1
    - [  0, 999537,   1, 321,       1] # EV_KEY / BTN_TOOL_RUBBER           1
    - [  2, 602892,   1, 320,       0] # EV_KEY / BTN_TOOL_PEN              0
    - [  2, 602892,   1, 321,       0] # EV_KEY / BTN_TOOL_RUBBER           0
    #   Pressing eraser
    - [  5, 289077,   1, 321,       1] # EV_KEY / BTN_TOOL_RUBBER           1
    - [  5, 292313,   1, 330,       1] # EV_KEY / BTN_TOUCH                 1
    - [  5, 509074,   1, 330,       0] # EV_KEY / BTN_TOUCH                 0
    - [  5, 578334,   1, 321,       0] # EV_KEY / BTN_TOOL_RUBBER           0
```

Aster-the-Med-Stu, 2020-04-10:
And I believe the on-board Wacom chipset is able to distinguish between Zbook X2 Stylus's side button and eraser. Samsung has a tool called Air Command, and it would show a pop-up menu when side button is pressed. It never appears when using Zbook X2's stylus on the back.

More device info under Windows:
[image]

jigpu, 2020-04-10:
Can you confirm that [this](https://www8.hp.com/us/en/workstations/zbook-x2.html) is your device, and that [this](https://store.hp.com/us/en/pdp/hp-zbook-x2-pen) is your pen?

I find it interesting that the Zbook page mentions that it uses EMR technology and yet you see the pen button act as an eraser. Some EMR tech encodes the eraser identically to a second button but I don't think I've ever seen a one-button EMR pen trigger an eraser event.

Please use our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh) to record your use of the pen. Once you have selected the stylus device and the script indicates it is recording, bring the pen into prox, press the button three times, and then remove the pen from prox. Next, turn the pen over and bring the eraser into prox and then back out of prox. Press CTRL+C to stop recording and attach the generated tarball to this issue.

jigpu, 2021-07-21:
Closing due to inactivity.

closed 2021-07-21 by jigpu