linuxwacom/xf86-input-wacom #110: Intuos 4 wheel not scrolling

penny-bit, 2020-04-11:
I'm trying to setup the wheel of my tablet in order to scroll up and down like a mouse scrolling wheel. In the Gnome GUI settings it doesn't let me configure the action only keys. I tried with xsetwacom but I don't know what to put as mapping value in order to scroll.

This is my output of xsetwacom:

```
Property 'Wacom Tablet Area' does not exist on device.
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "1" "button +1 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "2" "button +2 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "3" "button +3 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "8" "button +8 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "9" "button +9 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "10" "button +10 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "11" "button +11 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "12" "button +12 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Button" "13" "button +13 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "ToolDebugLevel" "0"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "TabletDebugLevel" "0"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Suppress" "2"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "RawSample" "4"
Property 'Wacom Pressurecurve' does not exist on device.
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Mode" "Absolute"
Property 'Wacom Hover Click' does not exist on device.
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Gesture" "off"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "ZoomDistance" "0"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "ScrollDistance" "0"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Property 'Wacom Rotation' does not exist on device.
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "RelWheelUp" "1" "button +5 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "RelWheelDown" "2" "button +4 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "AbsWheelUp" "3" "button +4 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "AbsWheelDown" "4" "button +5 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "AbsWheel2Up" "5" "button +4 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "AbsWheel2Down" "6" "button +5 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "StripLeftUp" "1" "button +4 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "StripLeftDown" "2" "button +5 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "StripRightUp" "3" "button +4 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "StripRightDown" "4" "button +5 "
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "Threshold" "0"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" "BindToSerial" "0"
Property 'Wacom Pressure Recalibration' does not exist on device.
```

As far as I understood it is something to do with `gnome-settings-daemon`, all the parameter above are "linked" with the Gnome GUI setting, because if I modify the GUI it doesn't reflect on the property list above.

As you explain well on your wiki I would like to create a `xsetwacom` script with my buttons and scroll wheel settings in it. This is what i was trying to do:
```
...
xsetwacom set "Wacom Intuos4 6x9 Pad pad" AbsWheelDown "key shift plus"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" Button 1 "key 1"  # button inside touchring
xsetwacom set "Wacom Intuos4 6x9 Pad pad" AbsWheelUp "key minus"
...
```
but it doesn't work. And at the end, I don't know which way to go: `gnome-settings-daemon` or `xsetwacom` script.
Please help me! 
Thanks!

jigpu, 2020-04-14:
Short version:

The only real solution is to have GNOME integrate support for this feature. You can add a "thumb up" to https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/118

Since GNOME intercepts "real" scroll events from the pad, the closest thing you can do is use xsetwacom to configure the ring to send keyboard arrow events instead (though this is obviously not quite the same thing and won't always work as intended):

~~~
xsetwacom set "Wacom Intuos4 6x9 Pad pad" AbsWheelDown "key down"
xsetwacom set "Wacom Intuos4 6x9 Pad pad" AbsWheelUp "key up"
~~~

***
Long version:

Unfortunately, the mechanism that GNOME uses to "link" the GUI with our driver somewhat prevents you making the override you'd like.

GNOME intercepts the button events sent by our driver, and then creates and sends its own event according to what is specified in the control panel. If you had the third ExpressKey mapped to the key "A", our driver would send `button +3` as described by xsetwacom, and then GNOME would see that event from the device and create a keypress/release for the "A" key.

If you wanted to program the first ExpressKey to perform an action supported only by xsetwacom, you could just use xsetwacom to change the `button +3` default to something else. GNOME would ignore the event and it would get sent to the rest of the OS.

The problem is that the touchring sends scroll up/down events by default (button +4 and button +5). GNOME intercepts them and will send whatever you've programmed them to instead. You can't tell GNOME that you _want_ scroll events at the moment, however, and you also can't use xsetwacom to have the ring send scroll events since those are exactly the events that GNOME is already intercepting. The only thing you can do is send events that are similar to scroll events, e.g. arrow events or page up/down events.

> As you explain well on your wiki I would like to create a xsetwacom script with my buttons and scroll wheel settings in it. This is what i was trying to do:
>
> ~~~
> xsetwacom set "Wacom Intuos4 6x9 Pad [...]

Aside from these not being scroll events, they should work and cause the ring to send "+" and "-" keyboard events. Does this not happen for you?

closed 2020-04-14 by jigpu

penny-bit, 2020-04-15:
I would go with the short one! :-)
So the problem is gnome, that dosn't recognize the event of scrolling, right? Do you think it could work using older version of gnome?
The only way I could set `xsetwacom` to make it scroll was to map `key up/down`, as you said, but it's not a real scroll event.
Thank you for your clear answer.

jigpu, 2020-04-15:
Older versions of GNOME would allow you to disable the gsdwacom plugin which was one possible workaround. This workaround is described in the bug report I linked to. I believe it is only possible up through version 3.22.

penny-bit, 2020-05-18:
It might be a silly question and I'm sorry for that, I'm not a programmer. I was wondering: It would be possible to hack KDE wacomtablet (https://github.com/KDE/wacomtablet) which is the KDE wacom control panel module to work on GNOME, so in this way I can set the tablet with KDE control panel and keep the rest of the controls on GNOME?

jigpu, 2020-05-18:
I'm not really sure. I'm not up to date on how the KDE control panel works, but in the past I believe it needed to connect to other KDE components to work. You might be able to get the control panel to load and apply some settings, but my gut says probably not.