linuxwacom/xf86-input-wacom #112: Intuos pro S (PTH-460) touch ring center button does nothing

RandomReaper, 2020-04-20:
The center button generates no event (tested with xev).
xserver-xorg-input-wacom 1:0.39.0-0ubuntu1

jigpu, 2020-04-20:
I assume you're using the GNOME desktop, which is consuming these events before xev sees them. You should be able to see them if you use something like `xinput test-xi2 <pad_id>` or run xev under a non-GNOME desktop.

RandomReaper, 2020-04-21:
Thank you for your response, I'm using xfce here. Here is some more information:

If I understand well, it maybe a problem with `ds.buttons` with a value of `0` for the center button.

Output of `xinput test-xi2 <my-pad-id>` for the first button press+release : 
```EVENT type 15 (RawButtonPress)
    device: 15 (15)
    detail: 1
    flags: 
    valuators:
          3: 0.00 (0.00)
          4: 0.00 (0.00)
          5: 0.00 (0.00)

EVENT type 17 (RawMotion)
    device: 15 (15)
    detail: 0
    flags: 
    valuators:
          3: 0.00 (0.00)
          4: 0.00 (0.00)
          5: 0.00 (0.00)

EVENT type 12 (PropertyEvent)
     property: 526 'Wacom Serial IDs'
     changed: modified
EVENT type 16 (RawButtonRelease)
    device: 15 (15)
    detail: 1
    flags: 
    valuators:
          3: 0.00 (0.00)
          4: 0.00 (0.00)
          5: 0.00 (0.00)

EVENT type 12 (PropertyEvent)
     property: 526 'Wacom Serial IDs'
     changed: modified
```

For the central button press+release: 
```
EVENT type 12 (PropertyEvent)
     property: 526 'Wacom Serial IDs'
     changed: modified
EVENT type 12 (PropertyEvent)
     property: 526 'Wacom Serial IDs'
     changed: modified
```
Debug info from `/var/log/Xorg.0.log`, first button press+release:
```
[  3335.960] (II) /dev/input/event23 (10:wcmReadPacket): fd=85
[  3335.960] (II) /dev/input/event23 (1:wcmReadPacket): pos=0 remaining=256
[  3335.961] (II) /dev/input/event23 (10:wcmReadPacket): buffer has 72 bytes
[  3335.961] (II) /dev/input/event23 (10:usbParseEvent): 
[  3335.961] (II) /dev/input/event23 (10:usbParseEvent): 
[  3335.961] (II) /dev/input/event23 (10:usbParseEvent): 
[  3335.961] (II) /dev/input/event23 (6:usbDispatchEvents): 3 events received
[  3335.961] (II) /dev/input/event23 (10:usbDispatchEvents): Dirty flag set on channel 17; sending event.
[  3335.961] (II) /dev/input/event23 (10:wcmEvent): channel = 17
[  3335.961] (II) /dev/input/event23 (10:wcmEvent): c=17 i=15 t=16 s=4294967295 x=0 y=0 b=1 p=0 rz=0 tx=0 ty=0 aw=0 aw2=0 rw=0 t=0 px=1 st=0 cs=4 
[  3335.961] (II) /dev/input/event23 (10:commonDispatchDevice): device type = 16
[  3335.961] (II) /dev/input/event23 (10:wcmCheckSuppress): level = 2 return value = 8
[  3336.050] (II) /dev/input/event23 (10:wcmReadPacket): fd=85
[  3336.050] (II) /dev/input/event23 (1:wcmReadPacket): pos=0 remaining=256
[  3336.050] (II) /dev/input/event23 (10:wcmReadPacket): buffer has 72 bytes
[  3336.050] (II) /dev/input/event23 (10:usbParseEvent): 
[  3336.051] (II) /dev/input/event23 (10:usbParseEvent): 
[  3336.051] (II) /dev/input/event23 (10:usbParseEvent): 
[  3336.051] (II) /dev/input/event23 (6:usbDispatchEvents): 3 events received
[  3336.051] (II) /dev/input/event23 (10:usbDispatchEvents): Dirty flag set on channel 17; sending event.
[  3336.051] (II) /dev/input/event23 (10:wcmEvent): channel = 17
[  3336.051] (II) /dev/input/event23 (10:wcmEvent): c=17 i=15 t=16 s=4294967295 x=0 y=0 b=0 p=0 rz=0 tx=0 ty=0 aw=0 aw2=0 rw=0 t=0 px=0 st=0 cs=4 
[  3336.051] (II) /dev/input/event23 (10:commonDispatchDevice): device type = 16
[  3336.051] (II) /dev/input/event23 (10:wcmCheckSuppress): level = 2 return value = 8
```

Center button press+release:
```[  3456.432] (II) /dev/input/event23 (10:wcmReadPacket): fd=85
[  3456.432] (II) /dev/input/event23 (1:wcmReadPacket): pos=0 remaining=256
[  3456.432] (II) /dev/input/event23 (10:wcmReadPacket): buffer has 48 bytes
[  3456.432] (II) /dev/input/event23 (10:usbParseEvent): 
[  3456.432] (II) /dev/input/event23 (10:usbParseEvent): 
[  3456.433] (II) /dev/input/event23 (6:usbDispatchEvents): 2 events received
[  3456.433] (II) /dev/input/event23 (10:usbDispatchEvents): Dirty flag set on channel 17; sending event.
[  3456.433] (II) /dev/input/event23 (10:wcmEvent): channel = 17
[  3456.433] (II) /dev/input/event23 (10:wcmEvent): c=17 i=15 t=16 s=4294967295 x=0 y=0 b=0 p=0 rz=0 tx=0 ty=0 aw=0 aw2=0 rw=0 t=0 px=1 st=0 cs=4 
[  3456.433] (II) /dev/input/event23 (10:commonDispatchDevice): device type = 16
[  3456.433] (II) /dev/input/event23 (10:wcmCheckSuppress): level = 2 return value = 8
[  3456.482] (II) /dev/input/event23 (10:wcmReadPacket): fd=85
[  3456.482] (II) /dev/input/event23 (1:wcmReadPacket): pos=0 remaining=256
[  3456.482] (II) /dev/input/event23 (10:wcmReadPacket): buffer has 48 bytes
[  3456.482] (II) /dev/input/event23 (10:usbParseEvent): 
[  3456.482] (II) /dev/input/event23 (10:usbParseEvent): 
[  3456.482] (II) /dev/input/event23 (6:usbDispatchEvents): 2 events received
[  3456.482] (II) /dev/input/event23 (10:usbDispatchEvents): Dirty flag set on channel 17; sending event.
[  3456.482] (II) /dev/input/event23 (10:wcmEvent): channel = 17
[  3456.482] (II) /dev/input/event23 (10:wcmEvent): c=17 i=15 t=16 s=4294967295 x=0 y=0 b=0 p=0 rz=0 tx=0 ty=0 aw=0 aw2=0 rw=0 t=0 px=0 st=0 cs=4 
[  3456.482] (II) /dev/input/event23 (10:commonDispatchDevice): device type = 16
[  3456.482] (II) /dev/input/event23 (10:wcmCheckSuppress): level = 2 return value = 8
```