linuxwacom/xf86-input-wacom #113: how can I map tablet to dual display as a single screen?

braindevices, 2020-04-21:
This was always my setting before 2019/10. After I upgrade the system, I can only map it to specify display. But I want to use my tablet on both display as they combined into one. All the wiki papers is about how to map the wacom tablet onto specific one. I actually want to do it opposite. Can anyone help me on this?

jigpu, 2020-04-21:
Can you please provide some additional details?
* Tablet model
* Distribution & version
* Desktop environment & version
* Driver version (`xsetwacom --version`)
* Describe the upgrade you performed (e.g was this just a run-of-the-mill software update or an upgrade to a new distribution version?)
* Describe what steps you are taking to try and map the tablet to the entire display

braindevices, 2020-04-23:
ubuntu 19.10, GNOME 3.34.2, I am using ubuntu default version not the classic version.
driver:
```
$ xsetwacom --version
0.36.1
```
tablet:
```
$ xinput list-props "Wacom Intuos3 6x8 Pad pad"
Device 'Wacom Intuos3 6x8 Pad pad':
        Device Enabled (169):   1
        Coordinate Transformation Matrix (171): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (301):     0
        Device Accel Constant Deceleration (302):       1.000000
        Device Accel Adaptive Deceleration (303):       1.000000
        Device Accel Velocity Scaling (304):    10.000000
        Device Node (293):      "/dev/input/event23"
        Wacom Serial IDs (333): 177, 0, 15, 0, 0
        Wacom Serial ID binding (334):  0
        Wacom Pressure Threshold (335): 0
        Wacom Sample and Suppress (336):        2, 4
        Wacom Enable Touch (337):       0
        Wacom Enable Touch Gesture (339):       0
        Wacom Touch Gesture Parameters (340):   0, 0, 250
        Wacom Tool Type (341):  "PAD" (351)
        Wacom Button Actions (342):     "Wacom button action 0" (343), "Wacom button action 1" (344), "Wacom button action 2" (345), "None" (0), "None" (0), "None" (0), "None" (0), "Wacom button action 3" (346), "Wacom button action 4" (347), "Wacom button action 5" (352), "Wacom button action 6" (353), "Wacom button action 7" (354)
        Wacom button action 0 (343):    1572865
        Wacom button action 1 (344):    1572866
        Wacom button action 2 (345):    1572867
        Wacom button action 3 (346):    1572872
        Wacom button action 4 (347):    1572873
        Wacom button action 5 (352):    1572874
        Wacom button action 6 (353):    1572875
        Wacom button action 7 (354):    1572876
        Wacom Strip Buttons (355):      "Wacom strip action 0" (356), "Wacom strip action 1" (357), "Wacom strip action 2" (358), "Wacom strip action 3" (359)
        Wacom strip action 0 (356):     1572868
        Wacom strip action 1 (357):     1572869
        Wacom strip action 2 (358):     1572868
        Wacom strip action 3 (359):     1572869
        Wacom Wheel Buttons (360):      "Wacom wheel action 0" (361), "Wacom wheel action 1" (362), "Wacom wheel action 2" (363), "Wacom wheel action 3" (364), "Wacom wheel action 4" (365), "Wacom wheel action 5" (366)
        Wacom wheel action 0 (361):     1572869
        Wacom wheel action 1 (362):     1572868
        Wacom wheel action 2 (363):     1572868
        Wacom wheel action 3 (364):     1572869
        Wacom wheel action 4 (365):     1572868
        Wacom wheel action 5 (366):     1572869
        Wacom Panscroll Threshold (349):        2600
        Device Product ID (294):        1386, 177
        Wacom Debug Levels (350):       0, 0
```
Before upgrade form ubuntu 18.04 to 19.10, I did not need to do anything special to map tablet to my dual displays. By default, it was using the entire combined screen. After system upgrade, the cursor only appear on my primary screen. I went into the system setting->wacom->Tablet panel, It said Tablet(absolute). I click Map to Monitor... `Map to single monitor` is unchecked. So it should just work.
I can check that box, then select it to may to either monitor. The transform matrix of Xinput output never changes.

I then use relative mode, which allows the tablet map to both dual displays. Then I switch it back to absolute mode. The tablet then successfully map to both displays. Still the transform matrix never changed. But after sleep-wakeup or logout-login or disconnect-reconnect usb, it maps back to primary display only again.

I haven't try any tricks from the wiki, because all of them are about taking effort to map it to single display.

jigpu, 2020-04-23:
Thanks, that helps quite a bit. Ubuntu 18.04 and 19.10 use the same driver version, so this is most likely a bug in GNOME rather than our driver. I don't recall seeing this issue with GNOME 3.34 in the past, but I can try to reproduce this in Ubuntu 19.10 in case the issue is specific to that distro.

I notice that you ran `xintput list-props` on the "pad" device. Is this what you did to see if the transform matrix was changing as you modified the settings in GNOME? The "pad" device doesn't control the pointer at all, so I doubt GNOME would modify its matrix even under normal conditions. I would run `xinput list-props` on the stylus device instead to see how the matrix changes.

braindevices, 2020-04-24:
@jigpu Thanks a lot! Yes you are correct. The stylus matrix actually changed. I end up write a script to reset the matrix at startup.

closed 2020-04-24 by braindevices