linuxwacom/xf86-input-wacom #116: Only able to access one screen at a time.

coreyryanhanson, 2020-05-09:
I am running Centos 8 and I have been unable to map my Wacom Intuos Pro to both monitors simultaneously.  I'm not sure why, but I'm forced to toggle between them entirely on an individual basis which has never happened on my older Centos 7 system. I've included some relevant hardware info, please let me know if I should add anything else.

xorg-x11-drv-wacom version: 0.36.1
NVIDIA driver version-  440.82 (installed via EL-repo)
The GPU is a GeForce RTX 2080

and the two displays are
AU Optronics Corporation (DP-2), - 1920 x 1080 res
Ancor Communications Inc PA248 (HDMI-0) 1920 x 1200 res

jigpu, 2020-05-13:
A few questions:
* Are you running CentOS 8.0 or 8.1 according to `cat /etc/redhat-release`?
* Are you running "CentOS Stream"?
* Are you running `xsetwacom set <id> MapToOutput next` to toggle displays, or another mechanism (and which)?

coreyryanhanson, 2020-05-13:
Actually, I just tried enabling CentOS Stream and that did the trick. Thanks!!!!!!!

closed 2020-05-13 by coreyryanhanson