linuxwacom/xf86-input-wacom #117: Intuos 4 PTK-540WL keeps producing events with pen rock steady

PrzemoF, 2020-05-09:
I hope it's the right place to file the bug, but it might be related to the kernel driver. My PTK-540WL (Intuos 4) produces events even with the pen rock steady, fixed in place by a book. It happens over USB and over bluetooth connection. Normally it's impossible to notice (gimp/krita works fine), but I have problem with FreeCAD - certain functions do recalculation when cursor is moved and flooding on the input makes them unusable. Video showing the problem during xev test: https://peertube.social/videos/watch/1dbfe959-6ab5-417a-a91d-9f669173df00
Info about my system:
Up to date linux fedora 32, gnome in xorg mode
`$ uname -a 
Linux localhost.localdomain 5.6.7-300.fc32.x86_64 #1 SMP Thu Apr 23 14:13:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
rpm -qi libwacom
Name        : libwacom
Version     : 1.3
Release     : 1.fc32
Architecture: x86_64
[..]
$ libwacom-list-local-devices
# Device node: /dev/input/event9
[Device]
Name=Wacom Intuos4 WL
ModelName=PTK-540WL
DeviceMatch=usb:056a:00bc;bluetooth:056a:00bd;
Class=Intuos4
Width=8
Height=5
IntegratedIn=
Layout=intuos4-6x9-wl.svg
Styli=0x802;0x804;0x806;0x80a;0x80c;0x902;0x90a;0x100802;0x100804;0x10080a;0x10080c;0x100902;0x10090a;0x120802;0x140802;0x14080a;0x160802;0x16080a;

[Features]
Reversible=true
Stylus=true
Ring=true
Ring2=false
Touch=false
TouchSwitch=false
StatusLEDs=Ring;
NumStrips=0
Buttons=9

[Buttons]
Left=A;B;C;D;E;F;G;H;I;
# Right=
# Top=
# Bottom=
# Touchstrip=
# Touchstrip2=
OLEDs=B;C;D;E;F;G;H;I;
Ring=A;
# Ring2=
EvdevCodes=0x100;0x101;0x102;0x103;0x104;0x105;0x106;0x107;0x108;
RingNumModes=4
Ring2NumModes=0
StripsNumModes=0

---------------------------------------------------------------
# Device node: /dev/input/event8
[Device]
Name=Wacom Intuos4 WL
ModelName=PTK-540WL
DeviceMatch=usb:056a:00bc;bluetooth:056a:00bd;
Class=Intuos4
Width=8
Height=5
IntegratedIn=
Layout=intuos4-6x9-wl.svg
Styli=0x802;0x804;0x806;0x80a;0x80c;0x902;0x90a;0x100802;0x100804;0x10080a;0x10080c;0x100902;0x10090a;0x120802;0x140802;0x14080a;0x160802;0x16080a;

[Features]
Reversible=true
Stylus=true
Ring=true
Ring2=false
Touch=false
TouchSwitch=false
StatusLEDs=Ring;
NumStrips=0
Buttons=9

[Buttons]
Left=A;B;C;D;E;F;G;H;I;
# Right=
# Top=
# Bottom=
# Touchstrip=
# Touchstrip2=
OLEDs=B;C;D;E;F;G;H;I;
Ring=A;
# Ring2=
EvdevCodes=0x100;0x101;0x102;0x103;0x104;0x105;0x106;0x107;0x108;
RingNumModes=4
Ring2NumModes=0
StripsNumModes=0

---------------------------------------------------------------
`
Also occasionally when I move pen between windows "paste" event happens. It's probably not related the the error above, but it's something new on my system after fedora 31-->32 update

PrzemoF, 2020-05-09:
```
/usr/bin/xsetwacom --get "Wacom Intuos4 WL Pen stylus" Suppress
2
```

PrzemoF, 2020-05-09:
Setting Suppress to 4 makes the situation much better with occasional event only

closed 2020-05-09 by PrzemoF

jigpu, 2020-05-13:
Increasing the suppress value is exactly what I would have suggested. It basically sets the minimum threshold of movement before the driver will bother to send any kind of event. Given that the Intuos4 has a resolution of 200 points/mm, a threshold of 4 is movement (or noise :D) on the order of 20 microns.