linuxwacom/xf86-input-wacom #118: Mapping bug with Ubuntu 20.04

lizliz, 2020-05-10:
Hi, I cannot seem to figure out how to fix the input on my wacom tablet after the upgrade to Ubuntu 20.04. I have a Wacom Intuos BT S pen input which was working perfectly fine before the upgrade, but the wacom driver is now buggy to the point of unuseable (specifics below), does anyone have any suggestions on how to manually control screen mapping? I have 3 monitors: two horizontal, and one rotated to be vertical, and not quite lined up on the bottom. I want the tablet to be just mapped to the left monitor, which is HDMI-1.

If I'm interpreting xrandr correctly, the virtual coordinates are: 
```
     0               1920              3000
   0                   A----------------|
                       |                |
                       |                |
                       |                |
667  B-----------------|   DP-3         |------------------|
     |                 |                |                  |
     |  HDMI-1         |    1080x1920   |   DVI-I-1        |
     | 1920x1080       |                |   1920x1080      |
     ------------------|                |------------------|
                       |                |
                       |----------------|
```

- First question, there used to be something in the settings to map the top buttons to actions (in particular, I liked setting one button to switch between screens for the mapping), but that doesn't appear to be included anymore, will that come back?
- Second main question, there is something massively wrong with the automatic single monitor mapping right now. For both of my regular screens, the mapping only uses the left half of my wacom tablet. It registers the left half, but the entire right half gets mapped to the right edge of the screen. However, at a minimum I really just need it mapped to the leftmost screen (although I do miss having the button on the tablet do automatic switching!)
- Potentially useful and almost definitely another bug, for the vertical monitor, it maps the tablet rotated, and again, only seems to work with the left side of the wacom tablet.

For context my output settings after running `xsetwacom set 15 MapToOutput HDMI-1` are: 

```
(base) liz@Gandalf:~$ xinput list-props 15
Device 'Wacom Co.,Ltd. Intuos BT S stylus':
	Device Enabled (152):	1
	Coordinate Transformation Matrix (154):	0.390244, 0.000000, 0.000000, 0.000000, 0.562500, 0.347396, 0.000000, 0.000000, 1.000000
	Device Accel Profile (283):	0
	Device Accel Constant Deceleration (284):	1.000000
	Device Accel Adaptive Deceleration (285):	1.000000
	Device Accel Velocity Scaling (286):	10.000000
	Device Node (275):	"/dev/input/event19"
	Wacom Tablet Area (679):	0, 0, 15200, 9500
	Wacom Rotation (680):	0
	Wacom Pressurecurve (681):	0, 0, 100, 100
	Wacom Serial IDs (401):	886, 1, 2, 0, 0
	Wacom Serial ID binding (652):	0
	Wacom Proximity Threshold (682):	30
	Wacom Pressure Threshold (653):	26
	Wacom Sample and Suppress (654):	2, 4
	Wacom Enable Touch (655):	0
	Wacom Hover Click (683):	1
	Wacom Enable Touch Gesture (656):	0
	Wacom Touch Gesture Parameters (657):	0, 0, 250
	Wacom Tool Type (658):	"STYLUS" (678)
	Wacom Button Actions (659):	"Wacom button action 0" (660), "Wacom button action 1" (661), "Wacom button action 2" (662), "None" (0), "None" (0), "None" (0), "None" (0), "Wacom button action 3" (663)
	Wacom button action 0 (660):	1572865
	Wacom button action 1 (661):	1572866
	Wacom button action 2 (662):	1572867
	Wacom button action 3 (663):	1572872
	Wacom Pressure Recalibration (685):	1
	Wacom Panscroll Threshold (676):	1300
	Device Product ID (276):	11551, 886
	Wacom Debug Levels (677):	0, 0
```
and my output of `xrandr`
```
(base) liz@Gandalf:~$ xrandr
Screen 0: minimum 320 x 200, current 4920 x 1920, maximum 16384 x 16384
DVI-I-1 connected 1920x1080+3000+667 (normal left inverted right x axis y axis) 474mm x 296mm
   1920x1080     61.09*+  60.00  
   1680x1050     59.88  
   1680x945      60.02  
   1400x1050     74.87    59.95  
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1440x900      74.98    59.90  
   1280x960      60.00  
[... 98 lines cut ...]
   800x512       60.17  
   700x525       74.76    70.06    59.98  
   800x450       59.95    59.82  
   720x480       60.00    59.94  
   640x512       75.02    60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    75.00    72.81    75.00    66.67    60.00    59.94  
   720x405       59.51    58.99  
   720x400       70.08  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       75.00    75.00    70.00    60.06  
   640x360       59.86    59.83    59.84    59.32  
   512x384       75.03    70.07    60.00  
   512x288       60.00    59.92  
   416x312       74.66  
   480x270       59.63    59.82  
   400x300       72.19    75.12    60.32    56.34  
   432x243       59.92    59.57  
   320x240       72.81    75.00    60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
```

whot, 2020-05-11:
fwiw, the bit that does the mapping is this:

> 	Coordinate Transformation Matrix (154):	0.390244, 0.000000, 0.000000, 0.000000, 0.562500, 0.347396, 0.000000, 0.000000, 1.000000

it's a 3x3 transformation matrix, the [arch wiki](https://wiki.archlinux.org/index.php/Calibrating_Touchscreen) has long explanation of how it works. This is supposed usually set by GNOME (or your desktop environment), the driver itself has absolutely no say in it. the property is provided by the server and driver-independent,  `xsetwacom` just changes the property. `xinput map-to-output` also does the same thing here.

I've paged most of this out, but IIRC the above means: 39% width, 56% height at an offset of 34%. Those numbers seem to be correct given the drawing you have.

As I said, the driver doesn't know anything about screen mappings. The candidates are the X server but that code hasn't changed in years, or GNOME if it applies the wrong matrix. I don't think that code has changed recently either though, unless Ubuntu ships custom patches here.

lizliz, 2020-05-11:
Thanks for responding!

Yes, I doubled checked the matrix last night too and I'm convinced it's right. The weirdest part of this bug is that the right half of the tablet only seems to register the y coordinate when on my desktop. When I connect the Wacom to my laptop (so single monitor) everything is fine, so I don't think theres anything wrong with the tablet itself. And the problem showed up when I upgraded, so there must be some change in the OS that I don't understand. 

Any guesses on where else I can look for what might be causing this bug?

whot, 2020-05-12:
any chance you can start with a single screen, then add the next one, etc. until the issue occurs? my guess would be it's either the three-screens (not exactly common) or the rotated one in the middle that may cause the issue.

In the end though where the matrix is numerically correct but the behaviour is wrong, the issue is in the X server. look for `fill_pointer_events()` in `dix/getevents.c` and once you've stopped screaming you can start debugging :) It's not pretty code, but happy to explain things.

Also: nvidia driver? if so, that's usually the cause :)

jigpu, 2020-05-13:
> Yes, I doubled checked the matrix last night too and I'm convinced it's right. The weirdest part of this bug is that the right half of the tablet only seems to register the y coordinate when on my [...]

Sounds like another case of your tablet model automatically switching into "Android mode". See [this page](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect) for details and a workaround. Some software (particularly fwupd) can cause the tablet to mis-identify a desktop Linux machines as Android, and its very possible that your upgrade to 20.04 brought in such software.

As for your other questions:

> * First question, there used to be something in the settings to map the top buttons to actions (in particular, I liked setting one button to switch between screens for the mapping), but that doesn't appear to be included anymore, will that come back?

Both the "Map Buttons" and "Switch Monitor" feature should still be there. I would try to get more support at the GNOME control center's issue tracker: https://gitlab.gnome.org/GNOME/gnome-control-center/issues. 

> * Potentially useful and almost definitely another bug, for the vertical monitor, it maps the tablet rotated, and again, only seems to work with the left side of the wacom tablet.

Definitely a second issue to create at https://gitlab.gnome.org/GNOME/gnome-control-center/issues. GNOME is probably rotating the sensor because that's normally the right thing to do when the sensor is embedded into a display. It doesn't necessarily make sense for a peripheral tablet, though it does have the advantage of making maximal use of the tablet area instead of just half the tablet.

lizliz, 2020-05-14:
Holy obnoxious bugs, batman, it worked! 

The issue was the android vs desktop thing, followed the instruction [here](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect) from jigpu.  The answer is to just hold down the left-most and right-most buttons for 3-4 seconds until the light turns on and off again. 

Ah, I am so happy! Thank you all for your help!

jigpu, 2020-05-14:
Glad that did the trick :) Wish it was possible to create a proper solution, but I don't know that it is...

closed 2020-05-14 by jigpu