linuxwacom/xf86-input-wacom #123: Undetected Button 3 on Wacom Artz II (UD-0608-R) Serial Tablet

davidpgil, 2020-05-24:
For some reason, Button 3 (the top half of the stylus button) does not seem to register at all and cannot be assigned to anything. I've tried the --shell syntax:
`xsetwacom set "Wacom Digitizer II stylus" "Button" "3" "button +3 "`

and the non-shell syntax:
`xsetwacom set "Wacom Digitizer II stylus" Button 3 3`

I am using Trisquel 9 (Ubuntu 18.04 based).

I used "inputattach --wacom_iv" to connect my tablet. I have confirmed that all stylus buttons work in WIndows, so this is not an issue where the button is broken.

However, I am able to set Button 2 to 3 to create a right-click, so I know I am configuring fine.

Any idea what may be going on?

whot, 2020-05-24:
Check with `libinput record` what key code the button sends but it's most likely that the button is the eraser toggle button. In which case you can't map it, we don't have support for remapping that button yet (it's rather complicated to implement).

davidpgil, 2020-05-25:
My tablet, as I am using it, does not show up in `libinput list-devices` ... it seems libinput cannot see this tablet. If I use xsetwacom list, I see:

```
Wacom Digitizer II stylus       	id: 23	type: STYLUS    
Wacom Digitizer II eraser       	id: 24	type: ERASER    
Wacom Digitizer II cursor       	id: 25	type: CURSOR 
```

`libinput record` command doesn't seem to exist.

whot, 2020-05-25:
install `libinput-utils` or some similarly named package from your distroy, that should include it.

davidpgil, 2020-05-25:
I just installed `libinput*` and still `libinput record` does not exist.

whot, 2020-05-26:
oh, 1.10.4. yeah, in that case it's not included yet, it was first added in 1.11 or 1.12. `evemu-record` should be available on your distro which is close enough for this.

davidpgil, 2020-05-26:
I ran `evemu-record` and I do not see any "PRESSURE" being registered when I press button 3. When I press the the stylus and when I press button 2 and the eraser, all of them register "PRESSURE". Now what?

whot, 2020-05-27:
I'll need the `evemu-record` logs attached here from a button press, anything else will just be guesswork.

davidpgil, 2020-05-27:
The following is an `evamu-record` log of myself pressing button 3 on my tablets stylus four times:

```
# EVEMU 1.3
# Kernel: 4.19.120-gnu-rt52
# DMI: dmi:bvnLENOVO:bvrGNET72WW(2.20):bd02/26/2015:svn<<REDACTED>>:pn20EF000HUS:pvr<<REDACTED>>:rvnLENOVO:rn20EF000HUS:rvrSDK0E50510WIN:cvn<<REDACTED>>:ct10:cvrNotAvailable:
# Input device name: "Wacom Digitizer II"
# Input device ID: bus 0x13 vendor 0x3e product 0000 version 0x5544
# Size in mm: 8x6
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 272 (BTN_LEFT)
#     Event code 273 (BTN_RIGHT)
#     Event code 274 (BTN_MIDDLE)
#     Event code 320 (BTN_TOOL_PEN)
#     Event code 321 (BTN_TOOL_RUBBER)
#     Event code 326 (BTN_TOOL_MOUSE)
#     Event code 330 (BTN_TOUCH)
#     Event code 331 (BTN_STYLUS)
#   Event type 2 (EV_REL)
#   Event type 3 (EV_ABS)
#     Event code 0 (ABS_X)
#       Value     4054
#       Min          0
#       Max      10240
#       Fuzz         0
#       Flat         0
#       Resolution 1270
#     Event code 1 (ABS_Y)
#       Value     2335
#       Min          0
#       Max       7680
#       Fuzz         0
#       Flat         0
#       Resolution 1270
#     Event code 24 (ABS_PRESSURE)
#       Value        0
#       Min         -1
#       Max        255
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 40 (ABS_MISC)
#       Value        2
#       Min          0
#       Max          0
#       Fuzz         0
#       Flat         0
#       Resolution   0
# Properties:
#   Property  type 0 (INPUT_PROP_POINTER)
#   Property  type 1 (INPUT_PROP_DIRECT)
N: Wacom Digitizer II
[... 1174 lines cut ...]
E: 5.503887 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 5.551843 0003 0000 4581	# EV_ABS / ABS_X                4581
E: 5.551843 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +48ms
E: 5.583880 0003 0001 3012	# EV_ABS / ABS_Y                3012
E: 5.583880 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +32ms
E: 5.599889 0003 0001 3013	# EV_ABS / ABS_Y                3013
E: 5.599889 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 5.615825 0003 0000 4580	# EV_ABS / ABS_X                4580
E: 5.615825 0003 0001 3014	# EV_ABS / ABS_Y                3014
E: 5.615825 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 5.615833 0003 0001 3015	# EV_ABS / ABS_Y                3015
E: 5.615833 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 5.631880 0003 0000 4579	# EV_ABS / ABS_X                4579
E: 5.631880 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 5.631887 0003 0000 4578	# EV_ABS / ABS_X                4578
E: 5.631887 0003 0001 3017	# EV_ABS / ABS_Y                3017
E: 5.631887 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 5.647885 0003 0001 3019	# EV_ABS / ABS_Y                3019
E: 5.647885 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 5.647892 0003 0000 4577	# EV_ABS / ABS_X                4577
E: 5.647892 0003 0001 3020	# EV_ABS / ABS_Y                3020
E: 5.647892 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 5.647893 0003 0001 3021	# EV_ABS / ABS_Y                3021
E: 5.647893 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
```

whot, 2020-05-28:
`E: 1.935906 0001 0141 0001	# EV_KEY / BTN_TOOL_RUBBER      1`

That's the key here. It's the usual button that switches the tablet to eraser mode. Since this is the same-ish sequence as if you turn the stylus around to use the eraser **end** on traditional tablets, we cannot handle this as button without special code - which we don't have yet. See https://github.com/linuxwacom/libwacom/pull/179 for one of the requirements to get there, eventually.

See also [this post](https://who-t.blogspot.com/2014/09/stylus-behaviour-on-microsoft-surface-3.html) from 2014.

Meanwhile, I'm closing this bug. This isn't device-specific and it'll work when/if we manage to get this functionality implemented. Until then it cannot work.

closed 2020-05-28 by whot

davidpgil, 2020-05-28:
@whot for clarity, I am posting a picture of the button I am pressing. the tip of my finger is touch the exact part of the pen I am touching. Are you saying, there is no way to remap this part of the stylus button?
[image]

whot, 2020-05-29:
correct, that button is hardwired to change the whole pen into an eraser type instead. To make this appear like a regular button, xf86-input-wacom would have to know this button is an eraser button (see the libwacom MR linked above) and then intercept the tool type switch and map it into a virtual button instead - that button could then be mapped. This is nontrivial and hasn't been done yet.

davidpgil, 2020-05-29:
The ArtZ II serial tablet does not have this switching to eraser mode functionality that I am aware of. The stylus I am using came with this tablet, so its not like a Surface Pro stylus. In Windows, all three buttons are mappable and do not seem to be hard-wired:

[ArtZ II User's Manual - Page 14](https://www.manualslib.com/manual/534977/Wacom-Artz-Ii.html?page=14#manual)

jigpu, 2020-05-29:
Just because the buttons are mappable in Windows doesn't mean the stylus isn't setting the eraser flag under the hood. :wink: That said, its entirely possible that this is a bug in the kernel driver. The [relevant function](https://github.com/torvalds/linux/blob/master/drivers/input/tablet/wacom_serial4.c#L309) uses the same bit to control `BTN_TOOL_RUBBER` and `BTN_STYLUS2`. Because the `MODEL_ARTPAD_II` doesn't have the `F_HAS_STYLUS2` flag set, the kernel treats the bit as `BTN_TOOL_RUBBER`. Its possible that the "real" eraser (for pens which have one) is actually reported in another bit for this particular model. But then again, I was told "[eraser works too](https://github.com/linuxwacom/xf86-input-wacom/issues/40#issuecomment-469213474)" before submitting the patch to add support... Maybe multiple bits get set? We'd need to watch how the bits in the report change when the 2nd button is pressed compared to the eraser...

davidpgil, 2020-05-29:
here is an `evemu-record` of pressing the eraser against the tablet 4 times:

```
# EVEMU 1.3
# Kernel: 4.19.120-gnu-rt52
# DMI: dmi:bvnLENOVO:bvrGNET72WW(2.20):bd02/26/2015:svnLENOVO:pn20EF000HUS:pvrThinkPadW541:rvnLENOVO:rn20EF000HUS:rvrSDK0E50510WIN:cvnLENOVO:ct10:cvrNotAvailable:
# Input device name: "Wacom Digitizer II"
# Input device ID: bus 0x13 vendor 0x3e product 0000 version 0x5544
# Size in mm: 8x6
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 272 (BTN_LEFT)
#     Event code 273 (BTN_RIGHT)
#     Event code 274 (BTN_MIDDLE)
#     Event code 320 (BTN_TOOL_PEN)
#     Event code 321 (BTN_TOOL_RUBBER)
#     Event code 326 (BTN_TOOL_MOUSE)
#     Event code 330 (BTN_TOUCH)
#     Event code 331 (BTN_STYLUS)
#   Event type 2 (EV_REL)
#   Event type 3 (EV_ABS)
#     Event code 0 (ABS_X)
#       Value     1654
#       Min          0
#       Max      10240
#       Fuzz         0
#       Flat         0
#       Resolution 1270
#     Event code 1 (ABS_Y)
#       Value     1953
#       Min          0
#       Max       7680
#       Fuzz         0
#       Flat         0
#       Resolution 1270
#     Event code 24 (ABS_PRESSURE)
#       Value        0
#       Min         -1
#       Max        255
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 40 (ABS_MISC)
#       Value        2
#       Min          0
#       Max          0
#       Fuzz         0
#       Flat         0
#       Resolution   0
# Properties:
#   Property  type 0 (INPUT_PROP_POINTER)
#   Property  type 1 (INPUT_PROP_DIRECT)
N: Wacom Digitizer II
[... 455 lines cut ...]
E: 2.511855 0003 0018 0063	# EV_ABS / ABS_PRESSURE         63
E: 2.511855 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 2.527842 0003 0000 3739	# EV_ABS / ABS_X                3739
E: 2.527842 0003 0001 3655	# EV_ABS / ABS_Y                3655
E: 2.527842 0003 0018 0030	# EV_ABS / ABS_PRESSURE         30
E: 2.527842 0001 014a 0000	# EV_KEY / BTN_TOUCH            0
E: 2.527842 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 2.527853 0003 0018 0020	# EV_ABS / ABS_PRESSURE         20
E: 2.527853 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 2.527856 0003 0000 3730	# EV_ABS / ABS_X                3730
E: 2.527856 0003 0001 3662	# EV_ABS / ABS_Y                3662
E: 2.527856 0003 0018 0002	# EV_ABS / ABS_PRESSURE         2
E: 2.527856 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 2.543855 0003 0018 0000	# EV_ABS / ABS_PRESSURE         0
E: 2.543855 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 2.543864 0003 0000 3734	# EV_ABS / ABS_X                3734
E: 2.543864 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 2.559858 0003 0000 3738	# EV_ABS / ABS_X                3738
E: 2.559858 0003 0001 3658	# EV_ABS / ABS_Y                3658
E: 2.559858 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 2.559867 0003 0000 3747	# EV_ABS / ABS_X                3747
E: 2.559867 0003 0001 3652	# EV_ABS / ABS_Y                3652
E: 2.559867 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
# Closing after 3s inactivity
```

whot, 2020-06-02:
@davidpgil  - we''ll need the output from `hid-recorder`, evemu is too high-level to debug a kernel driver issues.  See https://gitlab.freedesktop.org/libevdev/hid-tools/ if it's not packaged in your distro.

Also, please **attach** recordings or any file longer than 10-15 lines.

davidpgil, 2020-06-02:
I cannot get `hid-tools` to to install or function in my distro. I believe it because I use Trisquel which is a free-software only distro and does not allow `pip` because `pip` has proprietary software in it. Is there some alternative sofware I can use that will do what you need in order for me to help out?

whot, 2020-06-02:
you can run `hid-recorder` directly from the git tree. beyond that it's your job to figure out the rest, I don't have the time to work around Trisquel.

davidpgil, 2020-06-04:
I manged to get help with why the `hid-recorder` was not working for me on the [Trisquel forums](https://trisquel.info/en/forum/how-work-around-no-python-pip-trisquel). The issue with that is solved. 

When I run the `hid-recorder` it does not seem to see my serial wacom tablet at all:
'/dev/hidraw0:	Apple Inc. Apple Watch Magnetic Charging Cable
/dev/hidraw1:	SINOWEALTH Game Mouse
/dev/hidraw2:	SINOWEALTH Game Mouse
/dev/hidraw3:	Ultimate Gadget Laboratories Ultimate Hacking Keyboard
/dev/hidraw4:	Ultimate Gadget Laboratories Ultimate Hacking Keyboard
/dev/hidraw5:	Ultimate Gadget Laboratories Ultimate Hacking Keyboard
/dev/hidraw6:	Ultimate Gadget Laboratories Ultimate Hacking Keyboard
/dev/hidraw7:	Ultimate Gadget Laboratories Ultimate Hacking Keyboard'

whot, 2020-06-05:
oh, a serial tablet? I missed that (re-reading the first comment, yes, you're using inputattach). In that case it predates hid and has its own serial parser in the kernel. So it's down to debugging the kernel driver `wacom_w8000.c` (iirc). Not something I can help with though, we're looking at a device more than a decade old here too and I simply don't have the cycles for that.

davidpgil, 2020-06-05:
Can this issue be re-opened then? I might take a crack at it since it in my best interests. I have some C knowledge but probably not enough... I suspect this may not be rocket science though.

whot, 2020-06-09:
there are very few users of serial devices left given that most of them are around 15y old now. Leaving the issue open for something that is unlikely going to get fixed unless *you* submit a patch to the upstream kernel is just a waste of developer attention (because every few months we'll click on this and go "what was this again?").

Once we have upstream support for remapping eraser buttons this will "accidentally" go away anyway in that you can configure the problem away.

davidpgil, 2020-06-19:
@whot is there a ticket someone opened regarding being able to remap eraser buttons then? I'd like to follow/ contribute info to that ticket.

whot, 2020-06-20:
There is now :) See https://github.com/linuxwacom/xf86-input-wacom/issues/134