linuxwacom/xf86-input-wacom #124: rare wine + Zbrush bug.

jekanev, 2020-05-24:
Have issue with pressure curve described on screenshots. my config
OS: Arch Linux x86_64
Kernel: 5.4.42-1-lts
tablet: Wacom Intuos4
wine-stable 5.0-2
ZBrush 2020
I try different  ZBrush and wine verison on 39.0 driver version issue are same, only if i downgrade xf86-input-wacom to version 33.0 all work fine. Also i tried Phtoshop CS 2 on wine with latest driver version with no issue and Blender, Krita, Gimp work well too. So this bug related to ZBrush only and driver version greater 33.0.
xf86-input-wacom 33.0 version all ok
[image]
xf86-input-wacom 39.0 verson broken
[image]
I ok with 33.0 driver all seems work fine, may be some one find this useful.

whot, 2020-05-26:
Sounds like d958ab79d21b57141415650daac88f9369a1c861. Try setting the `Presssure2K` option in and xorg.conf snippet like outlined in that commit.

jekanev, 2020-05-26:
> Sounds like [d958ab7](https://github.com/linuxwacom/xf86-input-wacom/commit/d958ab79d21b57141415650daac88f9369a1c861). Try setting the `Presssure2K` option in and xorg.conf snippet like outlined in that commit.

thanks after this setup everything worked.

closed 2020-05-26 by jekanev

jeaye, 2020-11-30:
I just ran into this same thing, using version 0.39.0 and also with ZBrush on WINE. Following the referenced commit above, the fix was to create the following file and log out/log in.

```
$ cat /etc/X11/xorg.conf.d/wacom.conf 
Section "InputClass"
    Identifier "Wacom pressure compatibility"
    MatchDriver "wacom"
    Option "Pressure2K" "true"
EndSection
```

Thank you!