linuxwacom/xf86-input-wacom #125: Surface Pro 5 tilt does not work

tihoulas, 2020-05-25:
Hi, I have a surface pro 5 which uses a n-trig stylus through intel IPTS bus.
Using IPTS kernel drivers, and a 50-wacom.conf under /etc/X11/xorg.conf.d to use the wacom driver (instead of libinput), everything works fine except tilt.
Tilt is reported in myPaint debug tool as (0.000079, 0.000079) when hovering and (1, 1) while drawing.
Evtest reports non-fixed values, and using libinput tilt works (but this causes other problems).
>Input driver version is 1.0.1
>Input device ID: bus 0x44 vendor 0x45e product 0x1f version 0x0
>Input device name: "ipts 045E:001F"
>Supported events:
>  Event type 0 (EV_SYN)
>  Event type 1 (EV_KEY)
>    Event code 320 (BTN_TOOL_PEN)
>    Event code 321 (BTN_TOOL_RUBBER)
>    Event code 330 (BTN_TOUCH)
>    Event code 331 (BTN_STYLUS)
>  Event type 3 (EV_ABS)
>    Event code 0 (ABS_X)
>      Value   3773
>      Min        0
>      Max     9600
>      Resolution      37
>    Event code 1 (ABS_Y)
>      Value   2365
>      Min        0
>      Max     7200
>      Resolution      42
>    Event code 24 (ABS_PRESSURE)
>      Value      0
>      Min        0
>      Max     4096
>    Event code 26 (ABS_TILT_X)
>      Value   9000
>      Min        0
>      Max    18000
>      Resolution    5730
>    Event code 27 (ABS_TILT_Y)
>      Value   9000
>      Min        0
>      Max    18000
>      Resolution    5730
>    Event code 40 (ABS_MISC)
>      Value  33910
>      Min        0
>      Max    65535
>  Event type 4 (EV_MSC)
>    Event code 4 (MSC_SCAN)
>Properties:
>  Property type 1 (INPUT_PROP_DIRECT)

jigpu, 2020-06-01:
You will need to figure out what the ABS_TILT_X and ABS_TILT_Y values report from evtest before we can work on updating the driver. On Wacom devices, it is reported as the angle away from vertical (in the X and Y axis, respectively) in degrees. This clearly can't be the case for your Surface device since the maximum reported value is apparently 18000.

Perhaps it reports a value of 0 for all the way to horizontal to the left/top, 9000 for vertical, and 18000 for horizontal to the right/bottom? If so, it should be a simple matter of just scaling and centering the data. If the data is actually encoding something like Alt-Azimuth insead of Tilt X/Y, that's going to be much more difficult.

tihoulas, 2020-06-03:
Ok, I cheched using evtest, it's exactly that : 

- vertical position is (9000, 9000),
- max tilt right is (15400, 9000),
- max tilt left is (3000, 9000),
- max tilt up is (9000,3000),
- max tilt down is (9000, 15400).

So I guess those ABS_TILT_X and ABS_TILT_Y are just angles in degrees multiplied by 100 as the kernel driver cannot use floats, 90° being vertical.

jigpu, 2020-06-03:
Okay, good :) In that case it looks like the issue is in these blocks of code:

https://github.com/linuxwacom/xf86-input-wacom/blob/095ba9c48e8cf8a96970e211ed734ba5883a0025/src/wcmUSB.c#L715-L723

https://github.com/linuxwacom/xf86-input-wacom/blob/095ba9c48e8cf8a96970e211ed734ba5883a0025/src/wcmUSB.c#L753-L761

These blocks of code are responsible for adjusting the zero-point offset (`wcmTiltX` and `wcmTiltY`) as well as the scale factors (`wcmTiltFactX` and `wcmTiltFactY`) that are later used to transform an input number (e.g. 9000) into a number of degrees from vertical (e.g. 0). The scale factor calculations look fine -- they will result of a scale factor of 0.01 just as you suspect is necessary. Instead, the zero-point offset may be to blame for your issues. Specifically, they set the offset to 0 and "assume the range is centered on 0" when -- at least in this particular case -- it isn't.

You might play around with setting these values to something like -9000. That looks like it should get the values into the right range for the rest of the driver to handle. If you can make that change and observe what difference is made in MyPaint and other apps, it should be telling.

tihoulas, 2020-06-03:
That did the trick. Replacing:
>common->wcmTiltOffY = 0;
with:
>common->wcmTiltOffY = - (absinfo.minimum +
>>>>absinfo.maximum) / 2;
and the same for X was enough to fix the problem.
I guess someone using a wacom tablet should check whether this change doesn't break tilt support for wacom tablets.

jigpu, 2020-06-03:
Great :+1: I'd like to check with the kernel devs to see if the odd range can be changed to be centered around 0. Its more likely to lead to a situation that works better across the board than working around it in userspace. The patches produced by https://github.com/linux-surface/linux-surface seem to already use zero-centered axes (see [here](https://github.com/linux-surface/linux-surface/blob/d719da54538603bef8a79877c61161333d137a28/patches/5.4/0007-ipts.patch#L1951-L1954)), so I'm not sure where the min/max values that you're seeing came from. Can you let me know which kernel are you using, and what version is it?

tihoulas, 2020-06-19:
Sorry for the very slow answer. I am using ubuntu kernel 5.3.18 + linux-surface patches.
However this is now outdated, as the linux-surface devs are moving to a different driver which does not have this problem (part of the driver is run in user space and give centered float values).

closed 2020-06-19 by tihoulas