linuxwacom/xf86-input-wacom #126: xsetwacom returns 0 even when there's an error

maruseu, 2020-05-26:
xsetwacom only seems to return a non zero when its run with absolutely no arguments

```
$ xsetwacom --set "Wacom Intuos S Pen stylus" Area 0 0 0 && echo success || echo failure
'Area' requires exactly 4 value(s).
success
```

jigpu, 2020-06-01:
Looks like there are a few error paths that need to be updated...

AstroLad47, 2020-07-13:
This is a fairly major task. I started to patch a version for my own use. After fixing get() and set() and still seeing the same problem, I looked deeper and discovered that many (most???) of the functions do not return status.They print to STDERR, but don't return an error code to the caller.

Joshua-Dickens, 2022-05-25:
This should be fully fixed with the commits above and return error codes properly.

closed 2022-05-25 by Joshua-Dickens