linuxwacom/xf86-input-wacom #128: [Documentation/Feature request] Guidance on configuring additional input devices

vintagepc, 2020-06-05:
I have an Intuos3 with some of the optional accessories. I cannot find any documentation on how to configure additional styli as input devices (e.g. airbrush, marker pen). xsetwacom only lists the defaults:
```
Wacom Intuos3 9x12 Pad pad              id: 17  type: PAD       
Wacom Intuos3 9x12 Pen stylus           id: 18  type: STYLUS    
Wacom Intuos3 9x12 Pen eraser           id: 19  type: ERASER    
Wacom Intuos3 9x12 Pen cursor           id: 20  type: CURSOR    
```
evtest shows that additional tool options are supported (and the airbrush code is fired when I bring it in proximity. Similarly, the art pen does work with rotation if one configures the dynamics correctly.

```
    Event code 320 (BTN_TOOL_PEN)
    Event code 321 (BTN_TOOL_RUBBER)
    Event code 322 (BTN_TOOL_BRUSH)
    Event code 323 (BTN_TOOL_PENCIL)
    Event code 324 (BTN_TOOL_AIRBRUSH)
    Event code 326 (BTN_TOOL_MOUSE)
    Event code 327 (BTN_TOOL_LENS)
```

Perhaps I am misunderstanding how this should work, but right now, I only get two tool types in Gimp - stylus and eraser. I'm trying to find out how to configure it such that I can have each stylus option have its own remembered tool - for example, if I bring the airbrush over the pad, it changes to that tool, if I bring the eraser or pen in proximity, it changes to those (To clarify - I know gimp remembers the last thing you've selected with each input device and that functionality is not your responsibility - I am just looking for a way to add more "extended input devices" for gimp to use). 

I have the following package versions installed on OpenSuSE 42.3, and the data files look to contain the needed information/ID references for all of the tools.
xf86-input-wacom-0.32.0-4.5.x86_64
libwacom2-0.23-2.1.x86_64
libwacom-data-0.23-2.1.x86_64
libwacom-tools-0.23-lp150.1.9.x86_64

```
[Device]
Name=Wacom Intuos3 9x12
DeviceMatch=usb:056a:00b2
Class=Intuos3
Width=12
Height=9
Layout=intuos3-9x12.svg
Styli=0x913;0x91b;0x813;0x81b;0x823;0x82b;0x801;0x885;0x017;0x097;
```

Additionally, is it possible to reconfigure the input type? For example, if I have both a grip pen and a classic pen, can I set it up so the classic pen registers as `BTN_TOOL_PENCIL` instead? Icing on the cake would be the ability to configure erasers on a per-tool basis, since they do have different tool IDs

Thank you! - I am happy to help provide debugging or testing as necessary.

jigpu, 2020-06-05:
The single "Wacom Intuos3 9x12 Pen stylus" device is intended to take care of all the various stylus tools that you can use with the tablet. Regardless of if you bring a general pen, airbrush, marker, etc. the events are going to come out of that device. It is up to the application (e.g. GIMP) to handle different stylus tools differently. The X driver will change its tool ID properties to match whatever tool is in prox, allowing applications to distinguish not only pens of different type (pen/airbrush/marker/etc.) but also individual pens of the same type since they each have a unique serial number. As far as I'm aware GIMP only remembers and applies settings based on the X11 device, not the tool ID information we provide.

As far as reconfiguring the input type, that is not an ability which is exposed to the user. You can, of course, make changes to the driver or application (again, if the application looks at the tool ID data) to interpret a classic pen like a pencil however.

vintagepc, 2020-06-05:
Thanks for the explanation. Sounds like I had wrongly assumed that using these devices would add or expose additional nodes under xsetwacom.

vintagepc, 2020-06-05:
Side Q - Are each of the different input options configurable via KCM-tablet? I saw a video where a developer demonstrated this (and showed an airbush config panel) but I only ever see cursor and pen.

jigpu, 2020-06-05:
> Thanks for the explanation. Sounds like I had wrongly assumed that using these devices would add or expose additional nodes under xsetwacom.

I believe the xf86-input-libinput driver does this, but applications sometimes have trouble with how it handles pen input.

> Side Q - Are each of the different input options configurable via KCM-tablet? I saw a video where a developer demonstrated this (and showed an airbush config panel) but I only ever see cursor and pen.

I haven't really used the KCM module, so I'm not sure if they have that option or how it might be activated if they do. I would check with its developer for more information.

vintagepc, 2020-06-05:
fair enough, thanks for your time and input. I may explore the code change to recognize the classic as a pencil; just wanted to make sure it wasn't already hidden somewhere. 

One additional naïeve question if you'll bear with me... what is the intent of the "ToolSerials" X driver option? https://github.com/linuxwacom/xf86-input-wacom/commit/1284bc5a858557a2739b67de401d25bcab3e1cf6

Certainly the documentation explains what it does from a literal perspective. But what I'm missing is the context of "when or why would I want to use this?"

In any case, thank you for an excellent tablet experience out of the box that mostly just works - I picked up a cheap used wacom after many years of frustrations with an Aiptek APT-6000U which did silly things like report pressure on the wrong axis (completely unusable in gimp or inkscape by default, this predated being able to configure the axes), or the hotkeys didn't work, it'd get stuck in relative mode, etc. etc. (Note it wasn't the developers' fault by any means, that particular hardware was just a mess of different internals all under the same model number. Some needed the hyperpen driver, some needed the aiptek module, some worked with wacom drivers... and even within the same hardware, some needed the programming commands sent in a different order or things would not work)

jigpu, 2020-06-05:
> One additional naïeve question if you'll bear with me... what is the intent of the "ToolSerials" X driver option? 1284bc5
>
> Certainly the documentation explains what it does from a literal perspective. But what I'm missing is the context of "when or why would I want to use this?"

I'm embarrassed to say this, but I'd completely forgotten about that particular option... Even more embarrassing, I'm not 100% sure what it does or how it works. Its been quite a while since I provided the Acked-by for that patch :laughing: 

Just looking at the functions its modifying, it appears that it might be something similar to what you're asking for. "Hotplugging child devices" is how the eraser and cursor devices are added, so it looks like using this option would potentially create serial number-specific devices. The "type" information might affect exactly what devices are generated ("pen" and "airbrush" make both stylus and eraser devices; "artpen" just a stylus; "cursor" a cursor device?) but it isn't 100% clear. The serial number for a particular tool can be found by running `xsetwacom get <id> ToolSerial` on the stylus device while holding the pen in prox.

I don't know if the code for it still works, but I'd definitely go exploring if you're interested. There's some rough documentation of [how xorg.conf.d files work](https://github.com/linuxwacom/xf86-input-wacom/wiki/xorg.conf.d) on our wiki, and you can also run `man xorg.conf.d` or search around the net for more information.

Let me know what you find out! :+1:

vintagepc, 2020-06-05:
iiinteresting... I will try playing with it and see what happens. Nothing to be embarrassed about... Develop enough and the older stuff goes fuzzy.;)

vintagepc, 2020-06-05:
Well what do you know... that seems to be EXACTLY what I was looking for. I added a line to a conf.d snippet and *boom*, there's now "airbrush stylus" and "airbrush eraser" slave pointers.

I can probably use that for the classic vs grip pen too, by registering the classic as a different pen entry. 

So I guess this does remain a documentation task - that feature should be highlighted somewhere for configuring input devices with gimp or inkscape, not relegated to some obscure commit description! :smile:

vintagepc, 2020-06-05:
I've filed a PR for the manpage - no reason I can't do the legwork on that. My day was already pretty good after I found a supply of these optional styli for cheap... it got better when I found out about this option. :1st_place_medal:

whot, 2020-06-08:
FTR:

> I believe the alternative xf86-input-libinput driver does this, but applications sometimes have trouble with how it handles pen input.

that's correct, but that's also a function how the underlying libinput handles it. the X driver will create a new device once a device goes into proximity the first time. You can't pre-configure devices as such. Or at least that has to happen out-of-band (e.g. GNOME)

> that seems to be EXACTLY what I was looking for.

Yes, I can confirm that this option is what you need here. It forces the driver to create X devices for a specific serial number, in addition to the generic catch-all devices it usually creates.

PR #129 (2021-02-01): Better explanation of the "ToolSerials" option
Closes #128 

This adds a better explanation of the "ToolSerials" option and why someone might want to use it - this was not readily apparent, and it is very useful for tablets that support the feature when you have multiple physical styli
closed 2021-02-01 by jigpu