linuxwacom/xf86-input-wacom #131: Status of support for external serial (tty) tablets?

jlturriff, 2020-06-09:
I have an old Intuos model GD-0912-R tablet that connects via RS-232 serial port, that I originally bought to use with my Amiga computer, before USB had been invented.  I'd like to use it with my OpenSuSE Leap 15.1 desktop system, but I see that currently there is no support for such serial devices.
Can you please tell me what is the status of support for such devices?  The tablet works fine as far as I can tell, but with a Linux machine it appears that it's just a boat-anchor. :-)

whot, 2020-06-09:
there's some support in the kernel for serial devices (check the `wacom_w800.c` file) and they get hooked up through `inputattach`. That's the recommended approach, if you want this device to work you need to fix the kernel driver so it shows up as normal evdev device to userspace.

The git history of xf86-input-wacom (pre-2009) still has the serial parsing code (`src/wcmSerial.c`) for original Intuos devices I think, so with a suitable pickaxe and shovel you can go digging.

It's not something that we'll restore here in this driver, the number of users does not justify the effort of resurrecting and maintaining it. You can try your luck submitting this to the kernel if you get it to work at which point it'll just work with this driver.

closed 2020-06-09 by whot

jlturriff, 2020-06-10:
On 2020-06-09 03:02:37 Peter Hutterer wrote:
> there's some support in the kernel for serial devices (check the
> `wacom_w800.c` file) and they get hooked up through `inputattach`. That's
> the recommended approach, if you want this device to work [...]

	Okay, thanks; now I know where I stand.

Leslie