linuxwacom/xf86-input-wacom #132: Wacom Intuos S: Not working after sleep/hibernate/display change -> needs restart

pjaap, 2020-06-17:
Hi! If have a Wacom Intuos S connected with USB and it works fine so far.

But when I do one of the following:
 - put PC to sleep
 - hibernate
 - plug in an external display

the device is unusable with a jumpy cursor. I need to restart my computer.

Interesting part: If if plug the device in the status LED turns on for about 2 seconds and then gets very dim. If I restart my PC and plug the device in, the LED stays bright and the device is usable.

I think it has something to do with the X-Server. What can I do to provide more info?

Always reproducible on
- Debian 11, KDE, Kernel 5.6 and 5.7
- Kbuntu 20.04, Kernel 5.4 and 5.6

jigpu, 2020-06-25:
Sounds like a classic case of the tablet mis-identifying your PC as an Android device: https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect

closed 2020-06-25 by jigpu

pjaap, 2020-07-01:
Hi! Yes this did the trick! Thank you very much!