linuxwacom/xf86-input-wacom #133: Wacom Cintiq 16 - Pen not detected

JulesSha, 2020-06-19:
Good day/evening everyone,
I recently bought a Wacom tablet but for the life of me I can't make it work on Ubuntu (20.04 LTS). The tablet appears, I did my best following the tutorials to install the different drivers/packages needed but still the pen does not work (not detected by the computer).
If someone had the same issue and managed to solve it or if somebody has sufficient knowledge to lend me a hand: that'd be much appreciated!
Have a good day!

jigpu, 2020-06-23:
The Cintiq 16 should work out of the box with Ubuntu 20.04. You say that "the tablet appears", but that it is "not detected by the computer". Could you clarify what you mean? What works and what doesn't?

You can run [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball here so we can review your system's hardware/software/logs.

blenderli, 2020-06-26:
Hello jigpu
I have a similar issue with my Wacom cintiq pro 16 no Pen and No Touch. Screen is fine but Wacom is not  detect in lsusb. See Sysinfo 

[sysinfo.B9NayJdHWK.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/4836213/sysinfo.B9NayJdHWK.tar.gz)

jigpu, 2020-06-26:
Hi @blenderli -- can you confirm that the tablet's USB plug was connected when you ran sysinfo? The kernel logs don't mention the hardware at all (not even a failure)...

blenderli, 2020-06-27:
Hello jigpu
Yes my USB is plug in. It's so strange. At my work i use my Cintiq on Windows and it works. That means the USB Kabel can not be broken.

blenderli, 2020-06-27:
Hello jigpu
I fix it, but it was very strange in need to plug out all USB C Cable many times until it works. I cant believe it.

jigpu, 2020-06-29:
That is strange... I haven't heard of others needing to do that with their Cintiq 16. Since the kernel logs don't show anything tablet related, its possible you're experiencing a hardware issue (bad plug / cable?) or some kind of USB bug. It would be interesting to know if the tablet behaves any better with another OS (e.g. running Fedora 32 from a USB stick) or another computer.

blenderli, 2020-06-30:
Hoi jigpu

By rebooting my PC i ge this message. But now everything works. Maybe it's helpful for you. Have e nice time thanks.

[image]

closed 2020-07-17 by jigpu