linuxwacom/xf86-input-wacom #134: Allow remapping of eraser buttons

whot, 2020-06-20:
Some styli have the windows-style eraser button instead of the eraser at the back of the stylus. The button generates proximity events in the device, so the sequence on eraser button press is: stylus proximity out, eraser proximity in, and the reverse on release.

libwacom recently [gained the API](https://github.com/linuxwacom/libwacom/commit/edde85844cbfa146f7fb7205ddec2252053fc012) to identify those styli. This means in the xf86-input-wacom driver we should provide an option to switch the eraser button into a normal button mode.

FTR equivalent issue in libinput: https://gitlab.freedesktop.org/libinput/libinput/-/issues/370

davidpgil, 2020-06-20:
I hope this can solve my issue with my trusty ancient serial tablet.

whot, 2021-12-21:
As I commented in the libinput issue linked above, this requires a significant amount of complex heuristics - e.g. every pen interaction needs to be delayed in case the eraser comes into proximity. It's very unlikely that this will be added to the driver at any point in the forseeable future.

Pinglinux, 2022-08-04:
Anyone interested in to this request please contribute to libinput.

closed 2022-08-04 by Pinglinux