linuxwacom/xf86-input-wacom #135: CTL-472 is not working

ashutosh0gupta, 2020-06-27:
In the latest release of this driver is not working for CTL-472..In my investigation I found the following.

In wcmUSB.c, the following codes is not returning correct bits for CTL-472.

`ioctl(pInfo->fd, EVIOCGBIT(EV_KEY, (sizeof(unsigned long)
						* NBITS(KEY_MAX))), common->wcmKeys)`

At BTN_TOOL_PEN position, array common->wcmKeys is not set.
However, it is set at 0x110 position indicating that CTL-472 is declaring itself to be mouse.

Is there a workaround for this?

jigpu, 2020-06-29:
Can you please perform the following steps to get some additional device information:
1. Run [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh). When it is done, it will print out the name of a tarball containing the gathered data.
2. Drag-and-drop the tarball into the comment box here to attach it.
3. Run `sudo libinput record > ctl472-stylus.log`
4. Select the CTL-472 stylus device
5. You should see the message `recording to 'stdout'` -- press CTRL+C
6. Drag-and-drop the ctl472-stylus.log file into the comment box here to attach it also.

ashutosh0gupta, 2020-06-30:
[sysinfo.D2T1t6KhjF.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/4849555/sysinfo.D2T1t6KhjF.tar.gz)

libtools on my laptop has no option record. I am attaching the output of 
[libinput-list-devices.log](https://github.com/linuxwacom/xf86-input-wacom/files/4849946/libinput-list-devices.log)

whot, 2020-06-30:
`libinput record`, not `libtool` :)

ashutosh0gupta, 2020-06-30:
Sorry for the typo, I meant libinput

'
sudo libinput record
libinput: record is not a libinput command or not installed. See 'libinput --help'
'

whot, 2020-06-30:
try installing `libinput-utils` or some similar package

ashutosh0gupta, 2020-06-30:
I had older version of libinput. After compiling from the source, I managed to make it work.
[ctl-472.txt](https://github.com/linuxwacom/xf86-input-wacom/files/4851397/ctl-472.txt)

jigpu, 2020-06-30:
Thanks. It looks like your kernel is using the "hid-generic" driver rather than the "wacom" driver for some reason. The hid-generic driver isn't able to initialize and communicate with the hardware in its full tablet mode, which explains why it is misbehaving. Your sysinfo output clearly shows that the Wacom driver is available (both the system-provided version and the input-wacom backport), but there were some problems getting additional information on the drivers.

Please run the following commands and report back their output:
* `sudo modinfo wacom`
* `sudo modinfo hid-generic`

Also, please reboot your system with the tablet connected and then attach the dmesg log so we can check for errors that might not have been picked up by sysinfo:
* `dmesg > dmesg.log`

ashutosh0gupta, 2020-06-30:
[dmesg.log](https://github.com/linuxwacom/xf86-input-wacom/files/4852976/dmesg.log)

[wacom.txt](https://github.com/linuxwacom/xf86-input-wacom/files/4852982/wacom.txt)
[hid-generic.txt](https://github.com/linuxwacom/xf86-input-wacom/files/4852983/hid-generic.txt)

jigpu, 2020-06-30:
Thanks again. I see some errors related to secure boot in the log, which could explain why the input-wacom driver isn't being loaded. See [our wiki page on secure boot](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot) for more information. Try the steps listed under the "Enabling and Disabling Secure Boot" heading to temporarily disable secure boot to see if it fixes your issue. If it does, you may want to re-enable secure boot and then follow the rest of the guide to let you build signed modules instead.

ashutosh0gupta, 2020-06-30:
Thank you! This worked. Let me write my fix for someone else who may have the same problem.

I was trying to use CTL-472 tablet on Lubuntu-18.04. It did not work. My fix.

 * I installed latest releases of xf86-input-wacom and input-wacom
 * I need to follow the instructions in https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot to ensure that wacom driver is loaded

After that it worked.

closed 2020-07-01 by jigpu