linuxwacom/xf86-input-wacom #136: Build failed when pre-install xf86-input-wacom

YuquanW, 2020-07-20:
An error occured after I extracted the tar file and ran the command:

`if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"
`

The error was as follows:

> In file included from ../src/xf86Wacom.h:26:0,
                 from ../src/xf86Wacom.c:46:
../include/Xwacom.h:32:10: fatal error: X11/keysym.h: No such file or directory
 #include <X11/keysym.h>
          ^~~~~~~~~~~~~~
compilation terminated.

Could anyone help me solve this problem? Thanks.

whot, 2020-07-20:
```
$ find /usr/include -name keysym.h 
/usr/include/X11/keysym.h
$ rpm -qf /usr/include/X11/keysym.h 
xorg-x11-proto-devel-2019.1-3.fc32.noarch
$ rpm -ql xorg-x11-proto-devel-2019.1-3.fc32.noarch | grep ".pc"
....
/usr/share/pkgconfig/xproto.pc
```
So the header is supplied by whatever supplies `xproto.pc` and we require this in `configure.ac` line 53/56. Not immediately sure what's going on there, `configure` *should* be checking for xproto.

YuquanW, 2020-07-21:
> ```
> $ find /usr/include -name keysym.h
> /usr/include/X11/keysym.h
> $ rpm -qf /usr/include/X11/keysym.h
> xorg-x11-proto-devel-2019.1-3.fc32.noarch
> $ rpm -ql xorg-x11-proto-devel-2019.1-3.fc32.noarch | [...]

I have tried the commands above. The second one returned

`$ rpm -qf /usr/include/X11/keysym.h 
file /usr/include/X11/keysym.h is not owned by any package
 `

whot, 2020-07-21:
tbh, that seems like a packaging issue on the distro (Fedora? SuSE?) Are you up-to-date on all the packages? Is there another `keysym.h` floating around somewhere? Try `dnf install /usr/include/X11/keysym.h` and see what that picks out.

YuquanW, 2020-07-21:
> tbh, that seems like a packaging issue on the distro (Fedora? SuSE?) Are you up-to-date on all the packages? Is there another `keysym.h` floating around somewhere? Try `dnf install /usr/include/X11/keysym.h` and see what that picks out.

I updated all the packages on my Ubuntu 19.04 and tried  `$ rpm -qf /usr/include/X11/keysym.h`, it failed again. 

`sudo apt install /usr/include/X11/keysym.h ` returned an error in Chinese, which means "There is an unsupported file in command line: /usr/include/X11/keysym.h". After that I tried `sudo find / -name keysym.h` and it seemed like `/usr/include/X11/keysym.h` is the only path to the file.

whot, 2020-07-21:
`rpm` won't work too well on ubuntu, so ignore that part. I think there's an `apt-file` command or something to search for which package a file provides but I don't know off-hand what exactly it is.

Either way, I'm putting this down to a Ubuntu packaging bug. The file should be there, the `xproto.pc` which we use to link should point at `/usr/include` so from our side everything seems to be fine.

closed 2020-07-21 by whot

jigpu, 2020-07-21:
According to https://packages.ubuntu.com/bionic/all/x11proto-dev/filelist both `/usr/share/pkgconfig/xproto.pc` and `/usr/include/X11/keysym.h` are installed by the x11proto-dev package, which is a dependency of libx11-dev that should have already been installed...

@YuquanW did you install all the build dependencies first?
~~~
sudo apt-get install autoconf pkg-config make xutils-dev libtool xserver-xorg-dev$(dpkg -S $(which Xorg) | grep -Eo -- "-hwe-[^:]*") libx11-dev libxi-dev libxrandr-dev libxinerama-dev libudev-dev
~~~

Also, Ubuntu 19.04 isn't supported anymore. I would recommend allowing it to update to 20.04, which is the current long-term support release.

YuquanW, 2020-07-22:
> According to https://packages.ubuntu.com/bionic/all/x11proto-dev/filelist both `/usr/share/pkgconfig/xproto.pc` and `/usr/include/X11/keysym.h` are installed by the x11proto-dev package, which is a [...]

Yes, I installed all the build dependencies first. I'll try to update to 20.04, thank you very much.