linuxwacom/xf86-input-wacom #138: Intuos S: only 1/3 of it is somewhat working...

doej1367, 2020-08-04:
I got a problem with my Intuos S, 
where about a third of the tablet is projected onto my full display, 
making my pen go three times faster in the x direction than in the y direction 
and the other two thirds of the tablet are useless for the x direction but y dir is still picked up.
Any ideas how i could solve that?
Thanks in advance

**What ive tried so far**:
+ YT Tutorial: https://www.youtube.com/watch?v=f3C6OwjOJxI
  + https://help.ubuntu.com/community/Wacom/LatestDriver
  + https://linuxwacom.github.io/
  + https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source
  + https://github.com/linuxwacom/input-wacom/releases
+ Old issues: https://github.com/linuxwacom/xf86-input-wacom/issues/21

**my specs**: xUbuntu 18.04.4 LTS, 1920x1080, Intuos S 
[image]

**Settings**: .sh file i used for startup
`## Device names and ID numbers from 'xinput list' entered in a terminal.
#
## In the example "Device name" not ID # is used.  Note if you use the
## xorg.conf the "Device names" will be stylus, eraser, touch, and pad.
#
## If you are hot plugging use "Device name" as ID # can change.
#
## ClickForce changes name to Threshold with xf86-input-wacom 0.10.9 (11-19-10)
#
## Warning:  Changing Mode to either Absolute or Relative in stylus/eraser stops
## the mouse from being able to pull guidelines out of the ruler in Gimp.

## stylus = ID 11 = "Wacom Co.,Ltd. Intuos S stylus"
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" Suppress "4"  # data pt.s trimmed, default is 4, 0-20
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" RawSample "2"  # data pt.s filtered, default is 2, 0-100
#xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" ClickForce "27"  # pressure, default is 27, range is 0-2047
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" Threshold "27"  # pressure, default is 27, range is 0-2047
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" PressureCurve "5 10 90 95"  # Bezier curve, default is 0,0,100,100
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" TabletPCButton "on"  # stylus tip + button, or "off" for hover mode
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" Mode "Absolute"  # or Relative cursor movement
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" Button "1"  # left mouse click
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" Button "3"  # right mouse click
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" Button "2"  # middle mouse click
# default coordinates in /var/log/Xorg.0.log
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" Area 0 0 15200 9500
xsetwacom set "Wacom Co.,Ltd. Intuos S stylus" MapToOutput 1920x1080+0+0

[ then the same for eraser...]

[and some other things all commented out ...]

## Developed with duffman.c.d. & artik1024 (& SergeantOreo's Xorg.0.log for coordinates) @ Ubuntu forums
`

doej1367, 2020-08-04:
now I restarted without hotpluging my tablet and i think it works now... wierd but whatever...

doej1367, 2020-08-04:
ok now the only remainder of that whole session is an error message at startup that i havent had before... but other than that it all seems fine.

jigpu, 2020-08-04:
The source of this problem is the tablet's firmware mis-detecting Linux as Android. See https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect for more information and a workaround.

closed 2020-08-04 by jigpu