linuxwacom/xf86-input-wacom #146: Touch Ring not working as expected

StoicIndividual, 2020-10-15:
I have a Wacom Intuos Pro m and its Touch Ring has 4 settings and one button to move switch between all of them, with:
`xsetwacom set "Wacom Intuos Pro M Pad pad" AbsWheelDown "key PgDn"`
and
`xsetwacom set "Wacom Intuos Pro M Pad pad" AbsWheelUp "key PgUp"`
I bind PgDn and PgUp to **all** the settings, I can't seem to find an option to map a macro to 1 setting of the Touch Ring, how do I bind commands/macros to each individual setting?

jigpu, 2021-02-10:
The X driver (and by extension, the xsetwacom tool) doesn't know anything about the state of the mode switch. In order to have the ring perform different actions in different modes, you'd need to have some software which re-programs the ring whenever the mode changes. We have a program that does this for Wacom's ExpressKey Remote product (see the [ekr_monitor repository](https://github.com/linuxwacom/ekr_monitor)) but support for Intuos Pro and other products is still a "todo" item. It shouldn't take a lot of effort to extend the program but its been a low priority item for us so far...

jigpu, 2021-07-21:
Closing due to inactivity.

closed 2021-07-21 by jigpu