linuxwacom/xf86-input-wacom #149: Wacom DTH 1152 touchscreen not working in Ubuntu 20.04

ekaveeramaths, 2020-11-01:
I have connected my Interactive Wacom pen tablet DTH 1152 to Ubuntu. I can see the screen of my Ubuntu home page, but touchscreen is not working, i am unable to use pen on my wacom tablet. Please give me a solution.

jigpu, 2020-11-05:
Hi, please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball here.

mperreau, 2021-01-04:
I have the same issue

jigpu, 2021-01-05:
@mperreau same request as above: please run and attach the output from our sysinfo script. This will give us more information about what devices the system sees and any errors that might have been logged.

mperreau, 2021-01-24:
Thanks, here it is
[sysinfo.JLyJ43Fybs.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/5863134/sysinfo.JLyJ43Fybs.tar.gz)

jigpu, 2021-01-25:
Thanks. Can you confirm if the tablet was connected when you ran the script? I don't see any indication of a Wacom device at all; not even an informational message from the kernel that it had detected a USB device...

mperreau, 2021-02-06:
Yes, the tablet was connect by USB port (the bluetooth recognizes the presence of the tablet but cannot connect to it)

jigpu, 2021-02-09:
Can you please disconnect the tablet, reboot, connect the tablet, wait 30 seconds, and then run the following two commands:
~~~
dmesg > dmesg.txt
cat /proc/uptime
~~~
Afterwards, attach the dmesg in a reply and let me know what the output from the second command was.

mperreau, 2021-02-12:
[dmesg.txt](https://github.com/linuxwacom/xf86-input-wacom/files/5973571/dmesg.txt)
Here is the dmesg.txt
and the result of cat /proc/uptime is:
385257.66 589958.80
Thanks for your help
Michel

mperreau, 2021-02-12:
Hi, everything is working now !
Since the upgrade to the distribution 20 I have not made anything else than the usual upgrade proposed by ubuntu, and now this seems to be a miracle, tablet is perfectly working.
Thanks again
Michel

jigpu, 2021-02-12:
That's good news :smile:

I only see an Intuos tablet (CTL-6100WL) in your logs, not a DTH-1152 like the original poster. Can you confirm if that's the tablet you were trying to get connected? I can't think of much that would have changed recently to affect it, but hopefully whatever changed is a permanent fix for you :)

mperreau, 2021-02-19:
Yes I have not changed my tablet, Sorry, I aw that somebody had a problem with "a" wacom tablet after upgrading ubuntu and so I asked help without looking to the model of tablet.
Anyway I confirm anything works fine presently. I am using it without problem.
Thanks

closed 2021-02-23 by jigpu