linuxwacom/xf86-input-wacom #150: Wacom 4898 has no input driver or is it a misconfigurationm, Ubuntu

IRoninCoder, 2020-11-20:
I have a [HP x360 G3](https://support.hp.com/us-en/document/c06074000#AbT8) and it has Wacom tablet technology. I assume it's using a digitizer which should work with a Samsung S-pen, but it's not and I believe this is a problem with my Ubuntu which requires a fix/tweak.

    Ubuntu 20.04
    xrandr program version       1.5.0
    Server reports RandR version 1.6
    X11

The display manager makes no difference, GDM and OpenBox behave the same. Here are the identified devices:

    % xsetwacom list                                 
    Wacom HID 4898 Pen stylus               id: 12  type: STYLUS    
    Wacom HID 4898 Finger touch             id: 13  type: TOUCH     
    Wacom HID 4898 Pen eraser               id: 23  type: ERASER  

running `% xinput test 12` returns nothing when the S-pen is touching the screen. I also tried running with the name instead of ID, same behavior persists. No output is shown and the prompt stays still. Also tried the buttons on the S-pen, but nothing is displayed. Additionally there's no cursor identified when S-pen is in screen proximity/touching, just thought to mention this.

Next checked the `/var/log/Xorg.0.log`:

    [    13.433] (II) config/udev: Adding input device Wacom HID 4898 Pen (/dev/input/event6)
    [    13.433] (**) Wacom HID 4898 Pen: Applying InputClass "libinput tablet catchall"
    [    13.433] (**) Wacom HID 4898 Pen: Applying InputClass "Wacom USB tablet class"
    [    13.433] (**) Wacom HID 4898 Pen: Applying InputClass "Wacom tablet class"
    [    13.433] (II) LoadModule: "wacom"
    [    13.433] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
    [    13.437] (II) Module wacom: vendor="X.Org Foundation"
    [    13.437] 	compiled for 1.20.7, module version = 0.39.0
    [    13.437] 	Module class: X.Org XInput Driver
    [    13.437] 	ABI class: X.Org XInput driver, version 24.1
    [    13.437] (II) Build version: 0.39.0
    [    13.437] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
    	Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
    	Graphire4 4x5, Graphire4 6x8, BambooFun 4x5, BambooFun 6x8,
    	Bamboo1 Medium, Graphire4 6x8 BlueTooth, CTL-460, CTH-461, CTL-660,
    	CTL-461/S, Bamboo Touch, CTH-460/K, CTH-461/S, CTH-661/S1, CTH-461/L,
    	CTH-661/L, Intuos 4x5, Intuos 6x8, Intuos 9x12, Intuos 12x12,
    	Intuos 12x18, PTU600, PL400, PL500, PL600, PL600SX, PL550, PL800,
    	PL700, PL510, PL710, DTI520, DTF720, DTF720a, DTF521, DTU1931,
    	DTU2231, DTU1631, DTH1152, DTH1152, DTK2451, DTH2452, DTH2452,
    	Intuos2 4x5, Intuos2 6x8, Intuos2 9x12, Intuos2 12x12, Intuos2 12x18,
    	Intuos2 6x8 , Volito, PenStation, Volito2 4x5, Volito2 2x3,
    	PenPartner2, Bamboo, Bamboo1, Bamboo1 4x6, Bamboo1 5x8, Intuos3 4x5,
    	Intuos3 6x8, Intuos3 9x12, Intuos3 12x12, Intuos3 12x19,
    	Intuos3 6x11, Intuos3 4x6, Intuos4 4x6, Intuos4 6x9, Intuos4 8x13,
    	Intuos4 12x19, Intuos4 WL USB Endpoint,
    	Intuos4 WL Bluetooth Endpoint, Intuos5 touch S, Intuos5 touch M,
    	Intuos5 touch L, Intuos5 S, Intuos5 M, Intuos Pro S, Intuos Pro M,
    	Intuos Pro L, Cintiq 21UX, Cintiq 20WSX, Cintiq 12WX, Cintiq 21UX2,
    	Cintiq 24HD, Cintiq 22HD, Cintiq 24HD touch (EMR digitizer),
    	Cintiq 13HD, DTK2241, DTH2242, Cintiq 22HDT, Cintiq 27QHDT,
    	Cintiq 27QHDT, Mobilestudio Pro 13, MobileStudio Pro 16,
    	MobileStudio Pro 13, MobileStudio Pro 16, Cintiq 13 FHD Pro,
    	Cintiq 16 UHD Pro, Cintiq Pro 24, Cintiq Pro 32, Cintiq 13 FHD Pro,
    	Cintiq 16 UHD Pro, Cintiq Pro 24, Cintiq Pro 32, Cintiq Pro 24,
    	Mobilestudio Pro 13, MobileStudio Pro 16, MobileStudio Pro 13,
    	MobileStudio Pro 16, MobileStudio Pro 16, MobileStudio Pro 16,
    	TabletPC 0x90, TabletPC 0x93, TabletPC 0x97, TabletPC 0x9A,
    	CapPlus  0x9F, TabletPC 0xE2, TabletPC 0xE3, TabletPC 0xE5,
    	TabletPC 0xE6, TabletPC 0xEC, TabletPC 0xED, TabletPC 0xEF,
    	TabletPC 0x100, TabletPC 0x101, TabletPC 0x10D, TabletPC 0x116,
    	TabletPC 0x12C, TabletPC 0x4001, TabletPC 0x4004, TabletPC 0x5000,
    	TabletPC 0x5002, usb:172f:0024, usb:172f:0025, usb:172f:0026,
    	usb:172f:0027, usb:172f:0028, usb:172f:0030, usb:172f:0031,
    	usb:172f:0032, usb:172f:0033, usb:172f:0034, usb:172f:0035,
    	usb:172f:0036, usb:172f:0037, usb:172f:0038, usb:172f:0039,
    	usb:172f:0051, usb:172f:0052, usb:172f:0053, usb:172f:0054,
    	usb:172f:0055, usb:172f:0056, usb:172f:0057, usb:172f:0058,
    	usb:172f:0500, usb:172f:0501, usb:172f:0502, usb:172f:0503,
    	usb:1b96:0001, usb:17ef:6004
    [    13.437] (II) Using input driver 'wacom' for 'Wacom HID 4898 Pen'
    [    13.437] (**) Wacom HID 4898 Pen: always reports core events
[... 30 lines cut ...]
    [    13.477] (EE) Wacom HID 4898 Finger: Invalid type 'cursor' for this device.
    [    13.477] (II) Wacom HID 4898 Finger: type not specified, assuming 'touch'.
    [    13.477] (II) Wacom HID 4898 Finger: other types will be automatically added.
    [    13.477] (**) Wacom HID 4898 Finger touch: panscroll is 0
    [    13.477] (**) Wacom HID 4898 Finger touch: panscroll modified to 13
    [    13.477] (--) Wacom HID 4898 Finger touch: maxX=11752 maxY=6608 maxZ=0 resX=40000 resY=40000 
    [    13.477] (II) Wacom HID 4898 Finger touch: hotplugging dependent devices.
    [    13.477] (EE) Wacom HID 4898 Finger touch: Invalid type 'stylus' for this device.
    [    13.477] (EE) Wacom HID 4898 Finger touch: Invalid type 'eraser' for this device.
    [    13.477] (EE) Wacom HID 4898 Finger touch: Invalid type 'cursor' for this device.
    [    13.477] (EE) Wacom HID 4898 Finger touch: Invalid type 'pad' for this device.
    [    13.477] (II) Wacom HID 4898 Finger touch: hotplugging completed.
    [    13.519] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-WCOM4890:00/0018:056A:4898.0001/input/input22/event7"
    [    13.519] (II) XINPUT: Adding extended input device "Wacom HID 4898 Finger touch" (type: TOUCH, id 13)
    [    13.520] (**) Wacom HID 4898 Finger touch: (accel) keeping acceleration scheme 1
    [    13.520] (**) Wacom HID 4898 Finger touch: (accel) acceleration profile 0
    [    13.520] (**) Wacom HID 4898 Finger touch: (accel) acceleration factor: 2.000
    [    13.520] (**) Wacom HID 4898 Finger touch: (accel) acceleration threshold: 4
    [    13.520] (II) config/udev: Adding input device Wacom HID 4898 Finger (/dev/input/mouse1)
    [    13.520] (II) No input driver specified, ignoring this device.
    [    13.520] (II) This device may have been added with another device file.

Next installed `evtest` and `evemu-tools` attempting to follow [this](http://migueleonardortiz.com.ar/linux/measuring-level-of-pressure-in-wacom-tablets-or-other-devices/1988) page, but again running `evtest` and selecing `6` which is the pen, does nothing and prompt just sits there as pen touches the screen.

There's a line in `Xorg.0.log` suggesting an input driver is missing. Can anyone help me identify what input driver is required to make this work? or if that's just an after effect, then something else I should change/fix?

jigpu, 2020-11-20:
Your HP device has a pen sensor based on AES technology rather than the EMR technology used by the S-Pen. AES and EMR are incompatible, so it is not possible to use the S-Pen with most of the pen-capable tablet PCs on the market today (which have largely all moved to AES).

The original HP pen is probably the [HP Rechargeable Active Pen G3](https://store.hp.com/us/en/pdp/hp-rechargeable-active-pen-g3), though other AES pens should also work. Wacom, for example, sells the [Bamboo Ink](https://www.wacom.com/en-us/products/stylus/bamboo-ink) and [Bamboo Ink Plus](https://www.wacom.com/en-us/products/stylus/bamboo-ink-plus), though neither explicitly list the 1030 G3 in the compatibility list.

closed 2020-11-20 by jigpu

whot, 2020-11-22:
> [    13.520] (II) config/udev: Adding input device Wacom HID 4898 Finger (/dev/input/mouse1)
> [    13.520] (II) No input driver specified, ignoring this device.
> [    13.520] (II) This device may have been added with another device file.

> There's a line in Xorg.0.log suggesting an input driver is missing.

FTR, this is for the `mouse1` device node and we haven't had a driver (or a need for those drivers) for over a decade now. So you can ignore those lines.