linuxwacom/xf86-input-wacom #152: Wacom One Pen Display 13 (DTC133) - display unknwon (Arch Linux)

GonerGenesis, 2020-12-02:
Hi,
I tried to connect my new Wacom One. The input (pen) on the black tablet screen is working, but it would be nice to have the display not black.
I suppose the problem lies here: "modinfo: ERROR: Module hid_wacom not found" but I don't know how to get the module running.
[sysinfo.1F2oMzDdo7.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/5629447/sysinfo.1F2oMzDdo7.tar.gz)

GonerGenesis, 2020-12-02:
Nevermind - I just had to flip the connector at the tablet!
Beside I still get an error:
```
[ 3444.281104] input: Wacom One Pen Display 13 Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-11/2-11:1.0/0003:056A:03A6.0005/input/input30
[ 3444.281205] wacom 0003:056A:03A6.0005: hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom One Pen Display 13] on usb-0000:00:14.0-11/input0
[ 3444.281727] wacom 0003:056A:03A6.0006: Unknown device_type for 'Wacom Co.,Ltd. Wacom One Pen Display 13'. Ignoring.
```

jigpu, 2020-12-03:
Just to be clear, are the tablet and screen are both working correctly now? Some tablets have an extra interface that our driver cannot use which will produce an "Unknown device_type" error like that, even if the device is otherwise operating fine.

GonerGenesis, 2020-12-04:
The tablet screen is currently working as my 3rd screen and I'm able to draw pressure sensitive. Touch with real fingers doesn't work, but that should be correct for the Wacom One. So I don't really know which device is still missing.
```
xsetwacom --list dev
Wacom One Pen Display 13 Pen stylus	id: 11	type: STYLUS    
Wacom One Pen Display 13 Pen eraser	id: 12	type: ERASER
```
here is no display shown but it works 
[image]

jigpu, 2020-12-04:
Okay, in that case I'm pretty sure everything is fine. The failing interface has a very small descriptor so its probably used for something like firmware update or something else that has no impact on actually using the tablet. Your `xsetwacom` output also looks exactly like I would expect for that device; there's nothing missing that I can see. If you want to see a list of displays, you'd use the `xrandr` tool, not `xsetwacom`.

closed 2020-12-04 by jigpu