linuxwacom/xf86-input-wacom #154: Cannot remap Intuos S button to map event

lorddoskias, 2020-12-08:
I'm trying to bind my Intuos S pad 4th button to mouse even click 9, that's supposed to be one of the thumb buttons on a mouse. I have a mouse with such button and have configured that particular button to do something in MyPaint. However when I try to bind my intuous S 4th button (i.e index 8) to "9" via: 

`xsetwacom set "Wacom Intuos S Pad pad" Button 8 "9"` the output of` xsetwacom -s get "Wacom Intuos S Pad pad" all ` suggests that remapping is successful but no events are being sent: 

```
Property 'Wacom Tablet Area' does not exist on device.
xsetwacom set "Wacom Intuos S Pad pad" "Button" "1" "key +e -e "
xsetwacom set "Wacom Intuos S Pad pad" "Button" "2" "key +f -f "
xsetwacom set "Wacom Intuos S Pad pad" "Button" "3" "key +d -d "
xsetwacom set "Wacom Intuos S Pad pad" "Button" "8" "button +9 "
xsetwacom set "Wacom Intuos S Pad pad" "ToolDebugLevel" "0"
xsetwacom set "Wacom Intuos S Pad pad" "TabletDebugLevel" "0"
xsetwacom set "Wacom Intuos S Pad pad" "Suppress" "2"
xsetwacom set "Wacom Intuos S Pad pad" "RawSample" "4"
Property 'Wacom Pressurecurve' does not exist on device.
xsetwacom set "Wacom Intuos S Pad pad" "Mode" "Absolute"
Property 'Wacom Hover Click' does not exist on device.
xsetwacom set "Wacom Intuos S Pad pad" "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
xsetwacom set "Wacom Intuos S Pad pad" "Gesture" "off"
xsetwacom set "Wacom Intuos S Pad pad" "ZoomDistance" "0"
xsetwacom set "Wacom Intuos S Pad pad" "ScrollDistance" "0"
xsetwacom set "Wacom Intuos S Pad pad" "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Property 'Wacom Rotation' does not exist on device.
xsetwacom set "Wacom Intuos S Pad pad" "RelWheelUp" "1" "button +5 "
xsetwacom set "Wacom Intuos S Pad pad" "RelWheelDown" "2" "button +4 "
xsetwacom set "Wacom Intuos S Pad pad" "AbsWheelUp" "3" "button +4 "
xsetwacom set "Wacom Intuos S Pad pad" "AbsWheelDown" "4" "button +5 "
xsetwacom set "Wacom Intuos S Pad pad" "AbsWheel2Up" "5" "button +4 "
xsetwacom set "Wacom Intuos S Pad pad" "AbsWheel2Down" "6" "button +5 "
xsetwacom set "Wacom Intuos S Pad pad" "StripLeftUp" "1" "button +4 "
xsetwacom set "Wacom Intuos S Pad pad" "StripLeftDown" "2" "button +5 "
xsetwacom set "Wacom Intuos S Pad pad" "StripRightUp" "3" "button +4 "
xsetwacom set "Wacom Intuos S Pad pad" "StripRightDown" "4" "button +5 "
xsetwacom set "Wacom Intuos S Pad pad" "Threshold" "0"
xsetwacom set "Wacom Intuos S Pad pad" "BindToSerial" "0"
Property 'Wacom Pressure Recalibration' does not exist on device.
xsetwacom set "Wacom Intuos S Pad pad" "PanScrollThreshold" "13"
```

Nothing happens when I press the button. I also tried debugging this by running `xev -event button` and can confirm that when I tap the stylius or press one of its 2 buttons I get the expected events, however if I press the pad button while the pointer is in the area of xev I get no events. When I remap this button to a `"key e"` for example then it works as expected. I also tried remapping the key via `xinput set-button-map 19 1 2 3 4 5 6 7 9 9` as per this [blog post](https://wiki.mbirth.de/know-how/software/linux/remapping-mouse-buttons.html) to no avail.

I'm running ubuntu 20.04 with kernel 5.4.0-53-generic, libwacom is version 1.3-2ubuntu1, xf86-input-wacom is version 0.39-0 (default package in ubuntu 20.04).

jigpu, 2020-12-10:
If you're using the GNOME desktop, this is a known side-effect of how the desktop works. Rather than directly programming our driver (like xsetwacom does), GNOME instead intercepts the button events from our pad devices and re-writes them according to the control panel configuration. If a button isn't configured to do anything in the control panel, then GNOME swallows the event and prevents applications like xev from seeing it.

Basically, when you use xsetwacom to re-program one of the pad buttons to send a "button 9" event, GNOME is going to get first dibs on that event, see that the control panel doesn't have any configuration set for "button 9", and then do nothing with the event. The only reason that setting a key event (like `key e`) in xsetwacom works is that GNOME doesn't listen for key events from the pad.

In the past you could disable the GNOME integration if this side-effect was bothersome. Unfortunately this isn't the case anymore. I haven't tested it, but you could still get the behavior that you're looking for with a combination of the `xbindkeys` and `xdotool` applications. If you program the tablet button to send CTRL+SHIFT+F1, then you could add the following to `.xbindkeysrc` and (re)start the xbindkeys daemon. Afterwards, when you press the button you should trigger the keyboard shortcut, which will in turn trigger a click of mouse button 9. Convoluted, but appears to work on my computer...
~~~
"xdotool click 9"
    Control+Shift+F1
~~~

jigpu, 2021-07-21:
Closing due to inactivity.

closed 2021-07-21 by jigpu