linuxwacom/xf86-input-wacom #157: Stylus not recognized on Ubuntu 20.04

tormec, 2021-01-05:
Hi,
I have a Fujitsu T4220, whose stylus is not recognized under Ubuntu 20.04 (whereas it works well under Ubuntu 16.04 and 18.04).
I'm using a fresh installation of 20.04 on which I have run:
`sudo apt-get install build-essential autoconf linux-headers-$(uname -r)`

These are my packages:
```
uname -r

5.4.0-58-generic
```
```
apt list | grep libwacom

libwacom-bin/focal,now 1.3-2ubuntu1 amd64 [installed,automatic]
libwacom-bin/focal 1.3-2ubuntu1 i386
libwacom-common/focal,focal,now 1.3-2ubuntu1 all [installed]
libwacom-dev/focal,now 1.3-2ubuntu1 amd64 [installed]
libwacom-dev/focal 1.3-2ubuntu1 i386
libwacom2/focal,now 1.3-2ubuntu1 amd64 [installed,automatic]
libwacom2/focal 1.3-2ubuntu1 i386
```
```
apt list | grep xserver-xorg-input

xserver-xorg-input-all-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-all/focal,now 1:7.7+19ubuntu14 amd64 [installed,automatic]
xserver-xorg-input-evdev-dbg/focal 1:2.10.6-1 amd64
xserver-xorg-input-evdev-dev-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-evdev-dev/focal,focal 1:2.10.6-1 all
xserver-xorg-input-evdev-hwe-18.04-dbg/focal 3:14.5 amd64
xserver-xorg-input-evdev-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-evdev/focal 1:2.10.6-1 amd64
xserver-xorg-input-joystick-dev-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-joystick-dev/focal,focal 1:1.6.3-1build1 all
xserver-xorg-input-joystick-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-joystick/focal 1:1.6.3-1build1 amd64
xserver-xorg-input-kbd/focal 1:1.9.0-1build1 amd64
xserver-xorg-input-libinput-dev-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-libinput-dev/focal,focal 0.29.0-1 all
xserver-xorg-input-libinput-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-libinput/focal,now 0.29.0-1 amd64 [installed,automatic]
xserver-xorg-input-mouse/focal 1:1.9.3-1build1 amd64
xserver-xorg-input-mtrack-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-mtrack/focal 0.3.1-1build2 amd64
xserver-xorg-input-multitouch/focal 1.0~rc3-2 amd64
xserver-xorg-input-synaptics-dev-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-synaptics-dev/focal,focal 1.9.1-1ubuntu3 all
xserver-xorg-input-synaptics-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-synaptics/focal,now 1.9.1-1ubuntu3 amd64 [installed,automatic]
xserver-xorg-input-void/focal 1:1.4.1-1build3 amd64
xserver-xorg-input-wacom-hwe-18.04/focal 3:14.5 amd64
xserver-xorg-input-wacom/focal,now 1:0.39.0-0ubuntu1 amd64 [installed]
xserver-xorg-input-xwiimote/focal 0.5-1build3 amd64
```
And lastly:
```
xinput

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Fujitsu FUJ02E3                         	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Fujitsu FUJ02BF                         	id=9	[slave  keyboard (3)]
    ↳ Power Button                            	id=10	[slave  keyboard (3)]
    ↳ Fujitsu Application Panel buttons       	id=11	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=12	[slave  keyboard (3)]
```

tormec, 2021-05-19:
I was looking for a method to reload the stylus after boot up... and with a little googling I was able to resume the stylus after logout/login with this script

```
    #!/bin/sh
    # restart devices after suspension
    case "${1}" in
        resume|thaw)
            udevadm trigger --action=change
            ;;
    esac
```
which I called `10_reload_devices` saved in `/etc/pm/sleep.d` and made executable with `sudo chmod +x /etc/pm/sleep.d/10_reload_devices`.

The downside is that one has to re-login every time.

tormec, 2021-05-20:
Installing [input-wacom from source](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) solved this issue. 

Moreover, the stylus is recognized after suspension so that the previous script is no longer necessary.

tormec, 2021-05-20:
Maybe I've spoken too early!
At the second reboot, the stylus is no longer recognized.

tormec, 2021-05-21:
Thanks to [this post](https://ubuntuforums.org/showthread.php?t=1780154&page=36&p=11807798#post11807798) (reply 354), I think I managed to solve this issue.
**The cause:**
> The Distros including Ubuntu have added a new udev rule to enable the inputattach/wacom_w8001.ko combo which means they inadvertently disabled the normal driver providing out of the box support.

**The solution:**
in `/lib/udev/rules.d/60-inputattach.rules` (in earlier version of ubuntu it was `40-`) comment the line:
```SUBSYSTEM=="tty", KERNEL=="ttyS[0-9]*", ATTRS{id}=="FUJ02e5", ACTION=="add|change", RUN+="/usr/bin/inputattach --daemon --baud 19200 --w8001 /dev/%k"```

closed 2021-05-21 by tormec

superscotty19, 2023-11-12:
I've got the exact same tablet (T4220) and commenting does not help - still nothing on the stylus, but it did work in Windows. I know it's on ttyS4 because I did a test in the terminal (can't remember the steps) and I got (garbled) text feedback when I moved the stylus. I tried to go through all 40+ pages of the forum link posted and still nothing. Any other ideas?

tormec, 2023-11-12:
> I've got the exact same tablet (T4220) and commenting does not help

Unfortunately, that solution works with Ubuntu 20.04 but not with 22.04

whot, 2023-11-12:
Make sure inputattach isn't running:
```
$ sudo killall inputattach
```
Then check if the kernel handles this device anyway using 
```
$ sudo libinput record
$ sudo libinput debug-events
````
The former records kernel events but they can be hard to interpret, debug-events is a bit easier since those are the logical events parsed by libinput. If you don't see your device in `record`, don't bother with `debug-events`.

If the device works, then inputattach interferes somehow and we need some update to make sure it doesn't trigger for this particular device.

If the device does not work, then run `sudo inputattach --baud XYZ --w8001 /dev/ttyS4` manually and repeat the `record`/`debug-events` step. Test the different baud rates and other options inputattach has (`--help` should show them) until it works.

If nothing works, this device has a different protocol that's not yet supported though the chances of this happening for a ~16y old device is quite low.