linuxwacom/xf86-input-wacom #158: How to use right click in TouchPad mode for a Bamboo Wacom 2FG?

ralexrdz, 2021-01-14:
I can't find a way to either set one of the 4 side buttons or two finger tap for a rightclick action

here is my `xinput list-props 18` for the wacom touch

```
$ xsetwacom --version
0.39.0
$ xinput list-props 18
Device 'Wacom Bamboo 2FG 4x5 Finger touch':
	Device Enabled (165):	1
	Coordinate Transformation Matrix (167):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (298):	0
	Device Accel Constant Deceleration (299):	1.000000
	Device Accel Adaptive Deceleration (300):	1.000000
	Device Accel Velocity Scaling (301):	10.000000
	Device Node (290):	"/dev/input/event17"
	Wacom Tablet Area (748):	0, 0, 15360, 10240
	Wacom Rotation (749):	0
	Wacom Pressurecurve (750):	0, 0, 100, 100
	Wacom Serial IDs (470):	209, 1, 3, 0, 0
	Wacom Serial ID binding (751):	0
	Wacom Pressure Threshold (753):	0
	Wacom Sample and Suppress (754):	2, 4
	Wacom Enable Touch (755):	1
	Wacom Enable Touch Gesture (757):	1
	Wacom Touch Gesture Parameters (758):	832, 230, 250
	Wacom Tool Type (475):	"TOUCH" (474)
	Wacom Button Actions (759):	"Wacom button action 0" (760), "Wacom button action 1" (761), "Wacom button action 2" (762), "None" (0), "None" (0), "None" (0), "None" (0), "Wacom button action 3" (763)
	Wacom button action 0 (760):	1572865
	Wacom button action 1 (761):	1572866
	Wacom button action 2 (762):	1572867
	Wacom button action 3 (763):	1572872
	Wacom Panscroll Threshold (765):	1300
	Device Product ID (291):	1386, 209
	Wacom Debug Levels (766):	0, 0

```

using the `xev --event button` i wasn't able to produce a button 3 action using the touchpad

jigpu, 2021-01-25:
Are you using a GNOME-based environment (e.g. GNOME, Unity, Cinnamon, etc.)? If so, this would probably explain why you can't configure the pad to produce a right-click event. GNOME absorbs any mouse button events from our tablet to make its own tablet configuration code work, which means that even if you use xsetwacom to configure a right-click action, that event won't make its way through to applications. GNOME also does not have a way to set a right-click action from its tablet configuration tool, which makes this limitation even more frustrating...

If you're running some other environment (e.g. KDE, Mate, etc.) you should be able to run `xsetwacom set "Wacom Bamboo 2FG 4x5 Pad pad" button <button_id> button 3`, replacing `<button_id>` with either 1, 2, 3, or 8 depending on which button you want to program.

As for right-clicking with the touchpad, that should work fine without any configuration. The gesture our driver uses for right-click is a little different than some other drivers, so you'll need to make sure you're doing what ours expects. To perform a right-click, place one finger on the touchpad and then tap with a second finger (keeping the first in contact at all times). A right-click event should be immediately produced.

jigpu, 2021-07-21:
Closing due to inactivity.

closed 2021-07-21 by jigpu