linuxwacom/xf86-input-wacom #163: status LED is not accessible from xsetwacom

oxpa, 2021-02-08:
Hello.

I'm using Wacom PTH-651 (Intuos pro M) tablet. It has a sensor ring and 9  buttons one of which is in the center of the ring.
There are also 4 LEDs around the sensor ring. The button in the middle of the sensor ring toggles LEDs clockwise so that only one LED is active. So there are 4 states of the tablet and only 1 is active at the moment. This state is usually used to control tablet settings.
It is possible to know which LED is active using `/wacom_led/status_led0_select` contents for a device.

I've found an old thread https://sourceforge.net/p/linuxwacom/mailman/linuxwacom-devel/thread/20120123044314.GA28037%40yabbi.bne.redhat.com/#msg28714184 but it seems like the patch has never been accepted.

I couldn't find a way to get status LED state using xsetwacom program while it feels right to have it.
Feel free to point me to proper options if I missed them.

jigpu, 2021-02-10:
Your investigation is correct; this feature was never added to xsetwacom. The only method available to read the LED status is through sysfs. We could potentially add the feature into xsetwacom, but many of the questions about how to best represent them still remain. We have a bit more insight now that there are several devices with LEDs, but we'd still have to do some thinking. Is there a specific purpose you have in mind that makes access through xsetwacom better than sysfs?

oxpa, 2021-02-10:
I'm using [wacom-gui](https://github.com/tb2097/wacom-gui) to set up my tablet. It's a nice tool which has almost everything the native wacom application has. The application itself relies heavily on xsetwacom to query data from a tablet and set it's state.

Major feature wacom-gui lacks is profile switching. The native application can change tablet behavior based on LED state (so in case of my tablet it's not an LED state but a tablet state indicated by LED but never mind). wacom-gui already has profiles built-in and is easily patched to switch those profiles. And I even patched wacom-gui and wrote a simple shell script to switch profiles based on the LED state. But my approach relies on the fact that there is a single tablet in the system which is obviously wrong for general use. 

As everything else about a tablet is known from `xsetwacom --get ....` and the application (wacom-gui) itself is quite high level it seems natural to get LED state using the same xsetwacom. And switch profiles based on that info. It will also guarantee that whole thing works with multiple tablets.

And, well, in general, I'd say it's a bad design for a high level tool to go and touch one thing inside sysfs when everything else is done through a specialized low level tool.

So having LED state accessible for reading from xsetwacom would make other application(s) better ;) It may sound somehow funny and strange but, well, it is how it is.

Hope this answers your questions.

Regards,
Oxpa

jigpu, 2021-02-24:
Thanks for the detailed response! Unfortunately, after doing some thinking I'm afraid I'm moving more toward having LED control be done through a separate tool...

I can understand wanting to use xsetwacom as a "Swiss Army Knife" for interacting with the tablet, but that isn't its actual scope. It is a tool for statically querying / modifying the X11 driver state. It is able to communicate that state to the user (or from the user) in a format that is more understandable than the raw numbers you'd deal with when using `xinput --list-props` and `xinput --set-prop`. Tablet features which exist outside of this scope should be handled by other tools when possible. For example, using `upower` to get the state of a Bluetooth tablet's battery or `i4oled` to program the OLEDs on an Intuos4. The LEDs are similar: they aren't part of the X11 driver state and so shouldn't be exposed through xsetwacom.

It isn't "bad design" for wacom-gui to interact with multiple tools in order to achieve its goals. If anything, its a necessity in the UNIX world where command-line tools are expected to be modular. The xsetwacom command may do a lot of things, but it will never do it all. At some point that means wacom-gui is going to have to rely on other programs or custom code to get things done.

Now, admittedly, we don't have a command-line tool for interacting with the LEDs. The raw sysfs interface isn't exactly hard to work with (especially in the context of a Python program which already has hooks into libwacom), but it still might be a small tool that is worth creating. Making it a standalone tool also means that it will work fine under Wayland and other non-X environments.

whot, 2022-02-17:
Closing this, see https://github.com/linuxwacom/xf86-input-wacom/issues/163#issuecomment-785204677. We'd welcome a PR for a tool to expose the LED, but xsetwacom is not the best place for that.

closed 2022-02-17 by whot