linuxwacom/xf86-input-wacom #164: compiling on fedora for android x86 or are there compiliation tools for android x86?

StAletheus, 2021-02-16:
i've gotten the driver to compile and make, but i havent installed because i wanted to take the binary and put it in its proper system directory. this may not be the place for this. could someone direct me to the proper forum if not? (ie this may not be classified as an issue and may need to be addressed on a sourceforge disccussion forum, but i wanted to hold off on having to make an account unless necessary....getting this working IS a necessity though)

jigpu, 2021-02-16:
Android does not (and cannot!) use the xf86-input-wacom driver. To be clear, are you trying to compile this **xf86-input-wacom** X11 driver for Android, or are you trying to compile the (similarly named) **input-wacom** kernel driver?

StAletheus, 2021-02-16:
i was directed by [https://linuxwacom-discuss.narkive.com/sCsq5OO1/wacom-drivers-for-android-x86](this) here..well, to sourceforge.

jigpu, 2021-02-16:
We've shut down our old Sourceforge site and moved all discussion to here on Github.

The link you've provided is about compiling the (input-wacom) kernel driver. Please open a new issue at https://github.com/linuxwacom/input-wacom/issues/ where I can follow up with further instruction.

Apologies for any confusion...

closed 2021-02-16 by jigpu

StAletheus, 2021-02-16:
great thank you! sorry to clutter up this discussion.