linuxwacom/xf86-input-wacom #165: Inconsistent right click trigger in touch mode and not in accordance to the docs

heavyinfo, 2021-02-16:
Device : Intuos Pen & Touch small CTH-480

OS: Arch Linux 5.1015-arch1-1, KDE Plasma with Wacom Tablet Settings

Installed: input-wacom(included in kernel),  xf86-input-wacom-0.40.0-1, libwacom-1.8-1

```
xsetwacom list devices

Wacom Intuos PT S Pen stylus    	id: 13	type: STYLUS
Wacom Intuos PT S Pen eraser    	id: 14	type: ERASER
Wacom Intuos PT S Pad pad       	id: 15	type: PAD
Wacom Intuos PT S Finger touch  	id: 21	type: TOUCH
```

Right click on tablet touch mode is triggered randomly when double tapped with two (or more) fingers several times. 

I can confidently say that the behavior is not in accordance to supposed right click trigger mentioned earlier in other issues  

> As for right-clicking with the touchpad, that should work fine without any configuration. The gesture our driver uses for right-click is a little different than some other drivers, so you'll need to make sure you're doing what ours expects. To perform a right-click, place one finger on the touchpad and then tap with a second finger (keeping the first in contact at all times). A right-click event should be immediately produced.

Keeping one finger on the tablet and tapping using other finger doesn't trigger anything.

I tried changing Tap Time (100) in KDE Wacom settings but it didn't resolve this issue, so right now I have to double tap several times for a right click hoping to get lucky. No right click issues with Pen(Stylus) button 3.

Let me know if any further information is needed.

P.S.
It goes without saying that everyone who is using wacom tablet with Linux appreciates the existence of this project,  I appreciate everyone involved with this project too.

jigpu, 2021-02-23:
I've been testing this out and have found that right-click is indeed much more difficult to trigger than it should be. The instructions quoted do work, but the reliability is poor. Adjusting configuration options (increasing Tap Time and increasing Suppress) does not help much. Carefully performing the gesture (i.e. making sure the pointer does not move and making sure the 2nd-finger tap takes less than Tap Time milliseconds to complete) might slightly improve success rate, but I have to wonder if its just placebo...

These tests were with a 2nd-gen Intuos Pro L on Arch Linux with kernel 5.10.16 and xf86-input-wacom 0.40.0 under KDE Plasma. Similar results are expected in other situations since we have not touched the gesture code in quite a while...

heavyinfo, 2021-02-25:
Thanks for testing and acknowledging the issue. I was using the touch mode for mouse functions due to accessibility reasons, I have moved to mouse temporarily as not being able to trigger right click on-demand was affecting my flow.

> making sure the pointer does not move and making sure the 2nd-finger tap takes less than Tap Time milliseconds to complete

Since it's nearly impossible to keep the pointer still and that it's main cause of issue can we do away with it altogether? Like just using two finger tap detection to trigger right click?

jigpu, 2021-03-10:
The driver already allows a small amount of pointer motion to occur during the gesture, but since I cannot trigger the gesture even under "perfect" conditions (no pointer movement and an artificially long tap time) I suspect that the issue is more fundamental. Changing the kind of gesture to two-finger touch is unlikely to actually fix the bug.

I did more digging around and realized this was reported earlier this year as #103. It suggests that version 0.37.0 has functional right-click gestures -- I would suggest downgrading to it and seeing if your situation improves. Instructions for building and installing the driver from source can be found at https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver

heavyinfo, 2021-04-01:
Thanks for taking the time to verify the issue. Right now I'm using the tablet with just the pen, I will try older drivers later to see if it works fine.

Pinglinux, 2021-09-02:
Close this issue. It will be considered and resolved by #103.

closed 2021-09-02 by Pinglinux