linuxwacom/xf86-input-wacom #166: Cintiq Pro 32 - xsetwacom wacom devices are not a wacom device

Dwood15, 2021-03-02:
When I run 

`xsetwacom --list --verbose`

I get: 
```
... Found device 'Wacom Cintiq Pro 32 Pen' (21).
... 'Wacom Cintiq Pro 32 Pen' (21) is not a wacom device.
... Found device 'Wacom Cintiq Pro 32 Touch Finger' (22).
... 'Wacom Cintiq Pro 32 Touch Finger' (22) is not a wacom device.
... Found device 'Wacom Cintiq Pro 32 Pen Pen (0x8a8018a1)' (23).
... 'Wacom Cintiq Pro 32 Pen Pen (0x8a8018a1)' (23) is not a wacom device.
... Found device 'Wacom Cintiq Pro 32 Pen Eraser (0x8a8018a1)' (24).
... 'Wacom Cintiq Pro 32 Pen Eraser (0x8a8018a1)' (24) is not a wacom device.
```

xinput manages to find these just fine:

[image]

When I run `lsusb | grep -i wacom`
I get: 

[image]

`xsetwacom --version` reports `0.40.0`
`xinput --version` reports `1.6.3`

`
[image]

`libwacom-list-local-devices` 

returns: 
```# Device node: /dev/input/event23
[Device]
Name=Wacom Cintiq Pro 32
ModelName=DTH-3220
DeviceMatch=usb:056a:0352;
PairedID=usb:056a:0356;
Class=Cintiq
Width=27
Height=15
IntegratedIn=Display;
Styli=0x842;0x84a;0x80842;0x100802;0x100804;0x10080a;0x10080c;0x100842;0x10084a;0x100902;0x10090a;0x120802;0x140802;0x14080a;0x160802;0x16080a;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=0

---------------------------------------------------------------
# Device node: /dev/input/event22
[Device]
Name=Wacom Cintiq Pro 32
ModelName=DTH-3220
DeviceMatch=usb:056a:0352;
PairedID=usb:056a:0356;
Class=Cintiq
Width=27
Height=15
IntegratedIn=Display;
Styli=0x842;0x84a;0x80842;0x100802;0x100804;0x10080a;0x10080c;0x100842;0x10084a;0x100902;0x10090a;0x120802;0x140802;0x14080a;0x160802;0x16080a;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=0

---------------------------------------------------------------
# Device node: /dev/input/event21
[Device]
Name=Wacom ExpressKey Remote
ModelName=ACK-411050
DeviceMatch=usb:056a:0331;
Class=Remote
[...]```

Additionally, MapToOutput requires me to take my pen and interact with the screen before I can use it. I would have thought that the default for a tablet with an integrated display would default to the tablet mapping directly to the display. 

I've done a lot of wiki and stackoverflow browsing and haven't found anyone with these issues. Guess not a lot of people have Cintiqs. 

The OS I'm running is arch, on kernel ver. 5.10.11-arch1-1, driver version: `v2.00-0.47.0`

whot, 2021-03-03:
looks like the devices are handled by the xf86-input-libinput driver, most likely because xf86-input-wacom isn't installed (since it *should* take precedence). Installing it and restarting X should fix the issue.

Dwood15, 2021-03-08:
> looks like the devices are handled by the xf86-input-libinput driver, most likely because xf86-input-wacom isn't installed (since it _should_ take precedence). Installing it and restarting X should fix the issue.

What this turned out to be was a broken GLIBC from a `pacman -Syu` that had been forced closed before it could finish. The driver simply couldn't load, so was forced to fall back on the xf86 input driver.

After reinstalling arch (whee) everything works again and this issue is no longer replicating!

closed 2021-03-08 by Dwood15