linuxwacom/xf86-input-wacom #167: Finetip not working on Intuos Pro

Cat-Ion, 2021-03-03:
I'm running current master of input-wacom and xf86-input-wacom.  The regular stylus is working as expected, but when I try using the finetip instead, the LED on the touch ring turns green, and then blue when the tip is touched. It does not turn green again afterwards. This seems to disable the touch input. Using the regular stylus again reverts all of this and lets me use the tablet again. Is this behaviour known or expected, or did I misconfigure something?

I've tried setting all the tools to maximum debug mode, but do not see anything in the logs when using the finetip.

Here's my sysinfo: [sysinfo.qcuRT44vLr.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/6078255/sysinfo.qcuRT44vLr.tar.gz)

Cat-Ion, 2021-03-03:
Looking at the logs some more, I can see this:

    [ 18068.526] (II) Using input driver 'wacom' for 'Wacom Intuos Pro M Pen'
    ...
    [ 18068.578] (II) Wacom Intuos Pro M Pen stylus: hotplugging completed.
    [ 18068.595] (II) XINPUT: Adding extended input device "Wacom Intuos Pro M Pen stylus" (type: STYLUS, id 9)
    ....
    [ 18068.647] (II) config/udev: Adding input device Wacom Intuos Pro M Pen (/dev/input/mouse0)
    [ 18068.647] (**) Wacom Intuos Pro M Pen: Applying InputClass "Tablet"
    [ 18068.647] (II) Using input driver 'wacom' for 'Wacom Intuos Pro M Pen'
    [ 18068.647] (**) Wacom Intuos Pro M Pen: always reports core events
    [ 18068.647] (EE) Wacom Intuos Pro M Pen: usbDetect: can not ioctl version
    [ 18068.647] (EE) Wacom Intuos Pro M Pen: cannot identify device class.
    [ 18068.668] (EE) PreInit returned 8 for "Wacom Intuos Pro M Pen"
    [ 18068.668] (II) UnloadModule: "wacom"

I had similar behaviour (but for all inputs) with an older version of the kernel driver. Could it be that the kernel driver does not like the pen for some reason?

whot, 2021-03-03:
The last set of lines you can ignore, `/dev/input/mouse*` has been deprecated for over a decade and won't work with any driver but `xf86-input-mouse`. This indicates an error in your `xorg.conf.d` snippets, the wacom driver shouldn't be assigned to those event nodes anyway. Either way, it's a false positive, these lines have no effect on the issue.

jigpu, 2021-03-04:
Its been a long time since I've used the Finetip pen, but I believe this is normal behavior. When you use the Finetip pen the tablet switches into its "paper mode" and begins recording the strokes to its internal memory. You can press the button inside the touchring (the one with the LED) to make new pages and layers. See https://www.youtube.com/embed/UPG0-8hcrCw for more information. When the tablet is in paper mode I don't believe it responds to touch or other input.

The "Tuhi" application (https://github.com/tuhiproject/tuhi) can be used to download drawings from the internal memory.

closed 2021-07-21 by jigpu