linuxwacom/xf86-input-wacom #168: SAMSUNG Chrombook Pro stylus not recognized

waitlamp, 2021-03-13:
Hello,
I have filed an issue in wacom-hid-descriptors [#147](https://github.com/linuxwacom/wacom-hid-descriptors/issues/147)
On ubuntu 20.10, pressure can use, 
but xsetwacom cannot detect any wacom devices, GNOME either.
Is there any way to use xf86-input-wacom driver? 
thanks. 

here is sysinfo
[sysinfo.UwjeIiKZUM.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/6129840/sysinfo.UwjeIiKZUM.tar.gz)

whot, 2021-03-15:
```
$ cat > /etc/X11/xorg.conf.d/99-wacom-driver.conf <<EOF
Section "InputClass"
        Identifier "Wacom USB tablet class"
        MatchUSBID "2d1f:*"
        MatchDevicePath "/dev/input/event*"
        MatchIsTablet "true"
        Driver "wacom"
EndSection
EOF
```
should do the trick, based on the USB id from the wacom-hid-descriptor bug

waitlamp, 2021-03-15:
It work.
`xsetwacom --list` now list four devices. (some devices' name are same.)
```
$ xsetwacom --list
ACPI0C50:00 2D1F:0163 stylus    	id: 11	type: STYLUS    
ACPI0C50:00 2D1F:0163 stylus    	id: 12	type: STYLUS    
ACPI0C50:00 2D1F:0163 eraser    	id: 14	type: ERASER    
ACPI0C50:00 2D1F:0163 eraser    	id: 15	type: ERASER    
```
but GNOME setting still not recognized the stylus.
in Krita, Tablet Tester detected the Pen tip brought near / taken away 
~~but no pressure and X-Y axis , and when drawing, pressure sensitive work~~ 
Add: I have tried many times and a few times I could make X-Y axis and pressure level recognized.
(in most cases, only Pen tip brought near / taken away can be recognized)
And I tried to use touch/mouse and stylus at same time, pen tip caused mouse drifting, also moved mouse X-Y axis.
It seems pen still recognized as mouse when pen tip drawing on screen.

Then, I tried to use `xsetwacom set "11" "Mode" "Absolute"`
(both 12 / 14 / 15, devices have same devices name, I have to use ID instead of name.) 
but problem still exist.
Is a krita problem？

whot, 2021-03-18:
> but GNOME setting still not recognized the stylus.

that is primarily defined by the device being listed by `libwacom-list-local-devices`, and more specifically: it requires a `.tablet` file to be added to libwacom for any new devices.

`libinput`  has a `libinput debug-tablet` tool which I recommend to run to make sure the tablet output itself is sane. If those sliders work correctly, then at least we can rule out the kernel.

`xinput test-xi2` is an x-based test tool, watch the output as you interact with the window, you should see the valuator values change with the various movements (valuator 0,1 are x,y, 2 is usually pressure, etc.)

If that looks sane, the issue is most likely in Krita, yes.

waitlamp, 2021-06-20:
[image]

after added `.tablet` file, tablet can be detected
It work properly.
but the pen was recognized as AES Pen?

closed 2021-06-21 by waitlamp

whot, 2021-06-21:
cosmetic issue, it really doesn't matter. we just group the AES pens together but unless there's a button missing or something, it just doesn't matter