linuxwacom/xf86-input-wacom #170: Problem with button mapping on Wacom BambooFun CTE-450

elementgreen, 2021-03-31:
I'm not sure if this is the right project to file this bug report on, since I'm still becoming familiar with all the components which are part of the Wacom input subsystem.

OS: Ubuntu Linux 20.10
Desktop: Gnome
Drawing Tablet: Wacom Bamboo Fun CTE-450 (Small)

Problem:
When accessing the Wacom Tablet section of the Gnome Settings application and accessing the Map Buttons interface on the Tablet tab, the proper diagram is shown for this tablet (/usr/share/libwacom/bamboo-0fg-s-p-alt.svg).  However, the buttons don't seem to map correctly when pressing them.  Here is a list of what button is pressed followed by which one is activated on the screen interface:
< (A) => FN2
> (B) =>  NONE
FN1 (C) =>  <
FN2 (D) =>  FN1

The ring seems to work as expected though.

When dumping libinput events for the Pad device:
libinput debug-events /dev/input/event3
-event3   DEVICE_ADDED     Wacom BambooFun 4x5 Pad           seat0 default group1  cap:P buttons:4 strips:0 rings:1 mode groups:1
 event3   TABLET_PAD_BUTTON +0.000s	  0 pressed (mode 0)
 event3   TABLET_PAD_BUTTON +0.156s	  0 released (mode 0)
 event3   TABLET_PAD_BUTTON +1.812s	  1 pressed (mode 0)
 event3   TABLET_PAD_BUTTON +1.956s	  1 released (mode 0)
 event3   TABLET_PAD_BUTTON +3.508s	  2 pressed (mode 0)
 event3   TABLET_PAD_BUTTON +3.652s	  2 released (mode 0)
 event3   TABLET_PAD_BUTTON +8.016s	  3 pressed (mode 0)
 event3   TABLET_PAD_BUTTON +8.084s	  3 released (mode 0)

The above was output when pressing the buttons in A, B, C, D order (<, >, FN1, FN2).

When running xinput to query the state, the following is output (extracted relevant ButtonClass section):
...
ButtonClass
	button[1]=up
	button[2]=up
	button[3]=up
	button[4]=up
	button[5]=up
	button[6]=up
	button[7]=up
	button[8]=up
	button[9]=up
...

Pressing and holding A, B, C, D in order and running the above command for each, shows the following sequence:
< (A) => button[8]=down
> (B) => button[9]=down
FN1 (C) => button[1]=down
FN2 (D) =>  button[3]=down

It seems strange that xinput lists 9 buttons (there are only 4 and the ring is treated as a separate class).  Also the sequence 8, 9, 1, 3 is also rather strange.  xinput get-button-map shows a normal mapping (1 2 3 4 5 6 7 8 9).

The /usr/share/libwacom/bamboo-0fg-s-p-alt.tablet file seemed like a potential candidate and has the following lines related to the buttons:
[Buttons]
Top=A;B;C;D
EvdevCodes=0x116;0x115;0x110;0x111

I'm not sure what the EvdevCodes are though, so I'm unable to verify if these are perhaps incorrect.

Any tips on where the issue might be for the incorrect button mapping would be most appreciated.

whot, 2021-04-07:
> It seems strange that xinput lists 9 buttons (there are only 4 and the ring is treated as a separate class). Also the sequence 8, 9, 1, 3 is also rather strange. xinput get-button-map shows a normal mapping (1 2 3 4 5 6 7 8 9).

X button codes 4-7 are reserved for wheel clicks, so any 4-button device has at least buttons 1-8. In this case the buttons seem to map to left/right and 8/9, middle (button 2) is unused.

Anyway, judging by the libinput debug-events output the button order is correct-enough, the mapping issue you're seeing seems to be an issue with the libwacom SVG file which GNOME relies on to highlight the various bits. Check out the [README](https://github.com/linuxwacom/libwacom/blob/master/data/layouts/README.md) and see if the SVG matches all those requirements.

jigpu, 2021-07-21:
Smells like an ancient GNOME bug that affects all Bamboo devices. See, for example, my comment at https://bugzilla.gnome.org/show_bug.cgi?id=770464#c5 for some background. They haven't fixed it yet, to my knowledge. Might be worth filing an issue on their new bug tracker (https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues) since the old Bugzilla has been shutdown.

FYI, I wrote a script to at https://gist.github.com/jigpu/fde784840ba5731726e7382952b0d5ed which will help with configuring the driver to send events that GNOME can handle.

closed 2021-07-21 by jigpu