linuxwacom/xf86-input-wacom #172: Wacom Co., Ltd PTZ-630 [Intuos3 (6x8)] shows up as a "mouse" in Google Chrome.

bobajeff, 2021-04-13:
I've been testing the `PointerEvent` interface browser have. Here's code I used to test it:
https://github.com/bobajeff/test_pointer_events 

# Doesn't work with linuxwacom:
On _Chrome/Firefox_ on _Linux_ It's only showing up a `mouse` `pointerType`. (It's also not picking up any pressure or tilt either. I'm not sure if that simply blocked by it being seen as mouse though.).
[image]

# Works with Wacom's Windows Driver:
On _Chome_ on _Windows_ with Wacom's drivers it shows up correctly as a `pen` and I get _Pressure_ and _Tilt_ data. _(Window's version of Firefox for some reason forces emulation touch events)._

[image]

----
# Note:

I'm not sure if this is a driver issue or browser issue. [I've also test the _wacom_ tablet with the _WebHID API_](https://stackoverflow.com/questions/67013159/webhid-api-how-do-i-parse-data-from-inputreport-event) and it seems to be reporting incorrect HID data on the Linux version of Chrome while the Windows version seems to be reporting correct HID data. So that leads me to think it could be related to driver settings or something.

However, everything seem to work fine in Krita.

Here's my system info:
linuxwacom/wacom-hid-descriptors#152

whot, 2021-04-15:
tbh, I don't know how those tools work but: if it works fine in Krita that indicates that the driver works fine. After all, the driver has no idea whether Krita or any other client is running, it sends the events regardless.

The lowest-level testing tool under X is probably `xinput test-xi2` which should show all axes as they are sent to the clients. If you see the axes there, then the issue is with Chrome.

bobajeff, 2021-04-17:
Yep, just tested it. It does report all the sensor data.

Shows up like:
```log
EVENT type 17 (RawMotion)
    device: 2 (11)
    detail: 0
    flags: 
    valuators:
          0: 13765.00 (13765.00)
          1: 15398.00 (15398.00)
          2: 0.00 (0.00)
          3: 26.00 (26.00)
          4: 36.00 (36.00)
          5: -900.00 (-900.00)
```
[wacom-xinput.log](https://github.com/linuxwacom/xf86-input-wacom/files/6329492/wacom-xinput.log)

The Linux version of Chrome/Firefox seem to be at fault here.

closed 2021-04-17 by bobajeff