linuxwacom/xf86-input-wacom #173: Wacom IDs change

zoliky, 2021-04-19:
Wacom Model: Wacom Intuos S USB
Model Number: CTL-4100/K0-BX
Linux: Fedora Workstation 33 with GNOME on Xorg

Right after GNOME starts:

`$ xsetwacom --list devices | grep "Wacom Intuos"

Wacom Intuos S Pen stylus       	id: 13	type: STYLUS    
Wacom Intuos S Pad pad          	id: 14	type: PAD       
Wacom Intuos S Pen eraser       	id: 22	type: ERASER    
Wacom Intuos S Pen cursor       	id: 23	type: CURSOR
`

30 seconds later:

`$ xsetwacom --list devices | grep "Wacom Intuos"

Wacom Intuos S Pen stylus       	id: 13	type: STYLUS    
Wacom Intuos S Pen eraser       	id: 14	type: ERASER    
Wacom Intuos S Pen cursor       	id: 22	type: CURSOR    
Wacom Intuos S Pad pad          	id: 23	type: PAD 
`

Notice that the order of the "Pen eraser", "Pen cursor", and "Pad pad" change and they get different IDs as well.

**It leads to the following problem:**

I have a script which fires the following two command when GNOME starts so srolling works:

```
$ xsetwacom --set "Wacom Intuos S Pen stylus" Button 2 "pan"
$ xsetwacom --set "Wacom Intuos S Pen stylus" "PanScrollThreshold" 200
```

However after 30 seconds the scrolling feature stops and I need to issue the commands again manually using a terminal. That is happening because as you can see above the IDs changed,  even though it is not the "Stylus" which changed `xsetwacom` seems to reset everything.

After that 30 second everything stays permanent until reboot.

Any suggestion on how to make the Wacom allocate the same IDs right from the start?

whot, 2021-04-19:
> Any suggestion on how to make the Wacom allocate the same IDs right from the beginning?

not really possible since we cannot usually know ahead of time how many devices we will need, and even then - the driver has no way of doing so, IDs are a server-private implementation and the driver cannot guarantee it'll get the same ID.

The issue here is that you're relying on a command that is not compatible with hotplugged devices (which the X server has now supported since 2006 or so). You need to hook up your script to *monitor* devices being removed and re-added (that is the only time they can change IDs) and re-apply the settings. Short of that, your `sleep 30` is indeed the simplest solution.

Why this particular removal happens I don't know but if you attach the journal log from such an ID switch, it may be clearer.

zoliky, 2021-04-19:
```
Apr 20 02:03:45 zoliky.net gnome-software[2438]: enabled plugins: desktop-categories, fwupd, os-release, packagekit, packagekit-local, packagekit-offline, packagekit-proxy, packagekit-refresh, packagekit-upgrade, packagekit-url-to-app, appstream, fedora-pkgdb-collections, desktop-menu-path, fedora-langpacks, flatpak, hardcoded-blocklist, hardcoded-popular, modalias, packagekit-refine, rewrite-resource, odrs, packagekit-history, provenance, repos, systemd-updates, generic-updates, packagekit-refine-repos, provenance-license, icons, key-colors, key-colors-metadata
Apr 20 02:03:45 zoliky.net gnome-software[2438]: disabled plugins: dummy
Apr 20 02:03:45 zoliky.net gnome-software[2438]: Theme parsing error: gtk.css:2722:25: 'gtkopacity' is not a valid color name
Apr 20 02:03:45 zoliky.net gnome-software[2438]: Theme parsing error: gtk.css:2732:15: Junk at end of value for padding
Apr 20 02:03:45 zoliky.net gnome-software[2438]: Theme parsing error: gtk.css:2791:13: Junk at end of value for padding
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (EE) Wacom Intuos S Pad pad: Error reading wacom device : No such device
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) config/udev: removing device Wacom Intuos S Pad pad
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) UnloadModule: "wacom"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) systemd-logind: releasing fd for 13:71
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (EE) Wacom Intuos S Pen cursor: Error reading wacom device : No such device
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) config/udev: removing device Wacom Intuos S Pen stylus
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) UnloadModule: "wacom"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) systemd-logind: not releasing fd for 13:70, still in use
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) config/udev: removing device Wacom Intuos S Pen eraser
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) UnloadModule: "wacom"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) systemd-logind: not releasing fd for 13:70, still in use
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) config/udev: removing device Wacom Intuos S Pen cursor
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) UnloadModule: "wacom"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) systemd-logind: releasing fd for 13:70
Apr 20 02:03:45 zoliky.net kernel: input: Wacom Intuos S Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4.2/3-4.2:1.0/0003:056A:0374.0006/input/input29
Apr 20 02:03:45 zoliky.net kernel: input: Wacom Intuos S Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4.2/3-4.2:1.0/0003:056A:0374.0006/input/input31
Apr 20 02:03:45 zoliky.net kernel: wacom 0003:056A:0374.0006: hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos S] on usb-0000:00:14.0-4.2/input0
Apr 20 02:03:45 zoliky.net gnome-software[2438]: failed to set proxy: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._pk_2dengine_2derror_2dquark.Code3: setting the proxy failed: failed to get the uid
Apr 20 02:03:45 zoliky.net gnome-software[2438]: failed to set proxies: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._pk_2dengine_2derror_2dquark.Code3: setting the proxy failed: failed to get the uid
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) config/udev: Adding input device Wacom Intuos S Pen (/dev/input/mouse1)
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) No input driver specified, ignoring this device.
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) This device may have been added with another device file.
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) config/udev: Adding input device Wacom Intuos S Pad (/dev/input/js0)
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) No input driver specified, ignoring this device.
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) This device may have been added with another device file.
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) config/udev: Adding input device Wacom Intuos S Pen (/dev/input/event6)
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (**) Wacom Intuos S Pen: Applying InputClass "evdev tablet catchall"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (**) Wacom Intuos S Pen: Applying InputClass "libinput tablet catchall"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (**) Wacom Intuos S Pen: Applying InputClass "Wacom USB tablet class"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (**) Wacom Intuos S Pen: Applying InputClass "Wacom tablet class"
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) Using input driver 'wacom' for 'Wacom Intuos S Pen'
Apr 20 02:03:45 zoliky.net upowerd[722]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4.2/3-4.2:1.0/0003:056A:0374.0006/power_supply/wacom_battery_1
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (II) systemd-logind: got fd for /dev/input/event6 13:70 fd 35 paused 0
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (**) Wacom Intuos S Pen: always reports core events
Apr 20 02:03:45 zoliky.net /usr/libexec/gdm-x-session[1662]: (**) Option "Device" "/dev/input/event6"
[... cut ...]
Apr 20 02:03:49 zoliky.net audit: BPF prog-id=45 op=UNLOAD
Apr 20 02:03:49 zoliky.net systemd[1]: systemd-hostnamed.service: Succeeded.
Apr 20 02:03:49 zoliky.net audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Apr 20 02:03:49 zoliky.net audit: BPF prog-id=40 op=UNLOAD
Apr 20 02:03:49 zoliky.net audit: BPF prog-id=39 op=UNLOAD
```

whot, 2021-04-20:
> (EE) Wacom Intuos S Pad pad: Error reading wacom device : No such device
> (II) config/udev: removing device Wacom Intuos S Pad pad
> ...
>   (II) XINPUT: Adding extended input device "Wacom Intuos S Pad pad" (type: PAD, id 23)

So this is is basically the core of the issue - the device sees a read error and gets removed by udev (this triggered by the kernel) and then is re-added shortly after that. I'm assuming this is some device firmware re-initialization or something (@jigpu?) that triggers these events.

But the overall issue remains the same: you have a one-shot tool that assumes the device is never unplugged. That simply isn't the case.

An xorg.conf snippet would be more reliable here but since we don't support setting a button to "pan" you can only set one option - which doesn't really get you anywhere.  Adding "pan" as potential button action to the config parser would be viable though.

```
$ cat /etc/X11/xorg.conf.d/99-wacom-panscroll.conf
Section "InputClass"
   Identifier "Set panscroll for wacom pads"
   MatchProduct "Wacom Intuos S Pad pad "
   Option "PanScrollThreshold" "200"
EndSection
```

jigpu, 2021-05-13:
> So this is is basically the core of the issue - the device sees a read error and gets removed by udev (this triggered by the kernel) and then is re-added shortly after that. I'm assuming this is [...]

I'd put money on the re-initalization being caused by fwupd. It typically kicks in a few seconds after logging in and will cause this kind of disconnect / reconnect symptom. The Intuos is also one of the few devices that fwupd currently probes.

As for workarounds, you might take a peek at a script I wrote some time ago to handle xsetwacom configuration in the face of hotplug. You'll need to add a case for your particular device at the bottom of the script and add the commands that it should run. See https://gist.github.com/jigpu/1fad061de7e84c2a1ee3208e1a971342

zoliky, 2021-05-14:
Thank you. I'll take a look at it.

closed 2021-05-14 by zoliky

zoliky, 2021-05-18:
You were right. I tried your script and everything works fine. Thank you.

zoliky, 2021-05-18:
@jigpu You were right. I tried your script. Everything works fine now.