linuxwacom/xf86-input-wacom #174: Thinkvision M14t touch not recognized, Eraser issues

naib864, 2021-04-20:
Some issues I noticed while testing this device:

1. Touchscreen doesn't work. Not sure what I can do to troubleshoot, I couldn't find anything about this device under linux. `xev` won't output any event when touching the screen.

2. Eraser doesn't work properly.
The pen has two buttons, one seems to be the Eraser. Programs like Xournal++ will recognize the eraser when I press said button and float the pen a few millimeters above the screen (The eraser symbol will show up). But as soon as I try to erase by touching the screen with the tip while holding the button, the program will just switch to drawing.
The second button works as expected, even when the tip is touching the screen.

Output of `xsetwacom --list devices`:
```
Wacom Co.,Ltd. Pen and multitouch sensor stylus	id: 10	type: STYLUS    
Wacom Co.,Ltd. Pen and multitouch sensor eraser	id: 11	type: ERASER    
Wacom Co.,Ltd. Pen and multitouch sensor UNKNOWN touch	id: 13	type: TOUCH     
Wacom Co.,Ltd. Pen and multitouch sensor UNKNOWN pad	id: 14	type: PAD       
Wacom Co.,Ltd. Pen and multitouch sensor touch	id: 21	type: TOUCH     
Wacom Co.,Ltd. Pen and multitouch sensor touch	id: 22	type: TOUCH
```

Output of `xev` while only pressing tip on the screen:
```
MotionNotify event, serial 34, synthetic NO, window 0x3c00001,
    root 0x718, subw 0x0, time 1023140, (1048,681), root:(2974,687),
    state 0x100, is_hint 0, same_screen YES

```    
Output of `xev` while pressing tip on the screen and pressing button 2:
```
MotionNotify event, serial 34, synthetic NO, window 0x3c00001,
    root 0x718, subw 0x0, time 1158381, (1347,661), root:(3273,667),
    state 0x200, is_hint 0, same_screen YES

```
Output of `xev` while pressing tip on the screen and pressing button 1:
```
MotionNotify event, serial 34, synthetic NO, window 0x3c00001,
    root 0x718, subw 0x0, time 1287470, (822,624), root:(2748,630),
    state 0x100, is_hint 0, same_screen YES

```

The state will switch from 0x100 to 0x0 for a moment when pressing button 1, but will switch right back to 0x100.
OS is arch with latest kernel.

naib864, 2021-04-20:
Solved the touch issue, turns out the device has one mode for windows and one mode for Android/ChromeOS, which one can set on the monitor itself. Ironically, setting it to Windows will enable Linux to recognize the touch input, while Android/ChromeOS won't work.

naib864, 2021-04-20:
UPDATE:
The touch seems to work initially after connecting the monitor but as soon as the pen touches the monitor, touch will not work until the monitor is reconnected.

naib864, 2021-04-20:
UPDATE:

The monitor itself appears to have two modes: Android/ChromeOS and Windows.

In Windows mode the problem described above will appear. One thing to add: Touch will work initially but as soon as you use the pen, touch will cease to work until reconnecting the Monitor or switching the mode on the Monitor itself.

In Android/ChromeOS mode everything will be recognized as a mouse by all programs.

evilphish, 2021-05-28:
I am experiencing the same issue on Linux Mint 20.1 running 5.8.0.

*Regarding touch:*
Same issue - touch works until I bring the active stylus close to the display. As soon as the display recognizes the stylus, even if I take it away, normal touch no longer works till I replug the monitor.

*Regarding the stylus*
Top button always defaults to eraser. Same problem as naib has - in Xournal it recognizes the eraser but pressing down the stylus then draws instead.

For me the above stated commands return the same output. If there is anything I can do to help debug this I am happy to do so.

naib864, 2021-05-28:
> *Regarding touch:*
> Same issue - touch works until I bring the active stylus close to the display. As soon as the display recognizes the stylus, even if I take it away, normal touch no longer works till I replug the monitor.

Removing xf86-input-wacom and only using libwacom solved this issue for me. Eraser issue is still there, though.  

[Someone on reddit](https://www.reddit.com/r/thinkpad/comments/mvek4z/thinkvision_m14t_linux_issues/gz21u1i?utm_medium=android_app&utm_source=share&context=3) said the device works out of the box without issues on fedora (with wayland), couldn't test it for myself yet though.

evilphish, 2021-05-28:
I can confirm that removing input-wacom works but it also makes you lose the pressure sensitivity and all the other graphics tablet settings so for me this is not really a fix. Would be interesting to figure out why input-wacom does that

whot, 2021-06-21:
> Removing xf86-input-wacom and only using libwacom solved this issue for me. Eraser issue is still there, though.

libwacom is a text database with a few C apis around it, it has no (direct) effect on how the pen works. If you remove xf86-input-wacom, most likely you're falling back onto xf86-input-libinput for tablet handling and there are a few different behaviours there which could explain the issue.

If the pen doesn't work with libinput, that's a libinput issue (or hw issue), so I recommend filing [a libinput issue](https://wayland.freedesktop.org/libinput/doc/latest/reporting-bugs.html) to debug this. It may also shed some light on why the xf86-input-wacom driver doesn't work.

jigpu, 2021-06-21:
I have to wonder what is different between the device's "Windows" and "Android/Chrome" modes. If either of you could follow the [contributing instructions for wacom-hid-descriptors](https://github.com/linuxwacom/wacom-hid-descriptors#contributing) to submit logs from the device in both modes, it would help us understand what is going on.

A log for one of the modes (not sure which) has been submitted at https://github.com/linuxwacom/wacom-hid-descriptors/issues/157 and it shows the device appearing as 0x2D1F, so I suspect that the switch changes the VID and affects whether this driver or xf86-input-libinput are used.

evilphish, 2021-06-24:
Attached you can find the requested logs for both windows and android mode.

[sysinfo_m14t_windows-mode.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/6710977/sysinfo_m14t_windows-mode.tar.gz)
[sysinfo_m14t_android-mode.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/6710978/sysinfo_m14t_android-mode.tar.gz)

Quick observation: In android mode the issue with the screen no longer accepting touch input after seeing the pen once is not there but also the pen does not get detected while hovering, only when touching the screen directly, thus disabling "palm detection".

jigpu, 2021-07-13:
Thanks for the logs. It looks like the hardware is exposed identically in both modes, so the difference in behavior should be a matter of how the hardware sends events over the devices that get created. There are a couple of different devices:
* USB Endpoint 1, HID Multitouch Collection
* USB Endpoint 1, HID Pen Collection
* USB Endpoint 2, HID Multitouch Collection [but possibly only supports one finger?]

Can you capture a segment of HID recordings in both modes for me? You'll need to clone the [hid-tools repository](https://gitlab.freedesktop.org/libevdev/hid-tools) and then use its `hid-recorder` program. We can replay the recordings locally to examine how the xf86-input-wacom, xf86-input-libinput, and Wayland libinput drivers are all reacting to the HID events that get sent by the hardware.

1. Clone the hid-tools repository: `git clone https://gitlab.freedesktop.org/libevdev/hid-tools`
2. Switch the device into Windows mode
3. From inside the hid-tools directory, run `sudo ./hid-recorder` and then press CTRL+C rather than selecting any single device
4. Note the paths of any Wacom devices listed
5. Run `sudo ./hid-recorder PATH1 PATH2 [...etc...] | tee record-windows-mode.log`. You should see a bunch of data printed to the screen; all this data describes the HID devices and their report formats.
6. Perform each of the following actions. Even more data should be printed to the screen as data is sent by the hardware:
  a. Swipe a single finger on the screen three times so we can see what single-touch events look like
  b. Swipe two fingers on the screen three times so we can see what multitouch events look like
  c. Swipe the pen on the screen three times so we can see what pen events look like
  d. Hold down the top (eraser) button on the pen and swipe the screen three times so we can see what eraser events look like
  e. Swipe a single finger on the screen three times so we can see how the hardware switches back from pen to touch events
7. Press CTRL+C to quit recording
8. Switch the device into Android / Chrome OS mode
9. Repeat the instructions from step 3, this this logging to record-android-mode.log in step 5.
10. Attach both record-windows-mode.log and record-android-mode.log to this issue (you may need to place them both in a tarball or ZIP file first).

evilphish, 2021-07-13:
[record-android-mode.log](https://github.com/linuxwacom/xf86-input-wacom/files/6811240/record-android-mode.log)
[record-windows-mode.log](https://github.com/linuxwacom/xf86-input-wacom/files/6811242/record-windows-mode.log)
Your wish is my command sire :)

Btw do we ignore the bottom button for now? (I did *not* press it during this excercise, as requested)

jigpu, 2021-07-14:
Thanks! The data looks reasonable at a first glance, and I can reproduce the problem of touch being disabled after the pen is brought into prox. The hardware is clearly still sending those touch events, so I'll need to dig into what could be going wrong.

As for the pen's top button, it is being signaled as a standard barrel switch and the HID driver reports a `BTN_STYLUS` event to match. The xf86-input-libinput driver reports that event as a middle mouse button, which Xournal appears to treat as an eraser.  The events look fine (the button remains held down while the tip is in contact), so my initial guess is that bug here is in Xournal -- it shouldn't suddenly switch from eraser to pen just because the tip is down. I'll need to look a little more closely to be sure that the events are sane.

Finally, it appears that Android mode operates by sending pen events out of the Endpoint 2 device I mentioned above. In this mode the pen appears like a finger to the system, so you won't get any additional functionality through it. It should be fine as a workaround, but Windows mode is definitely what we want to have working since you'll get the full pen experience.

evilphish, 2021-07-14:
Great, if there is anything I can add I'd be happy to do so :)

dkadioglu, 2021-08-30:
Any news on this? If helpful, I can also offer to help in testing or gathering further info. Thanks for looking into this.

Pinglinux, 2021-09-02:
@evilphish @dkadioglu As Peter suggested, please file a libinput issue at https://wayland.freedesktop.org/libinput/doc/latest/reporting-bugs.html and share the link to the libinput issue to this page. Someone at libinput project will look into it. Thank you.

dkadioglu, 2021-09-17:
done: https://gitlab.freedesktop.org/libinput/libinput/-/issues/658

closed 2021-09-17 by Pinglinux