linuxwacom/xf86-input-wacom #175: ELAN2514 04F3:2B05 :: module not loaded on OpenSUSE kernel 5.11.15

frgomes, 2021-04-25:
I've read somewhere (and I cannot find anymore!) that when the kernel driver does not load we should inform you guys about that. Sorry... that''s all I remember. I hope my report here happens to be useful.

My stylus apparently works, but it does not show up on KDE. Being a newbie on this matter, I had a difficult time understanding why I was not able to calibrate the stylus on the correct screen.

### Environment

```bash
$ uname -a
Linux mars 5.11.15-1-default #1 SMP Fri Apr 16 16:47:34 UTC 2021 (64fb5bf) x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
LSB Version:    n/a
Distributor ID: openSUSE
Description:    openSUSE Tumbleweed
Release:        20210423
Codename:       n/a
```

### Creating configuration file

I had to create the file below:

file: /usr/share/libwacom/ELAN2514_00_04F3_2B05.tablet
```
# ELAN touchscreen/pen sensor present in the HP Envy x360 Convertible 15-ee0xxx

[Device]
Name=ELAN 2514
ModelName=
DeviceMatch=i2c:04f3:2b05
Class=ISDV4
Width=14
Height=8
IntegratedIn=Display;System
Styli=0xffffe;0xfffff

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=0
```

### Loading the kernel module

```bash
$ sudo modprobe wacom
```

### My X11 configuration

I've run ``xinput_calibrator` --output-filename`` and I've obtained the lines below:

file: /etc/X11/xorg.conf.d/70-wacom.conf
```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ELAN 2514 04F3:2B05"
        Option  "MinX"  "-40050"
        Option  "MaxX"  "16742991"
        Option  "MinY"  "93207"
        Option  "MaxY"  "16746145"
        Option  "SwapXY"        "0" # unless it was already set to 1
        Option  "InvertX"       "0"  # unless it was already set
        Option  "InvertY"       "0"  # unless it was already set
EndSection
```

frgomes, 2021-04-28:
I'm not able to configure in KDE via [wacomtablet](https://invent.kde.org/system/wacomtablet) because the tablet is not listed by the command below:

    $ xsetwacom list devides

frgomes, 2021-04-28:
https://bugs.kde.org/show_bug.cgi?id=436324

jigpu, 2021-06-21:
It looks like the kernel is able to configure the sensor since your KDE bug shows an `ELAN2514:00 04F3:2B05` device in the output of `xinput list`. However, since your sensor is from Elan rather than Wacom, the xf86-input-wacom driver doesn't work with it by default. This is why it does not appear in `xsetwacom` (which only lists devices known to xf86-input-wacom).

You could try creating a file named `/etc/X11/Xorg.conf.d/90-wacom-override.conf` with the following contents and rebooting. This will tell Xorg to have our driver try to power the sensor even though it normally wouldn't. This may or may not work, however, since the driver is not designed for Elan sensors.
~~~
Section "InputClass"
        Identifier "Wacom driver override"
        MatchProduct "ELAN 2514 04F3:2B05"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
~~~

closed 2021-06-21 by jigpu

frgomes, 2021-06-22:
I inform that I've got some information wrong in my first report. In particular, the proper device name, which must go onto the ``MatchProduct`` line. So, I've fixed that, based onto your suggestion in the previous post.

This is below my ``/etc/X11/xorg.conf.d/90-wacom-override.conf``:
```
Section "InputClass"
        Identifier "Wacom driver override"
        MatchProduct "ELAN2514:00 04F3:2B05"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
```

Yes, it works! KDE is now able to manage the tablet.

frgomes, 2023-06-07:
Full configuration which is known to work well at:
https://gist.github.com/frgomes/58a6a7bc57cfbb9c77fcd46774811c9d