linuxwacom/xf86-input-wacom #176: Replacement for Twinview parameter

JohnnyJayJay, 2021-04-27:
Hello, 

I'm trying to set up my Wacom Intuos BT M on my Manjaro computer. So far, everything seems to work fine. However, I have a dual-screen setup and I'd like to only use the first screen for the tablet, as this feels more natural and is easier to handle. It seems that in the past, the "Twinview" parameter was used to configure this, but according to the wiki, it has been removed without replacement. Unfortunately I haven't found any new resources on this topic so far, only references to "Twinview". Is there a workaround? A new way to do this? If there isn't, could this be implemented in some form again?\
Thanks in advance.

whot, 2021-04-27:
`xsetwacom MapToOutput` should work for that but it's a runtime command (since outputs can change at runtime too). GNOME should automatically map things correctly too, but I'm not sure what other DEs do.

JohnnyJayJay, 2021-04-28:
Oh, thank you, it seems I didn't look at the Wiki correctly. It's all [well-described](https://github.com/linuxwacom/xf86-input-wacom/wiki/Dual-and-Multi-Monitor-Set-Up) after all. `MapToOutput` does indeed work for me.

closed 2021-04-28 by JohnnyJayJay