linuxwacom/xf86-input-wacom #177: Weird pan behaviour when screen rotates

Lucaacer, 2021-05-20:
I am running Ubuntu 20.04 on a Lenovo thinkpad Yoga 11E, which has got both a touchscreen and a stylus (not a wacom one, I suppose. It should be a N-Trig one).

I wanted to enable pan scrolling, so I moved from libinput to wacom by adding to /usr/share/X11/xorg.conf.d/70-wacom.conf these lines

# Lenovo thinkpad pro pen
Section "InputClass"
        Identifier "Lenovo thinkpad pen pro"
        MatchProduct "ELAN901C:00 04F3:2A42"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection

and now xsetwacom --list gives this output

ELAN901C:00 04F3:2A42 touch     	id: 12	type: TOUCH     
ELAN901C:00 04F3:2A42 stylus    	id: 13	type: STYLUS    
ELAN901C:00 04F3:2A42 eraser    	id: 17	type: ERASER 

On every boot this script is being run in order to enable scrolling/pan while moving my pen and holding down button number 2

xsetwacom set 13 button 2 pan
xsetwacom set 17 "PanScrollThreshold" 600

and I must admit that it's a nice feature that woks very well system wide.

But here comes the issue... When the screen rotates, touch, stylus and eraser follow rotation accordingly and I can touch, select, write, erase as usual, On the other side, pan does not work and it's 90 degrees off.

So I tried 

xsetwacom set "ELAN901C:00 04F3:2A42 stylus" Rotate cw

which actually lets me scrolling/panning in portrait mode, but the postion of the cursor, while using the stylus, is quite erratic. It looks there is an umpredictable offset which makes the stylus itself unusable for anything else but scrolling/panning.

Please, have a look at this bug and let me know if I can help you with any log or info.

One more detail, the behaviour is the same both if I rotate the screen with xrandr and if the screen automatically rotates due to the desktop settings (Ubuntu gnome takes care of autorotation in settings).

Thanks.

jigpu, 2021-06-21:
Testing with Fedora rather than Ubuntu, it appears that GNOME's device rotation code might be incorrect. When I rotate the screen on my device and run `xinput list-props <stylus>` I see that it has modified the device's "Coordinate Transformation Matrix" property to rotate the output coordinates to match the screen. It should, however, have modified our "Wacom Rotation" property instead. The former does not work properly with panning.

The erratic behavior that you see when only using the "Rotate" command is caused by rotation being performed twice -- once by the property that GNOME is setting and a second time by the property you've set through xsetwacom. To get a more sensible behavior where panning works properly, use the following commands. The first resets the "Coordinate Transformation Matrix" property so that the double-rotation doesn't happen and the second does the rotation in a pan-compatible way.
~~~
xsetwacom set "ELAN901C:00 04F3:2A42 stylus" MapToOutput desktop
xsetwacom set "ELAN901C:00 04F3:2A42 stylus" Rotate cw
~~~

closed 2021-06-21 by jigpu

jigpu, 2021-06-21:
Bug has been opened on GNOME's bug tracker for its "mutter" project: https://gitlab.gnome.org/GNOME/mutter/-/issues/1860