linuxwacom/xf86-input-wacom #178: Using this driver for touchscreens breaks touch-enabled Qt software

Pointedstick, 2021-05-25:
The xorg config snippet that this driver installs (`70-wacom.conf`) contains a section that will take over handling of touchscreens from Libinput. Unfortunately the driver does not handle touch events in a way that Qt expects, and as a result, touch support for Qt software gets broken; the cursor is moved to the touched location, but the thing you touched does not get activated. Perhaps a click is being sent rather than a touch? See https://bugreports.qt.io/browse/QTBUG-84756 for more details.

Consider either investigating and fixing the mis-handled touch events, or else removing the part of the config file that takes over input handling for touchscreens, and just leave it to Libinput instead.

Some distros have been patching this out downstream for quite some time. See
- Ubuntu: https://bugs.launchpad.net/ubuntu/+source/xf86-input-wacom/+bug/1774242
- openSUSE: https://bugzilla.suse.com/show_bug.cgi?id=1172669

Fixing this issue would allow them to drop their patches.

whot, 2021-05-26:
ftr, the reason the wacom driver is assigned to those devices is so that tablet and touch are handled by the same driver. This allows for touch arbitration and for the (wacom-driver-specific) gestures.

@jigpu disabled gestures in 31a5405f7d9405bc514585709161287b0c67386e for touchscreens, so this should all work now? What version of the driver are you running?

Pointedstick, 2021-05-27:
I am running 0.40.0 on Fedora 34 and disabling gestures is not enough to fix it for me. See the related discussion in https://bugzilla.suse.com/show_bug.cgi?id=1172669.

jigpu, 2021-05-28:
I used `libinput record` and `libinput replay` along with `xinput test-xi2` to compare the events generated by the Wacom and libinput drivers from a simulated Cintiq 24HDT. Running a diff doesn't reveal any obviously-smoking guns, with events generally being functionally identical across the runs (different valuator range and choice of touch IDs make the logs not perfectly identical). We're sending both real Touch events and emulated Motion events just like libinput.

The only notable differences I see are:
* The choice of valuator axis is different: libinput uses Abs MT Position X/Y while we use Abs X/Y
* A (very) few [Raw]TouchUpdate events may be present in one log but not the other

I also noticed that disabling averaging (e.g. with `xsetwacom set <touch> RawSample 1`) causes the driver to never send TouchUpdate events (every update is sent as TouchBegin). I will send a fix for this shortly, but I doubt it is the cause of the Qt problem.

I need to do additional testing with a Wacom tablet PC touchscreen to see if I can find any notable differences there. I'd also like to experiment with changing the valuator axis labels to see if that might change how Qt treats the device...

jigpu, 2021-07-20:
I spent some more time looking at this and haven't been able to find anything wrong with the implementation. Once I ensure `Gesture Off` is set, Qt software operates fine regardless of the valuator axis label or the slight event difference I noted previously.

One thing I have noticed, however, is that KDE seems to set `Gesture On` after logging in. This overrides the new driver default we set in version 0.38.0 as well as overriding any user- or distro-supplied xorg.conf.d file which sets `Gesture Off`. The only way to get touch working properly under KDE is to manually run `xsetwacom set <id> gesture off` after logging in. This is obviously a severe usability issue...

@Pointedstick can you switch back to using our driver for touch and doing some tests? I would be interested in the following:
* Distro and desktop environment you're using
* Run `xsetwacom -V` to double-check the driver version
* Run `xsetwacom get <id> gesture` with the ID of your touchscreen to see if the in-driver gestures are turned on, despite the new default
* Run `xsetwacom set <id> gesture off` with the ID of your touchscreen and then see if controls respond any better

Pointedstick, 2021-07-27:
Thanks for investigating, and sorry for the delayed response. I was out on vacation.

I did a quick check and couldn't find anything in out software that explicitly sets `gesture on` but I can keep looking if you're sure it's our issue. Any chance you knew of which component was responsible?

Anyway, here's the rest of the requested output:

```
$ xsetwacom -V
0.40.0

$ xsetwacom get 9 gesture
on

$ xsetwacom set 9 gesture off
[the problem is instantly fixed]
```

Pointedstick, 2021-08-03:
Never mind. I found the KDE software responsible: it was our Wacom tablet configuration UI, which has a way to enable gestures. It is set to on by default. I will turn it off by default and add a message warning people that touch support might break if they turn it on. There is no action for you you take here. Sorry for the noise.

closed 2021-08-03 by Pointedstick

Pointedstick, 2021-08-03:
You can follow https://bugs.kde.org/show_bug.cgi?id=440556 if you're interested in the KDE side.