linuxwacom/xf86-input-wacom #180: Unable to map WACOM tablet to multi monitor (fedora 34 X11)

lverissimo, 2021-07-25:
My wacom tablet (CTL-4100 (Wacom Intuos S) suddenly mapped to only one of my 2 displays. In the wacom tablet config tab I see the map to single display is off. When I click it on, I can then map the tablet to a specific display. But when I click it off, it automatically map to primary display. The same tablet works well in relative mode, and it works well on another RHEL 8.4 laptop I have.  Any help is welcome.

Fedora 34
Kernel: 5.13.4-200.fc34.x86_64
Gnome: GNOME Shell 40.3

xrandr --listactivemonitors
```
Monitors: 2
 0: +*eDP-1 1920/309x1080/174+0+1050  eDP-1
 1: +HDMI-1 1680/474x1050/296+240+0  HDMI-1

```

xsetwacom --get "Wacom Intuos BT S Pen stylus" all
```
Option "Area" "0 0 15200 9500"
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "DebugLevel" "0"
Option "CommonDBG" "0"
Option "Suppress" "2"
Option "RawSample" "4"
Option "PressCurve" "0 0 100 100"
Option "Mode" "Absolute"
Option "TPCButton" "off"
Option "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
Option "Gesture" "off"
Option "ZoomDistance" "0"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Option "CursorProx" "30"
Option "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Option "Threshold" "26"
Option "Serial" "0"
Option "PressureRecalibration" "on"
Option "PanScrollThreshold" "1300"
```

xinput list-props "Wacom Intuos BT S Pen stylus"
```
Device 'Wacom Intuos BT S Pen stylus':
	Device Enabled (187):	1
	Coordinate Transformation Matrix (189):	1.000000, 0.000000, 0.000000, 0.000000, 0.507042, 0.492958, 0.000000, 0.000000, 1.000000
	Device Accel Profile (315):	0
	Device Accel Constant Deceleration (316):	1.000000
	Device Accel Adaptive Deceleration (317):	1.000000
	Device Accel Velocity Scaling (318):	10.000000
	Device Node (312):	"/dev/input/event4"
	Wacom Tablet Area (353):	0, 0, 15200, 9500
	Wacom Rotation (354):	0
	Wacom Pressurecurve (355):	0, 0, 100, 100
	Wacom Serial IDs (321):	886, 159545647, 2146, 0, 0
	Wacom Serial ID binding (322):	0
	Wacom Proximity Threshold (356):	30
	Wacom Pressure Threshold (323):	26
	Wacom Sample and Suppress (324):	2, 4
	Wacom Enable Touch (325):	0
	Wacom Hover Click (357):	1
	Wacom Enable Touch Gesture (326):	0
	Wacom Touch Gesture Parameters (327):	0, 0, 250
	Wacom Tool Type (328):	"STYLUS" (348)
	Wacom Button Actions (329):	"Wacom button action 0" (330), "Wacom button action 1" (331), "Wacom button action 2" (332), "None" (0), "None" (0), "None" (0), "None" (0), "Wacom button action 3" (333)
	Wacom button action 0 (330):	1572865
	Wacom button action 1 (331):	1572866
	Wacom button action 2 (332):	1572867
	Wacom button action 3 (333):	1572872
	Wacom Pressure Recalibration (358):	1
	Wacom Panscroll Threshold (346):	1300
	Device Product ID (313):	1386, 886
	Wacom Debug Levels (347):	0, 0
```

whot, 2021-07-26:
This would indicate a GNOME mutter or control-center bug - the `Coordinate Transformation Matrix` is set in the above output and mutter is the only thing that would set this in a normal setup.

venturewise, 2022-01-01:
Thanks @whot for your comment. Are you suggesting that the Coordinate Transformation Matrix should not be set in order for the multi-monitor mapping to work?
 
I only just started seeing it with my upgrade of Pop!_OS 21.10 this week. I'm using an Intuos Pro M (PTH-660), and I also tried 
with a PTH-651.

Pop!_OS 21.10 
Kernel: 5.15.8-76051508-generic
GNOME 40.0.0 on X11

```
Monitors: 3
 0: +*DP-2 1920/344x1080/193+2560+286  DP-2
 1: +DP-3 2560/641x1600/401+0+0  DP-3
 2: +DP-0 2560/641x1600/401+4480+0  DP-0

Device 'Wacom Intuos Pro M Pen stylus':
	Device Enabled (155):	1
	Coordinate Transformation Matrix (157):	0.272727, 0.000000, 0.363636, 0.000000, 0.675000, 0.178750, 0.000000, 0.000000, 1.000000
(etc...)
```

To recap the issue: When using multiple monitors, "Touchpad (Relative)" tracking mode works as expected, but "Tablet (Absolute)" mode maps only to monitor 1 (the laptop's display) although the 'Map to Monitor' option is unchecked.

whot, 2022-01-04:
> Are you suggesting that the Coordinate Transformation Matrix should not be set in order for the multi-monitor mapping to work?

Correct. If the matrix is the unity matrix the tablet input spans the accessible desktop (i.e. all monitors). If it is set, it is mapped to the portion of the desktop that the matrix refers to. You can easily test this by manually running xinput to set the matrix to the unity matrix (pretty sure mutter doesn't monitor this actively so it won't reset it immediately). 

If the mapping happens with "Map To Monitor" unchecked the bug is in mutter somewhere.

closed 2022-01-04 by whot

jigpu, 2022-01-07:
Issue has been filed with GNOME: https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/1568