linuxwacom/xf86-input-wacom #185: Lenovo m14t: Touch ceases to work after using pen; re-enabling possible with using eraser or reconnecting tablet.

felixbecker2, 2021-10-16:
Ahoj,

*(I think this is an issue for the wacom Xorg driver, because [people report that it is not happening with Wayland](https://www.reddit.com/r/thinkpad/comments/mvek4z/thinkvision_m14t_linux_issues/), and for myself it also does not happen when I use it under the console with [`gpm`](https://wiki.archlinux.org/title/General_purpose_mouse), that's why I report it at [`xf86-input-wacom`](/linuxwacom/xf86-input-wacom); if you think it belongs elsewhere please move or tell me.)*

I have a [Lenovo m14t monitor](https://www.lenovo.com/us/en/p/accessories-and-software/monitors/office/62a3uar1us) with multitouch and stylus. It has a wacom tablet, and the stylus is an active one with battery, pressure sensitivity and two buttons: One "stylus button" and the other turns it into an eraser.

Initially after connecting the monitor, touch input and stylus input work.

Whenever I used the stylus at least once, touch input ceases to work, even when the stylus is removed from the screen.

Touch input can be restored if I:

* Disconnect and reconnect the USB connection[1], *or*
* holding down the eraser button on the stylus while removing the stylus from the screen, *or*
* switching to another virtual console and back to Xorg.

Whenever I remove the stylus from the screen, the following line appears in the Xorg log file (`/var/log/Xorg.0.log` on my system):

```
(EE) Wacom Co.,Ltd. Pen and multitouch sensor Stylus eraser: usbParse: Ignoring event from invalid serial 0
```

System information:

* Operating system: [Artix Linux](https://artixlinux.org/) with OpenRC (that is an [Arch Linux](https://archlinux.org/) without systemd),
* Kernel version: 5.14-[pf4](https://gitlab.com/post-factum/pf-kernel/-/wikis/README),
* `Xorg -version`: `X.Org X Server 1.20.13`
* Version of package `xf86-input-wacom`: `0.40.0-1.1`
* `lsusb`:
  ```
  Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 127: ID 2d1f:524c Wacom Taiwan Information Co. Ltd. Pen and multitouch sensor
  Bus 001 Device 124: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
  Bus 001 Device 005: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
  Bus 001 Device 003: ID 6080:8061 AMR-4630-XXX-0- 0-1023 USB KEYBOARD
  Bus 001 Device 108: ID 04b3:301e IBM Corp. IBM USB Travel Keyboard with Ultra Nav
  Bus 001 Device 107: ID 04b3:3016 IBM Corp. UltraNav Keyboard Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
  ```
* `lsusb -vv -d 2d1f:524c`:
  ```
  Bus 001 Device 004: ID 2d1f:524c Wacom Taiwan Information Co. Ltd. Pen and multitouch sensor
  Device Descriptor:
    bLength                18
    bDescriptorType         1
    bcdUSB               2.00
    bDeviceClass            0 
    bDeviceSubClass         0 
    bDeviceProtocol         0 
    bMaxPacketSize0        64
    idVendor           0x2d1f Wacom Taiwan Information Co. Ltd.
    idProduct          0x524c 
    bcdDevice            0.01
    iManufacturer           1 Wacom Co.,Ltd.
    iProduct                2 Pen and multitouch sensor
    iSerial                 0 
    bNumConfigurations      1
    Configuration Descriptor:
      bLength                 9
      bDescriptorType         2
      wTotalLength       0x003b
      bNumInterfaces          2
      bConfigurationValue     1
      iConfiguration          0 
      bmAttributes         0xe0
        Self Powered
        Remote Wakeup
      MaxPower              400mA
      Interface Descriptor:
[... 400 lines cut ...]
      Detail for Valuator 4:
        Label: None
        Range: 0.000000 - 1.000000
        Resolution: 1 units/m
        Mode: absolute
        Current value: 0.000000
      Class originated from: 24. Type: XIValuatorClass
      Detail for Valuator 5:
        Label: None
        Range: 0.000000 - 1.000000
        Resolution: 1 units/m
        Mode: absolute
        Current value: 0.000000
      Class originated from: 24. Type: XITouchClass
      Touch mode: direct
      Max number of touches: 5
  ```
* Attached is a logfile ([`Xorg.1.log`](https://github.com/linuxwacom/xf86-input-wacom/files/7357680/Xorg.1.log)) from a fresh start of Xorg with `X -retro -logverbose 6 :1`, with doing some touch, then some stylus, then some touch, and then terminating the server with `<CTRL>`+`<ALT>`+`<Backspace>`.
* Using touchscreen or stylus does not produce further log messages with verbosity 6, except of the already mentioned `(EE) Wacom Co.,Ltd. Pen and multitouch sensor Stylus eraser: usbParse: Ignoring event from invalid serial 0` when removing the pen.

[1] I use my monitor not directly via USB C, but with a ["Wacom Link Plus" adapter](https://estore.wacom.com/en-US/wacom-link-plus-dongle.html), which takes HDMI output from the computer, power from a power supply and a USB connection from the computer and combines that into USB-C with video via USB-C alt mode, power for the monitor and USB data connection from the monitor to the computer. So I can disconnect just the data connection while leaving the monitor powered and displaying video output. The problem I descibed however also appeared with direct connection to USB-C, so I see the Wacom Link Plus adapter not interfering here.

---

[Xorg.1.log](https://github.com/linuxwacom/xf86-input-wacom/files/7357680/Xorg.1.log)

felixbecker2, 2021-11-26:
Just a note for anyone who reads this thread and is wondering about progress: All things which are also related to this problem sadly happen to be discussed only in issue #186. So please see there for ideas, workarounds, maybe finally solutions &hellip;

felixbecker2, 2022-02-24:
The patch mentioned in [this comment](/linuxwacom/xf86-input-wacom/issues/186#issuecomment-1029493500) seems to solve this issue, see [here](https://github.com/linuxwacom/xf86-input-wacom/issues/186#issuecomment-1049647946).

Pinglinux, 2022-03-18:
Benjamin's hid-generic patches have been merged.

closed 2022-03-18 by Pinglinux