linuxwacom/xf86-input-wacom #189: Pen keeps skipping!

quantatanu, 2021-11-18:
Hi, before installing anything my wacom ctl 4100 small was working well (without pressure sensitivity though), then I ran "./autogen.sh" for no reason and now the pen keeps skipping when I apply slight pressure.. it's like those old ball-pens when there are air bubbles trapped inside the refill.

jigpu, 2021-11-30:
Apologies for the delay. Are you still experiencing this problem?

After running `./autogen.sh` did you run any other commands? Typically you would run `./autogen.sh` and then `make` and finally `sudo make install`.

quantatanu, 2021-12-11:
Hi, thanks for your reply. I am not sure what I did, but the problem went away and everything started working smoothly.

jigpu, 2021-12-13:
Alright, that's okay :slightly_smiling_face:

Just to confirm, when you say you don't have any option to configure the tablet, do you mean that the "xsetwacom" command is missing? Or that your desktop (GNOME, KDE, etc.) doesn't display your tablet in the control panel?

quantatanu, 2021-12-17:
@jigpu hi,

I am on **linux mint 20.2**, `xsetwacom` command is there, although I don't know the options to set **express keys** with it. When I search `wacom` in the **start menu** I see a **Graphics Tablet**, nothing happens when I click on it!

jigpu, 2021-12-17:
I can't help with the "Graphics Tablet" issue (I'm not familiar with the control panel that Mint wrote), but to configure the ExpressKeys through xsetwacom you can use commands like the following to have them generate keypresses A through D. You can run `man xsetwacom` and scroll down to the "Button" parameter for more info on what else you can have the ExpressKeys do.

~~~
$ xsetwacom set "Wacom Intuos S Pad pad" button 1 key a
$ xsetwacom set "Wacom Intuos S Pad pad" button 2 key b
$ xsetwacom set "Wacom Intuos S Pad pad" button 3 key c
$ xsetwacom set "Wacom Intuos S Pad pad" button 8 key d
~~~
(the button numbering 1, 2, 3, 8 looks strange but I think is correct for your tablet...)

quantatanu, 2021-12-30:
@jigpu are you available right now?
I was afraid to try anything because I was in the middle of a project! But then something unfortunate happened! I didn't do anything, suddenly one fine morning... I can't draw a circle... a circle becomes a pure ellipse (circles are mathematically ellipses :) )... that is, the vertical direction got shrunk, a circle maps to a horizontal ellipse! I am unable to fix it... I tried `xsetwacom --list devices` and this time I get only two outputs:

`
:~$ xsetwacom --list devices -v
Wacom Co.,Ltd. Intuos S stylus  	id: 18	type: STYLUS    
Wacom Co.,Ltd. Intuos S eraser  	id: 19	type: ERASER 
 `  
 
 unlike last time, when along with these two, there were: `pen cursor` and `Pad pad`! Therefore, this gives:
` 
 :~$ xsetwacom set "Wacom Intuos S Pad pad" button 1 key a
Cannot find device 'Wacom Intuos S Pad pad'.
`

Can you please help me fix the scaling/mapping issue? By the way a few details about my system:
------------------------------------------------------------------------------------
Laptop: Dell Inspoiron 14 (5491 2-in-1)
OS: Linux Mint 20.2 
Tab: Wacom Intuos CTL 4100
xsetwacom version 0.39.0 (Xserver-xorg-input-wacom: 1:0.39.0ubuntu1)

quantatanu, 2021-12-30:
This is strange, after a `reboot` everything came back to normal! I still don't exactly know how to assign express keys to do specific tasks.. for example, how can I set an express key to do `ctrl+z` for example?

Thanks

Pinglinux, 2022-03-17:
As Jason pointed out, we don't know how configuration is done under Mint. IF you are willing to use xsetwacom command, please  checkout the examples here: https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-1%3A-xsetwacom-and-xorg.conf

jigpu, 2022-03-17:
It looks like you're getting bit by the "[Android Misdetect](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect)" issue. This affects the CTL-4100 and causes the device mapping to change. The smoking gun pointing to a misdetect issue (as opposed to something else) is that the device is being listed as a "Wacom Co.,Ltd. Intuos S" instead of a "Wacom Intuos S".

The link above describes the problem in more detail, along with a workaround. Briefly: you can force the tablet back into its PC mode by holding down the left- and right-most ExpressKeys at the same time for at least 3 seconds. You should see the tablet's LED turn off and then turn back on brighter than it was before.

closed 2022-03-17 by jigpu