linuxwacom/xf86-input-wacom #197: Keep the serial code alive?

whot, 2021-11-29:
I was reworking the driver a fair bit recently and I'm now wondering: is it worth keeping the ISDV4 serial backend alive? When was the last time Wacom released a compatible device for this code?

cc @Pinglinux, @jigpu

Pinglinux, 2021-11-29:
We can remove the ISDV4 serial code. There is no new devices with serial connection. For users need the code for their old systems, they can always use the last xf86-inout-wacom before this release.

whot, 2021-11-29:
It's not quite that easy, most users would use whatever the distros provide rather than self-compiling and while it was annoying to change for my patchset, the effort's been done. So keeping it doesn't matter too much in terms of maintenance but untested code is risky code.

So it's mostly a function of how old those devices are. I suspect by now they're all over a decade old? Do you remember the last time you released a device that was a serial one?

Pinglinux, 2021-11-29:
June 2012 was the last time we discussed new serial device patches. I don't think we had other serial device after that.

whot, 2021-11-29:
hmm, that's almost a decade, long enough I guess. I'd venture that most of those don't run the latest distributions anyway then. Thanks!

jigpu, 2022-01-04:
It looks like the ISDv4 code survived your series of pull requests. Should this issue be closed, or do we want to go through the effort of removing the code?

whot, 2022-01-04:
It did survive but it's completely untested (unless you did test it) and my only laptop with a serial device no longer boots. I'll send a PR to remove it asap.

PR #211 (2022-01-12): Remove support for serial devices from the driver
The serial code inside the driver is effectively unused and definitely
untested. It's been a decade since the last serial devices came out and few of
those will be in active use, the few that are are likely supported by
the inputattach code we keep in place for now.

Fixes #197
closed 2022-01-12 by jigpu