linuxwacom/xf86-input-wacom #208: Touchpad "natural scrolling" Intuos Pro

curiousercreative, 2021-12-15:
New drawing tablet, seems to be working nicely except that I cannot for the life of me get the touchpad scrolling to invert away from "natural scrolling" despite the natural scroll setting being present in mouse & touchpad settings. Results are the same regardless of USB vs bluetooth connection.

Wacom Intuos Pro (Medium) `$lsusb => 056a:0357 Wacom Co., Ltd PTH-660 [Intuos Pro (M)]`
X11 windowing
Pop!_OS 21.04
Kernel 5.13.0-7620-generic

Relates to https://github.com/linuxwacom/xf86-input-wacom/issues/28

jigpu, 2021-12-16:
If you have applied the change mentioned in the issue you linked to, then you may need to file an issue against xf86-input-libinput rather than our driver. Please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball to this issue so I can see which driver is in use. Once you're set up to use xf86-input-libinput the natural scrolling should work properly; if it doesn't I can direct you to that driver's bug tracker.

curiousercreative, 2021-12-19:
@jigpu thanks for the prompt response. I actually hadn't followed the instructions from the linked issue, for some reason I thought it was outdated. I've just now briefly tried those instructions without success, but I've also resigned myself to believing that leaving it as is (at least for now) is the better option. Thanks!

closed 2021-12-19 by curiousercreative